margelo/react-native-vision-camera #2805: 💭 error installing v4

mbenassarou, 2024-04-25:
### Question

Hello all, I am not very experimented sorry if my question is dumb
I have an error when I try to install v4 on an expo managed project
command : npx expo install react-native-vision-camera@4.0.0
error :
Error: Cannot find module './expo-plugin/withVisionCamera'
Require stack:
- /Users/x/project/node_modules/react-native-vision-camera/app.plugin.js
- /Users/x/project/node_modules/@expo/config-plugins/build/utils/plugin-resolver.js
- /Users/x/project/node_modules/@expo/cli/build/src/install/utils/autoAddConfigPlugins.js
- /Users/x/project/node_modules/@expo/cli/build/src/install/installAsync.js
- /Users/x/project/node_modules/@expo/cli/build/src/install/index.js
- /Users/x/project/node_modules/@expo/cli/build/bin/cli
- /Users/x/project/node_modules/expo/bin/cli

### What I tried

I just downgraded to v3.9.2
I don't know what to try actually
Can you help me ?

### VisionCamera Version

3.9.2

### Additional information

- [X] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22💭+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

nonam4, 2024-04-25:
Use `4.0.1` version. It have a fix for expo plugin

mbenassarou, 2024-04-26:
Thank you very much. I'm closing the issue