margelo/react-native-vision-camera #2806: [v3.9.2][Android] Race condition if takeFocus is called too frequently 🐛

ldalzottomp, 2024-04-25:
### What's happening?

I am not sure if V3 is still mainainted on Android but I have the following issue that I don't reproduce on V4.

If I frenetically call camera.focus(). Then, at some point the app crashes indicating a timeout in the camera precapture.

Note : I reproduce in the example app, if I dwngrade to 3.9.2 and disable the double tap gesture.

### Reproduceable Code

```tsx
TODO
```

### Relevant log output

```shell
TODO
```

### Camera Device

```json
{
  "formats": [],
  "sensorOrientation": "landscape-left",
  "hardwareLevel": "full",
  "maxZoom": 8,
  "minZoom": 1,
  "maxExposure": 20,
  "supportsLowLightBoost": false,
  "neutralZoom": 1,
  "physicalDevices": [
    "wide-angle-camera"
  ],
  "supportsFocus": true,
  "supportsRawCapture": false,
  "isMultiCam": false,
  "minFocusDistance": 10,
  "minExposure": -20,
  "name": "0 (BACK) androidx.camera.camera2",
  "hasFlash": true,
  "hasTorch": true,
  "position": "back",
  "id": "0"
}
```

### Device

Galaxy S9 - SM-G960F (Android 10, One UI 2.5)

### VisionCamera Version

3.9.2

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

DSKonstantin, 2024-04-29:
Hi there. The same on POCO. Any ideas?

mrousavy, 2024-04-29:
Is that still an issue on 4.x.x?

DSKonstantin, 2024-04-29:
But on 4.0.0 I see black screen on android...

ldalzottomp, 2024-04-30:
@mrousavy No I don't reproduce on V4 anymore. That was just for the record because I didn't know if the V3 is still maintained or not.

mrousavy, 2024-05-08:
Hey - so I think I'm going to close this now, as V4 is out and I think it makes more sense to dedicate my time to V4. It's just much more stable and the future of VisionCamera.

Thanks for the detailed issue report though.

closed 2024-05-08 by mrousavy