margelo/react-native-vision-camera #2808: 🐛 Android device reprots back camera has one physical device when three present

jkaufman, 2024-04-26:
### What's happening?

Moto G 5G 2022 is reporting only one physical device on the back camera, but there are three present.

### Reproduceable Code

```tsx
import { Camera } from 'react-native-vision-camera'
let devices = Camera.getAvailableCameraDevices()
console.log(devices)
```

### Relevant log output

```shell
n/a
```

### Camera Device

```json
{
	"formats": [
		{
			"autoFocusSystem": "contrast-detection",
			"fieldOfView": 75.04000877522147,
			"maxFps": 30,
			"maxISO": 6400,
			"minFps": 10,
			"minISO": 100,
			"photoHeight": 2296,
			"photoWidth": 4080,
			"supportsDepthCapture": false,
			"supportsPhotoHdr": false,
			"supportsVideoHdr": false,
			"videoHeight": 1080,
			"videoStabilizationModes": [

			],
			"videoWidth": 1920
		},
		{
			"autoFocusSystem": "contrast-detection",
			"fieldOfView": 75.04000877522147,
			"maxFps": 30,
			"maxISO": 6400,
			"minFps": 10,
			"minISO": 100,
			"photoHeight": 1792,
			"photoWidth": 4080,
			"supportsDepthCapture": false,
			"supportsPhotoHdr": false,
			"supportsVideoHdr": false,
			"videoHeight": 1080,
			"videoStabilizationModes": [

			],
			"videoWidth": 1920
		},
		{
			"autoFocusSystem": "contrast-detection",
			"fieldOfView": 75.04000877522147,
			"maxFps": 30,
			"maxISO": 6400,
			"minFps": 10,
			"minISO": 100,
			"photoHeight": 2160,
			"photoWidth": 3840,
			"supportsDepthCapture": false,
			"supportsPhotoHdr": false,
[... 2203 lines cut ...]
	"supportsFocus": true,
	"supportsLowLightBoost": false,
	"supportsRawCapture": false
}
```

### Device

Moto G 5G 2022

### VisionCamera Version

4.0.0

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [X] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

jkaufman, 2024-04-26:
[CameraX lists](https://developer.android.com/media/camera/camerax/devices) the "Moto G(22)" as tested and supported.

zzz08900, 2024-04-26:
It's just phone vendors not respecting Android spec, should not affect your app though since you'll just be using whatever the phone gives you. 
I have seen this multiple times, on multiple brands and models.

jkaufman, 2024-04-26:
Understood, thanks, so I shouldn't expect _any_ camera API to report these devices accurately. Only the OEM camera will use all lenses.

zzz08900, 2024-04-26:
Usually we go by capabilities and pixel count, in most cases select the most powerful device/the device with highest resolution makes the phone throw in all the cams it has.

But sometimes even the capabilities aren't trustworthy, cough, Samsung phones, cough. In those cases we had to make phone specific patches.

jkaufman, 2024-04-26:
Thanks, this is all very helpful. I have a pretty funky use case that benefits from a wide lens and lower-than-maximum capture resolution, but broad compatibility is most important.

mrousavy, 2024-04-29:
Yea, I use the default Camera API - if it is not listed there, then its the vendor's fault.

Download the [Camera2 API Probe app from the PlayStore](https://play.google.com/store/apps/details?id=com.airbeat.device.inspector) to see if that can detect it.

closed 2024-04-29 by mrousavy

jkaufman, 2024-04-29:
As expected, it reports a single rear-facing camera with Level 3 support. Interesting that there is no zoom.

```
Model:moto g stylus 5G (2022)
Manufacturer:motorola
Build:12
SDK:31
------------------------------------------
Camera ID: 0
Camera Facing: BACK
Hardware Support Level:LEVEL_3
Manual exposure:1
Auto exposure:1
Auto exposure, always flash:1
Auto exposure, auto flash:1
Auto exposure, auto flash, red eye reduction:0
AE Lock:1
Compensation Range: [-18, 18]
Compensation Step: 1/6
Target FPS  Range: [10, 10]
Target FPS  Range: [15, 15]
Target FPS  Range: [15, 20]
Target FPS  Range: [20, 20]
Target FPS  Range: [15, 24]
Target FPS  Range: [24, 24]
Target FPS  Range: [10, 30]
Target FPS  Range: [24, 30]
Target FPS  Range: [30, 30]
Target FPS  Range: [60, 60]
Manual focus:1
Auto focus:1
Auto focus macro:1
Auto focus continuous picture:1
Auto focus continuous video:1
Auto focus EDOF:0
White balance off:1
Automatic white balance:1
WB: cloudy day:1
WB: daylight:1
WB: fluorescent:1
WB: incandescent:1
WB: shade:1
WB: twilight:1
WB: warm fluorescent:1
AWB Lock:1
RawCapture:1
BurstCapture:1
availableCapability: BACKWARD_COMPATIBLE
availableCapability: CONSTRAINED_HIGH_SPEED_VIDEO
availableCapability: RAW
availableCapability: YUV_REPROCESSING
availableCapability: PRIVATE_REPROCESSING
availableCapability: READ_SENSOR_SETTINGS
availableCapability: MANUAL_SENSOR
availableCapability: BURST_CAPTURE
availableCapability: MANUAL_POST_PROCESSING
Stabilization Mode: Optical
Stabilization Mode: Video
previewSize: 4080 * 2296 (510:287)
previewSize: 4080 * 1792 (255:112)
previewSize: 3840 * 2160 (16:9)
previewSize: 3264 * 2448 (4:3)
previewSize: 3264 * 1836 (16:9)
previewSize: 3264 * 1432 (408:179)
previewSize: 2304 * 1728 (4:3)
previewSize: 2304 * 1296 (16:9)
previewSize: 2304 * 1012 (576:253)
previewSize: 1920 * 1080 (16:9)
previewSize: 1920 * 842 (960:421)
previewSize: 1600 * 1200 (4:3)
previewSize: 1600 * 900 (16:9)
previewSize: 1600 * 702 (800:351)
previewSize: 1558 * 720 (779:360)
previewSize: 1516 * 720 (379:180)
previewSize: 1440 * 1080 (4:3)
previewSize: 1280 * 960 (4:3)
previewSize: 1280 * 720 (16:9)
previewSize: 1280 * 562 (640:281)
previewSize: 1024 * 768 (4:3)
previewSize: 960 * 720 (4:3)
previewSize: 960 * 540 (16:9)
previewSize: 960 * 422 (480:211)
previewSize: 720 * 480 (3:2)
previewSize: 640 * 480 (4:3)
previewSize: 352 * 288 (11:9)
previewSize: 320 * 240 (4:3)
previewSize: 256 * 192 (4:3)
previewSize: 240 * 320 (4:3)
previewSize: 176 * 144 (11:9)
photoSize: 4080 * 2296 (510:287)
photoSize: 4080 * 1792 (255:112)
photoSize: 3840 * 2160 (16:9)
photoSize: 3264 * 2448 (4:3)
photoSize: 3264 * 1836 (16:9)
photoSize: 3264 * 1432 (408:179)
photoSize: 2304 * 1728 (4:3)
photoSize: 2304 * 1296 (16:9)
photoSize: 2304 * 1012 (576:253)
photoSize: 1920 * 1080 (16:9)
photoSize: 1920 * 842 (960:421)
photoSize: 1600 * 1200 (4:3)
photoSize: 1600 * 900 (16:9)
photoSize: 1600 * 702 (800:351)
photoSize: 1558 * 720 (779:360)
photoSize: 1516 * 720 (379:180)
photoSize: 1440 * 1080 (4:3)
photoSize: 1280 * 960 (4:3)
photoSize: 1280 * 720 (16:9)
photoSize: 1280 * 562 (640:281)
photoSize: 1024 * 768 (4:3)
photoSize: 960 * 720 (4:3)
photoSize: 960 * 540 (16:9)
photoSize: 960 * 422 (480:211)
photoSize: 720 * 480 (3:2)
photoSize: 640 * 480 (4:3)
photoSize: 352 * 288 (11:9)
photoSize: 320 * 240 (4:3)
photoSize: 256 * 192 (4:3)
photoSize: 240 * 320 (4:3)
photoSize: 176 * 144 (11:9)
videoSize: 4080 * 2296 (510:287)
videoSize: 4080 * 1792 (255:112)
videoSize: 3840 * 2160 (16:9)
videoSize: 3264 * 2448 (4:3)
videoSize: 3264 * 1836 (16:9)
videoSize: 3264 * 1432 (408:179)
videoSize: 2304 * 1728 (4:3)
videoSize: 2304 * 1296 (16:9)
videoSize: 2304 * 1012 (576:253)
videoSize: 1920 * 1080 (16:9)
videoSize: 1920 * 842 (960:421)
videoSize: 1600 * 1200 (4:3)
videoSize: 1600 * 900 (16:9)
videoSize: 1600 * 702 (800:351)
videoSize: 1558 * 720 (779:360)
videoSize: 1516 * 720 (379:180)
videoSize: 1440 * 1080 (4:3)
videoSize: 1280 * 960 (4:3)
videoSize: 1280 * 720 (16:9)
videoSize: 1280 * 562 (640:281)
videoSize: 1024 * 768 (4:3)
videoSize: 960 * 720 (4:3)
videoSize: 960 * 540 (16:9)
videoSize: 960 * 422 (480:211)
videoSize: 720 * 480 (3:2)
videoSize: 640 * 480 (4:3)
videoSize: 352 * 288 (11:9)
videoSize: 320 * 240 (4:3)
videoSize: 256 * 192 (4:3)
videoSize: 240 * 320 (4:3)
videoSize: 176 * 144 (11:9)
```

mrousavy, 2024-04-29:
lol, yea. this is Android for 'ya

jkaufman, 2024-04-29:
Hah, I got started with, what, API 7? I am _well aware_.