margelo/react-native-vision-camera #2813: 🐛 camera keeps being restarted when using codeScanner

cvarjao, 2024-04-27:
### What's happening?

The app seems to get into some sort of restarting camera loop which eventually cause it to crash

### Reproduceable Code

```tsx
// From Example https://github.com/mrousavy/react-native-vision-camera/blob/0e5a888c00fa4226d8acf7df6d56689d76e8ea71/package/example/src/CodeScannerPage.tsx#L72

<Camera
  style={StyleSheet.absoluteFill}
  device={device}
  isActive={isActive}
  codeScanner={codeScanner}
  torch={torch ? 'on' : 'off'}
  enableZoomGesture={true}
/>
```

### Relevant log output

```shell
I see the following lines repeating over and over:
00:13:24.307  I  captureNV21: Dimensions for the current request (1280x720) differ from the previous request (1280x960). Restarting camera
00:13:24.523  I  captureRGBA: Dimensions for the current request (1280x960) differ from the previous request (1280x720). Restarting camera
```

### Camera Device

```json
{
  "formats": [],
  "sensorOrientation": "landscape-left",
  "hardwareLevel": "limited",
  "maxZoom": 10,
  "minZoom": 1,
  "maxExposure": 9,
  "supportsLowLightBoost": false,
  "neutralZoom": 1,
  "physicalDevices": [
    "telephoto-camera"
  ],
  "supportsFocus": true,
  "supportsRawCapture": false,
  "isMultiCam": false,
  "minFocusDistance": 0,
  "minExposure": -9,
  "name": "0 (BACK) androidx.camera.camera2",
  "hasFlash": false,
  "hasTorch": false,
  "position": "back",
  "id": "0"
}
```

### Device

Android 9 Emulator (Pixel 2 API 28)

### VisionCamera Version

3.9.2 - 4.0.1

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

zzz08900, 2024-04-28:
Last time I checked, RNVC does all kinds of wired things within simulators. On iOS simulators it outright refuse to work.

mrousavy, 2024-04-29:
> RNVC does all kinds of wired things within simulators

Well it's hard to simulate Cameras. Also, CameraX handles that part. The logs appear to be from the Camera simulation, not something I wrote.

> On iOS simulators it outright refuse to work

Not my fault, Apple doesn't provide Cameras on Simulators. It's a _Simulator_, not an _Emulator_.

mrousavy, 2024-04-29:
Just to clarify, you are seeing exactly the same behaviour on 3.9.x and on 4.x.x? Because I'm sure that 4.x.x should work.

cvarjao, 2024-04-29:
That is fair. I am really trying to troubleshoot an issue with Android 9 device (black screen). I do not have access to the device, so I was looking for ways to reproduce the error.
I've tested specifically with 3.9.2 and 4.0.1

zzz08900, 2024-04-29:
I'm not intended to blame anything on RNVC.
I'm just stating the fact that this lib does not work with simulators and, potentially telling OP that he'd better get a real phone to test this lib out.

iqbaldev46, 2024-04-29:
> The actual problem I am trying to troubleshoot is that the camera show a back screen on some Android 9 devices (Specifically LG G6 H873. It seems fine on Samsung and a few other devices I've tried).

On 4.0.1, can you try to add the prop photoQualityBalance='speed' or photoQualityBalance='quality' and see if it fixes the black screen issue.

cvarjao, 2024-04-29:
I can try that. For clarity, we are only using the camera for reading QR code.

cvarjao, 2024-04-30:
I've got a hold of a LG Device which I can reproduce the error, not much in logcat that makes sense to me, but now I can play with different settings:

```text
15:03:45.117  I  Camera #0: Unavailable!
15:03:45.120  I  Updating CameraSession...
15:03:45.133  I  PreviewView onMeasure(1440, 2352)
15:03:45.133  D  coverSize :: 1920x1080 (0.5625), 1440x2352 (0.6122448979591837)
15:03:45.133  D  Fitted dimensions set: 1440x2560
15:03:45.165  I  PreviewView Surface created! Surface(name=null)/@0x592a6c
15:03:45.166  I  Setting Preview Output...
15:03:45.167  I  PreviewView Surface updated! Surface(name=null)/@0x592a6c 1920 x 1080
15:03:45.312  I  Camera 0: Opened!
15:03:45.382  I  Successfully configured Camera #0!
15:03:45.382  I  Resizing PreviewView to 1680 x 720...
15:03:45.384  I  Configuring Session for Camera #0...
15:03:45.386  I  Adding 1280 x 720 CodeScanner Output in Format #35...
15:03:45.410  I  PreviewView Surface updated! Surface(name=null)/@0x592a6c 1680 x 720
15:03:45.519  I  Considering local module com.google.mlkit.dynamite.barcode:10000 and remote module com.google.mlkit.dynamite.barcode:0
15:03:45.520  I  Selected local version of com.google.mlkit.dynamite.barcode
15:03:45.574  I  vendor::lge::hardware::configstore::V1_0::IConfigStore::getUBWCUsageFlag retrieved: 0 (default)
15:03:45.591  I  Camera 0: Creating Capture Session #1000... Hardware Level: 3} | Outputs: [android.hardware.camera2.params.OutputConfiguration@d89b816e]
15:03:45.592  I  Using legacy API (<28)
15:03:45.614  I  Camera 0: Capture Session #1000 configured!
15:03:45.617  I  Successfully configured Session with 1 outputs for Camera #0!
15:03:45.617  I  Successfully updated CameraSession Configuration! isActive: false
15:03:45.618  I  invokeOnInitialized()
15:03:45.618  I  invokeOnStopped()
15:03:45.620  I  Updating CameraSession Configuration...
15:03:45.621  W  Preview Output is null, aborting...
15:03:45.621  I  Successfully updated CameraSession Configuration! isActive: true
15:03:45.621  I  invokeOnStarted()
15:03:45.623  I  Updating CameraSession Configuration...
15:03:45.625  I  Configuring Session for Camera #0...
15:03:45.625  I  Closing BarcodeScanner..
15:03:45.631  I  Adding 1680 x 720 Preview Output...
15:03:45.634  I  Resizing PreviewView to 1680 x 720...
15:03:45.638  I  Adding 1280 x 720 CodeScanner Output in Format #35...
15:03:45.649  I  Considering local module com.google.mlkit.dynamite.barcode:10000 and remote module com.google.mlkit.dynamite.barcode:0
15:03:45.649  I  Selected local version of com.google.mlkit.dynamite.barcode
15:03:45.651  I  Camera 0: Creating Capture Session #1001... Hardware Level: 3} | Outputs: [android.hardware.camera2.params.OutputConfiguration@9f263cc6, android.hardware.camera2.params.OutputConfiguration@ec50e69b]
15:03:45.651  I  Using legacy API (<28)
15:03:45.690  I  Camera 0: Capture Session #1001 configured!
15:03:45.690  I  Camera 0: Capture Session #1000 closed!
15:03:45.691  I  Successfully configured Session with 2 outputs for Camera #0!
15:03:45.712  I  Successfully updated CameraSession Configuration! isActive: true
```

mrousavy, 2024-05-06:
@cvarjao That's VisionCamera 3. Can you please try with VisionCamera 4 and post the logs here? I doubt that the issue is reproduceable there.

cvarjao, 2024-05-08:
we updated to `4.0.1` and the problem with the physical device went away. So, we can close this one since the emulator issue might be unrelated. Thanks!

closed 2024-05-08 by cvarjao

jenskuhrjorgensen, 2025-05-14:
> Well it's hard to simulate Cameras. Also, CameraX handles that part. The logs appear to be from the Camera simulation, not something I wrote.
>
>
> Not my fault, Apple doesn't provide Cameras on [...]

@mrousavy does this answer mean, that it is currently expected that the Android emulator camera is flickering like in the OP video when using frame processors in the newest version of `react-native-vision-camera` (4.6.4)?