margelo/react-native-vision-camera #2814: 🐛 Camera really broken. Doesn't fix flat tires

raedle, 2024-04-28:
### What's happening?

My car has a flat tire. I took a picture of the flat tire with an app that uses your library, but the tire is still flat. Please fix this in your library! I’m still stuck on the side of the road, and I need this really urgent get home

### Reproduceable Code

```tsx
Repro is here: https://x.com/mrousavy/status/1784663046230516131
```

### Relevant log output

```shell
09:03:46 I ReactNativeJS: Successfully took picture
09:03:47 I App: Couldn't fix flat tire
```

### Camera Device

```json
{}
```

### Device

iPhone 3G

### VisionCamera Version

Latest

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

eledahl, 2024-04-29:
Might be an iOS issue. My HTC Dream fixed my flat fine.

mrousavy, 2024-04-29:
you forgot to ping me and say "urgent fix needed" 😂

mrousavy, 2024-04-29:
> Device: iPhone 3G

this has been fixed in the latest iPhone 4, please try to upgrade soon. 👍

closed 2024-04-29 by mrousavy