margelo/react-native-vision-camera #2815: Hello! Love the work you've done, but I am having issues trying to send frames from my mobile app to my flask backend over websockets.💭

HADO564, 2024-04-28:
### Question

So as the title says, I am having issues with managing and sending the frames over a socket connection to a server that will process the frames in a flask backend. I tried going through the old issues, and found that one of the solutions is to make your own plugin. So thats just what i tried doing. Followed the documentation available. But now im getting gradle errors, like
"XyzFrameProcessorPlugin.kt:3:28 Unresolved reference: frameprocessor". 
Apparently all of them are these types of errors, starting from MainApplication.kt to the new plugins. I used the automatic plugin builder CLI. 

I could share the repo if that helps. 

### What I tried

I tried using the steps mentioned in documentation. I dont know much about Java or Kotlin, but still, using the example plugin in the docs that just returns "cat", it throws the error during compile time, when kotlin or java is being compiled (in other words, the plugin). Some urgent help would be great.

### VisionCamera Version

4.01

### Additional information

- [X] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22💭+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

zzz08900, 2024-04-29:
Don't see any native code on your repo. Is that the repo for your frame processor plugin? If that is then you are doing it wrong, very wrong.

There's an issue on how to start creating your own frame processor plugin(opened by me, a few days ago), take a look at that and fix your approach.

zzz08900, 2024-04-29:
On a side note, the actual idea of frame processor plugin is that you can pass a frame to native side, process that frame with native (Java/KT/Swift) code and return results to js side, in almost real time.

So if you know nothing about Java/KT/Android then you could find yourself stranded in the middle of no where with no way out.

closed 2024-04-29 by mrousavy

HADO564, 2024-04-29:
Despite the changes, and following the docs exactly (both automatic and manual creation of plugins), the app crashes at compile time. Im coding in VsCode and using the command line android tools. The app compiles flawlessly if i dont create my own plugin. In the same repo under the "another one" commit are all the files that are being used to create a plugin. 

I think the issue is arising from incomplete dependencies as the error states, are you sure we dont need to mention them in the build.gradle files?

mrousavy, 2024-04-29:
Hey - it sounds like you're not very familiar with native development, might be good to get a good foundation of how things work in native Android and iOS first, or at least how the build toolchains work.

I'd recommend you to develop in Android Studio and Xcode, not VSCode.

Android Studio also would've told you that you were probably trying to use `frameprocessors`, not `frameprocessor`.
It'll give you hints about errors and tell you where you're going wrong.

HADO564, 2024-04-29:
Fair enough, is there a way to do a workaround instead of doing native development? I would rather very much be able to do this in javascript only. I tried using worklets, but to no avail. Just a pointer in the right direction would help, so that I dont spend hours pulling my hair over issues beyond my skills. 
Essentially just want to send frames from the frameprocessor to a backend for processing via socket io.

mrousavy, 2024-04-29:
A very trivial approach is to do `frame.toArrayBuffer()`, copy that into a `number[]`, use `runOnJS` to call fetch(..) and upload that to your server. Maybe you can find a JS based compression library for that as well, not sure.
But this is going to be REALLY slow.

That's what native plugins are for; they're faster, more efficient, can do stuff like good compression, etc.

HADO564, 2024-04-29:
Alright, thanks for your help man. I guess I'll have to go with an entire video instead of frame by frame processing. Might add some burden to the backend, but should work. 
Again, thanks dude.

vishal-affinsys, 2025-01-25:
Do you have any plugin which can send the data to the backend server, because I am also facing the exact similar issue, I am trying to connect to websocket and send base64 data over it, but this whole process is very slow

HADO564, 2025-01-27:
> Do you have any plugin which can send the data to the backend server, because I am also facing the exact similar issue, I am trying to connect to websocket and send base64 data over it, but this whole process is very slow

Unfortunately no. I ditched the idea because we were on a time schedule and couldn't experiment