margelo/react-native-vision-camera #2820: 🐛

jodrescher, 2024-04-29:
### What's happening?

Everything works perfectly in debug mode for Android. But when I install the app with “--mode=release”, the code in “runAsync” is not executed or the app crashes. I was able to trace it back so far that it makes a difference in FrameProcessorPlugins.ts if I change the line “const asyncContext = Worklets.createContext(‘VisionCamera.async’)” to “const asyncContext = Worklets.defaultContext”. Everything works with the DefaultContext, even in the release build. Could there be an error here, or am I doing something wrong in the usage? react-native-vision-camera:4.0.1 and react-native-worklets-core:1.2.0

### Reproduceable Code

```tsx
runAsync(frame, () => {
  'worklet';
  console.log('test');
  ...
});

tested also with example app, and wrap the existing frameprocessor code inside runAsyn:
const frameProcessor = useFrameProcessor((frame) => {
    'worklet'

    runAtTargetFps(10, () => {
      runAsync(frame, () => {
        'worklet'
        console.log(`${frame.timestamp}: ${frame.width}x${frame.height} ${frame.pixelFormat} Frame (${frame.orientation})`)
        examplePlugin(frame)
        exampleKotlinSwiftPlugin(frame)
      })
    })
  }, [])
```

### Relevant log output

```shell
no other log output or errors or something corresponding to the async function
```

### Camera Device

```json
{
                             "formats": [],
                             "hardwareLevel": "limited",
                             "hasFlash": true,
                             "hasTorch": true,
                             "id": "0",
                             "isMultiCam": false,
                             "maxExposure": 20,
                             "maxZoom": 8,
                             "minExposure": -30,
                             "minFocusDistance": 0,
                             "minZoom": 1,
                             "name": "0 (BACK) androidx.camera.camera2",
                             "neutralZoom": 1,
                             "physicalDevices": [
                             "wide-angle-camera"
                             ],
                             "position": "back",
                             "sensorOrientation": "landscape-left",
                             "supportsFocus": true,
                             "supportsLowLightBoost": false,
                             "supportsRawCapture": false
                             }
```

### Device

Samsung A55

### VisionCamera Version

4.0.1

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2024-04-29:
> no other log output or errors or something corresponding to the async function

I need log outputs or error messages or anything. Please post those, otherwise I cannot help you. On Android, use `adb logcat`.

mrousavy, 2024-04-29:
Ah nevermind you just posted error messages this second lol

mrousavy, 2024-04-29:
Just to clarify; this works fine in Debug, but fails in Release?

jodrescher, 2024-04-29:
> Just to clarify; this works fine in Debug, but fails in Release?

Yes, exactly, I had this with older versions of vision-camera and worklets-core too and hoped that the current version would fix it.

mrousavy, 2024-04-29:
I am looking at the code and I don't really see where the initialization is different, are you 100% sure that it works with the `Worklets.defaultContext`, but doesn't work with the `Worklets.createContext('VisionCamera.async')`?

jodrescher, 2024-04-29:
> I am looking at the code and I don't really see where the initialization is different, are you 100% sure that it works with the `Worklets.defaultContext`, but doesn't work with the `Worklets.createContext('VisionCamera.async')`?

> And does this crash on iOS as well, or only Android?

I have now tested the Example app again on a Samsung A55 (Android 14), a Samsung A41 (Android 12) and an iPhone 12 mini (iOS 17.4.1). The same result on all devices, everything works in debug mode, even with “Worklets.createContext(‘VisionCamera.async’)”. In release mode, however, the app crashes. If I change to “Worklets.defaultContext” it works on all devices, even in release mode.

mrousavy, 2024-04-29:
This is so weird. I mean we definitely want a separate Context in case you are using react-native-worklets-core for something else (which is usually the default context), so not sure why this is happening.

mrousavy, 2024-05-01:
Hey - I spent a little time investigating the parallel memory allocations and actually found & fixed a bug! Try this and see if this fixes your issue: https://github.com/mrousavy/react-native-vision-camera/pull/2829

mrousavy, 2024-05-01:
For me, `runAsync` runs perfectly fine now

jodrescher, 2024-05-02:
> Hey - I spent a little time investigating the parallel memory allocations and actually found & fixed a bug! Try this and see if this fixes your issue: #2829

I tested it with the current clean repo version and it sometimes works on iOS, but most of the time the app crashes (9 out of 10 cases). The fix is only for iOS? I get the following error message in XCode: Thread 9: EXC_BAD_ACCESS (code=1, address=0x80000000000000d0)

```shell
Thread 9 Queue : mrousavy/VisionCamera.video (serial)
#0	0x0000000102fcd4c4 in long std::__1::__libcpp_atomic_refcount_increment[abi:ue170006]<long>(long&) [inlined] at /Applications/Xcode.app/Contents/Developer/Platforms/iPhoneOS.platform/Developer/SDKs/iPhoneOS17.4.sdk/usr/include/c++/v1/__memory/shared_ptr.h:106
#1	0x0000000102fcd4c4 in std::__1::__shared_weak_count::__add_weak[abi:ue170006]() [inlined] at /Applications/Xcode.app/Contents/Developer/Platforms/iPhoneOS.platform/Developer/SDKs/iPhoneOS17.4.sdk/usr/include/c++/v1/__memory/shared_ptr.h:210
#2	0x0000000102fcd4c0 in std::__1::weak_ptr<RNWorklet::JsiWorkletContext>::weak_ptr(std::__1::weak_ptr<RNWorklet::JsiWorkletContext> const&) [inlined] at /Applications/Xcode.app/Contents/Developer/Platforms/iPhoneOS.platform/Developer/SDKs/iPhoneOS17.4.sdk/usr/include/c++/v1/__memory/shared_ptr.h:1722
#3	0x0000000102fcd4b4 in std::__1::weak_ptr<RNWorklet::JsiWorkletContext>::weak_ptr(std::__1::weak_ptr<RNWorklet::JsiWorkletContext> const&) [inlined] at /Applications/Xcode.app/Contents/Developer/Platforms/iPhoneOS.platform/Developer/SDKs/iPhoneOS17.4.sdk/usr/include/c++/v1/__memory/shared_ptr.h:1720
#4	0x0000000102fcd4b4 in std::__1::enable_shared_from_this<RNWorklet::JsiWorkletContext>::weak_from_this[abi:ue170006]() [inlined] at /Applications/Xcode.app/Contents/Developer/Platforms/iPhoneOS.platform/Developer/SDKs/iPhoneOS17.4.sdk/usr/include/c++/v1/__memory/shared_ptr.h:1963
#5	0x0000000102fcd4b4 in RNWorklet::JsiWorkletContext::invokeOnWorkletThread(std::__1::function<void (RNWorklet::JsiWorkletContext*, facebook::jsi::Runtime&)>&&) at ../react-native-vision-camera/package/example/node_modules/react-native-worklets-core/cpp/WKTJsiWorkletContext.cpp:146
#6	0x0000000102fd631c in RNWorklet::JsiWorkletContext::createCallInContext(facebook::jsi::Runtime&, facebook::jsi::Value const&, RNWorklet::JsiWorkletContext*)::$_5::operator()(facebook::jsi::Runtime&, facebook::jsi::Value const&, facebook::jsi::Value const*, unsigned long) const::'lambda'(std::__1::function<void (facebook::jsi::Runtime&)>&&)::operator()(std::__1::function<void (facebook::jsi::Runtime&)>&&) const [inlined] at ../react-native-vision-camera/package/example/node_modules/react-native-worklets-core/cpp/WKTJsiWorkletContext.cpp:309
#7	0x0000000102fd62cc in RNWorklet::JsiWorkletContext::createCallInContext(facebook::jsi::Runtime&, facebook::jsi::Value const&, RNWorklet::JsiWorkletContext*)::$_5::operator()(facebook::jsi::Runtime&, facebook::jsi::Value const&, facebook::jsi::Value const*, unsigned long) const::'lambda0'(facebook::jsi::Runtime&, std::__1::shared_ptr<RNWorklet::PromiseParameter>)::operator()(facebook::jsi::Runtime&, std::__1::shared_ptr<RNWorklet::PromiseParameter>) const [inlined] at ../react-native-vision-camera/package/example/node_modules/react-native-worklets-core/cpp/WKTJsiWorkletContext.cpp:344
#8	0x0000000102fd618c in decltype(std::declval<RNWorklet::JsiWorkletContext::createCallInContext(facebook::jsi::Runtime&, facebook::jsi::Value const&, RNWorklet::JsiWorkletContext*)::$_5::operator()(facebook::jsi::Runtime&, facebook::jsi::Value const&, facebook::jsi::Value const*, unsigned long) const::'lambda0'(facebook::jsi::Runtime&, std::__1::shared_ptr<RNWorklet::PromiseParameter>)&>()(std::declval<facebook::jsi::Runtime&>(), std::declval<std::__1::shared_ptr<RNWorklet::PromiseParameter>>())) std::__1::__invoke[abi:ue170006]<RNWorklet::JsiWorkletContext::createCallInContext(facebook::jsi::Runtime&, facebook::jsi::Value const&, RNWorklet::JsiWorkletContext*)::$_5::operator()(facebook::jsi::Runtime&, facebook::jsi::Value const&, facebook::jsi::Value const*, unsigned long) const::'lambda0'(facebook::jsi::Runtime&, std::__1::shared_ptr<RNWorklet::PromiseParameter>)&, facebook::jsi::Runtime&, std::__1::shared_ptr<RNWorklet::PromiseParameter>>(RNWorklet::JsiWorkletContext::createCallInContext(facebook::jsi::Runtime&, facebook::jsi::Value const&, RNWorklet::JsiWorkletContext*)::$_5::operator()(facebook::jsi::Runtime&, facebook::jsi::Value const&, facebook::jsi::Value const*, unsigned long) const::'lambda0'(facebook::jsi::Runtime&, std::__1::shared_ptr<RNWorklet::PromiseParameter>)&, facebook::jsi::Runtime&, std::__1::shared_ptr<RNWorklet::PromiseParameter>&&) [inlined] at /Applications/Xcode.app/Contents/Developer/Platforms/iPhoneOS.platform/Developer/SDKs/iPhoneOS17.4.sdk/usr/include/c++/v1/__type_traits/invoke.h:340
#9	0x0000000102fd6184 in void std::__1::__invoke_void_return_wrapper<void, true>::__call[abi:ue170006]<RNWorklet::JsiWorkletContext::createCallInContext(facebook::jsi::Runtime&, facebook::jsi::Value const&, RNWorklet::JsiWorkletContext*)::$_5::operator()(facebook::jsi::Runtime&, facebook::jsi::Value const&, facebook::jsi::Value const*, unsigned long) const::'lambda0'(facebook::jsi::Runtime&, std::__1::shared_ptr<RNWorklet::PromiseParameter>)&, facebook::jsi::Runtime&, std::__1::shared_ptr<RNWorklet::PromiseParameter>>(RNWorklet::JsiWorkletContext::createCallInContext(facebook::jsi::Runtime&, facebook::jsi::Value const&, RNWorklet::JsiW [... cut ...]
#59	0x00000001050467bc in _dispatch_client_callout ()
#60	0x00000001050498e0 in _dispatch_continuation_pop ()
#61	0x0000000105060cc8 in _dispatch_source_latch_and_call ()
#62	0x000000010505f718 in _dispatch_source_invoke ()
#63	0x000000010504e4a4 in _dispatch_lane_serial_drain ()
#64	0x000000010504f408 in _dispatch_lane_invoke ()
#65	0x000000010505c404 in _dispatch_root_queue_drain_deferred_wlh ()
#66	0x000000010505ba38 in _dispatch_workloop_worker_thread ()
#67	0x00000001fc8f4f20 in _pthread_wqthread ()

```

namnotfake, 2024-05-02:
When i use runAsyn but is not use runAsync app crash
[image]

my code:
const runOnJs = Worklets.createRunOnJS(() => {
    router.push("/");
  });

  const faceDetectionProcess = useFrameProcessor(
    (frame: Frame) => {
      "worklet";
      try {
        console.log(plugin);
        runAsync(frame, () => {
          "worklet";
          const internal = frame as FrameInternal;
          internal.incrementRefCount();
          // @ts-ignore
          const faces = plugin?.call(frame) as Face[];
          if (faces.length > 0) {
            runOnJs();
          }
        });
      } catch (error) {
        console.log(error);
      }
    },
    [runOnJs]
  );
> runs perfectly fine now

nonam4, 2024-05-02:
I think that #2829 is the fix for #2579 issue.

This `runAsync` crash is also happening on android but #2829 pull doesn't have any changes on android files.

mrousavy, 2024-05-02:
Okay yea well then this might be an actual Worklets bug.

I think it's really weird that it works with the `defaultContext` but not with the one we create.

nonam4, 2024-05-02:
On worklets-core `0.4.0` It works if we create the context from a string:
```
Worklets.createRunInContextFn(func..., 'AnyStringValue' )
``` 
Instead of:
```
const asyncContext = Worklets.createContext('VisionCamera.async')
Worklets.createRunInContextFn(func..., asyncContext)
```

I don't know why but it works

mrousavy, 2024-05-02:
> On worklets-core `0.4.0` It works if we create the context from a string:
>
> ```
> Worklets.createRunInContextFn(func..., 'AnyStringValue' )
> ```
>
> Instead of:
>
> ```
> const asyncContext = [...]

That didn't work. The type for the second parameter is `IWorkletContext | 'default'`, and you are not passing in a `IWorkletContext`, nor `'default'`, you are passing in a different string. TypeScript should've already told you that this isn't acceptable by adding squiggly red lines below that. However, Worklets Core didn't throw in that case, it just automatically fell back to the `'default'` context, see [`JsiWorkletApi::createRunInContextFn(...)`](https://github.com/margelo/react-native-worklets-core/blob/8a4e971e4b42411e24979d8cc2ccd6d09ad7c804/cpp/WKTJsiWorkletApi.h#L100-L105).

So your example of passing `'AnyStringValue'` is exactly the same as passing in `'default'`, or in react-native-worklets-core 1.x.x. where the string parameter has been removed, this is now exactly the same as `Worklets.defaultContext`.

mrousavy, 2024-05-02:
> yes yes, I had this TS warning but I rewrite it forcing my string to be from `any` type.

Yea, it's not a warning it's an error. Basically the force cast to any didn't change anything - as I said, `'default'` has the same effect as passing in any other string.

jodrescher, 2024-05-03:
> Take your time, I'm in no hury 😀
> Also thanks for your amazing work

I can only agree, thank you for all your hard work.

ismaelsousa, 2024-05-09:
> When i use runAsyn but is not use runAsync app crash. ![image](https://private-user-images.githubusercontent.com/147969833/327374432-ce091efc-a28e-467a-a232-3b5cb41d82c3.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTUyNzMzMzgsIm5iZiI6MTcxNTI3MzAzOCwicGF0aCI6Ii8xNDc5Njk4MzMvMzI3Mzc0NDMyLWNlMDkxZWZjLWEyOGUtNDY3YS1hMjMyLTNiNWNiNDFkODJjMy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNTA5JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDUwOVQxNjQzNThaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1kZTkzN2RiNDVjZWQ3ZGQ5MzYzMzVkZTEzODQxYjQwMDNhOWY3ZjFhMWZmMThjNzA2MzdhZDk0ZTJkZDU5Mjc3JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.dMeZugoA7PJibNCJZETB7BFlIqevmHzjIPmsAd8IKIA)
> 
> my code: const runOnJs = Worklets.createRunOnJS(() => { router.push("/"); });
> 
> const faceDetectionProcess = useFrameProcessor( (frame: Frame) => { "worklet"; try { console.log(plugin); runAsync(frame, () => { "worklet"; const internal = frame as FrameInternal; internal.incrementRefCount(); // @ts-ignore const faces = plugin?.call(frame) as Face[]; if (faces.length > 0) { runOnJs(); } }); } catch (error) { console.log(error); } }, [runOnJs] ); my config in babel.config module.exports = function (api) { api.cache(true); return { presets: ["babel-preset-expo"], plugins: [ [ "@tamagui/babel-plugin", { components: ["tamagui"], config: "./tamagui.config.ts", logTimings: true, disableExtraction: process.env.EXPO_PUBLIC_NODE_ENV === "development", }, ], "react-native-reanimated/plugin", ["react-native-worklets-core/plugin"], ], }; };
> 
> > runs perfectly fine now

I am facing the same

bglgwyng, 2024-05-10:
I got a tombstone for this.
[tombstone.zip](https://github.com/mrousavy/react-native-vision-camera/files/15270439/tombstone.zip)

nmajumder12, 2024-05-11:
> I tested it with the current clean repo version and it sometimes works on iOS, but most of the time the app crashes (9 out of 10 cases). The fix is only for iOS? I get the following error message in [...]

Running this in an LLM parser returns the following:

`Here's a breakdown of what the stack trace is showing:

It starts with a crash that seems to originate from a call to RNWorklet::JsiWorkletContext::invokeOnWorkletThread.
The crash occurs while trying to call a lambda function within createCallInContext.
There's a series of nested function calls involving std::__1::function and std::__1::shared_ptr.
The crash ultimately occurs while processing a sample buffer in the CameraSession.
To debug this issue, you might want to look into the implementation of RNWorklet::JsiWorkletContext and CameraSession. It appears that there might be some issue with memory management or thread synchronization when processing the sample buffer.`

Based on that information, it seems the error is occurring most likely due to context switching within the worklets module. I wonder if it is because `Worklets.createRunOnJs` is unaware of the non default context. Would explain why it is working on default context.

tfcornerstone, 2024-05-23:
@mrousavy The following is causing issues in `runAsync`, it resolves when you get rid of `visionCameraProcessorPlugin.call`. It seems `visionCameraProcessorPlugin.call` works perfectly fine in default context, but not as expected in `VisionCamera.async`. 

The only difference between default context and `VisionCamera.async` context I noticed is the fact that `workletInvoker` uses different dispatch Queues. I am assuming if something does go wrong on the custom dispatch queue on the VisionCameraProxyDelegate, it will stop calling the frameProcessor...which is why runAsync does not execute...or app crashes. 

```
var visionCameraProcessorPlugin = VisionCameraProxy.initFrameProcessorPlugin(
  MODULE_NAME
);

var scanCodes = (frame, options) => {
  "worklet";
  if (visionCameraProcessorPlugin == null) {
    throw new Error(`Failed to load Frame Processor Plugin "${MODULE_NAME}"!`);
  }
  frame.incrementRefCount && frame.incrementRefCount();
  const nativeCodes = visionCameraProcessorPlugin.call(
    frame,
    options
  );
  frame.decrementRefCount && frame.decrementRefCount();
  if (!Array.isArray(nativeCodes)) {
    console.log(
      `Native frame processor ${MODULE_NAME} failed to return a proper array!`,
      nativeCodes
    );
    return [];
  }
  return nativeCodes.slice().map((nativeBarcode) => normalizeNativeBarcode(nativeBarcode, frame));
};

const frameProcessor = useFrameProcessor(
        (frame) => {
            //Context: VisionCamera
            'worklet'
            runAsync(frame, () => {
                //Context: VisionCamera.async
                'worklet'
                const barcodes = scanCodes(frame, options);
                if (barcodes.length > 0) {
                    onBarcodeScannedJS(barcodes);
                }
            });

        },
        [],
    );

IvanByRool, 2024-06-09:
Hey,

I'm wondering if there is a fix or workaround for the problems with runAsync ?

I'm getting randomly RNWorklet::JsiWorklet::getCode while using it with ios.

I couldn't find a solution for this yet, did I miss something ?

If you need logs or anything to help you debug, let me know what you need. 

Thanks a lot, 
Ivan

tfcornerstone, 2024-07-10:
Related: https://github.com/margelo/react-native-worklets-core/pull/196/commits/552186a62b688c8e125e01fac51850a28086106a#diff-5626d984fb6c601e169493e0cdcd6f411cda50ad0672031afc78883b8289f60cR29 ...

mrEuler, 2024-08-07:
I have same issue with useSkiaFrameProcessor and runAsync worklet inside. Witht the debug mode everything works, but with a release mode -- 9/10 times have crashes. SkiaFrameProcessor by itslef works fine, I think I have in a runAsync issue.

zzz08900, 2024-08-14:
We also experienced the same issue, with `runAsync` in frame processor the app crashes after a few frames on Android. While on iOS the frame processor stopped executing in less than 1 sec.

Below is the crash scene from Android.
```
08-14 12:25:29.397 27950 27950 F DEBUG   : *** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
08-14 12:25:29.397 27950 27950 F DEBUG   : Build fingerprint: 'HUAWEI/CTRN-AL20/HWCTRN-Q:12/HUAWEICTR-AL20/104.2.0.178C00:user/release-keys'
08-14 12:25:29.397 27950 27950 F DEBUG   : Revision: '0'
08-14 12:25:29.397 27950 27950 F DEBUG   : ABI: 'arm'
08-14 12:25:29.397 27950 27950 F DEBUG   : Timestamp: 2024-08-14 12:25:28.914062761+0800
08-14 12:25:29.397 27950 27950 F DEBUG   : Process uptime: 0s
08-14 12:25:29.397 27950 27950 F DEBUG   : Cmdline: com.exampleReactNativeApp
08-14 12:25:29.397 27950 27950 F DEBUG   : pid: 27635, tid: 27682, name: ionCamera.video  >>> com.exampleReactNativeApp <<<
08-14 12:25:29.397 27950 27950 F DEBUG   : uid: 10213
08-14 12:25:29.397 27950 27950 F DEBUG   : signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x6e2f6574
08-14 12:25:29.397 27950 27950 F DEBUG   :     r0  6e2f6574  r1  69626f4d  r2  c253ace8  r3  ffffffff
08-14 12:25:29.397 27950 27950 F DEBUG   :     r4  efbe4f10  r5  efbe4f10  r6  b86254a0  r7  c253ac58
08-14 12:25:29.397 27950 27950 F DEBUG   :     r8  d2b64a40  r9  c25fac84  r10 b8624dd0  r11 b8625224
08-14 12:25:29.397 27950 27950 F DEBUG   :     ip  c25fbc04  sp  c253ac08  lr  c25e43f1  pc  c25d9b3a
08-14 12:25:29.397 27950 27950 F DEBUG   : backtrace:
08-14 12:25:29.397 27950 27950 F DEBUG   :       #00 pc 00039b3a  /data/app/~~IpN2405MYXlSHrcJwd3tiA==/com.exampleReactNativeApp-PwM3DMOnX0PEvhkK9L6J6Q==/base.apk!librnworklets.so (RNWorklet::JsiWorkletContext::invokeOnWorkletThread(std::__ndk1::function<void (RNWorklet::JsiWorkletContext*, facebook::jsi::Runtime&)>&&)+78) (BuildId: cfa9d690ac39aadc28c7bcb883b822bdba60c09a)
08-14 12:25:29.397 27950 27950 F DEBUG   :       #01 pc 000443ed  /data/app/~~IpN2405MYXlSHrcJwd3tiA==/com.exampleReactNativeApp-PwM3DMOnX0PEvhkK9L6J6Q==/base.apk!librnworklets.so (BuildId: cfa9d690ac39aadc28c7bcb883b822bdba60c09a)
08-14 12:25:29.397 27950 27950 F DEBUG   :       #02 pc 0004b473  /data/app/~~IpN2405MYXlSHrcJwd3tiA==/com.exampleReactNativeApp-PwM3DMOnX0PEvhkK9L6J6Q==/base.apk!librnworklets.so (RNWorklet::JsiPromiseWrapper::runComputation(facebook::jsi::Runtime&, std::__ndk1::function<void (facebook::jsi::Runtime&, std::__ndk1::shared_ptr<RNWorklet::PromiseParameter>)>)+90) (BuildId: cfa9d690ac39aadc28c7bcb883b822bdba60c09a)
08-14 12:25:29.397 27950 27950 F DEBUG   :       #03 pc 0004b2a1  /data/app/~~IpN2405MYXlSHrcJwd3tiA==/com.exampleReactNativeApp-PwM3DMOnX0PEvhkK9L6J6Q==/base.apk!librnworklets.so (RNWorklet::JsiPromiseWrapper::createPromiseWrapper(facebook::jsi::Runtime&, std::__ndk1::function<void (facebook::jsi::Runtime&, std::__ndk1::shared_ptr<RNWorklet::PromiseParameter>)>)+88) (BuildId: cfa9d690ac39aadc28c7bcb883b822bdba60c09a)
08-14 12:25:29.397 27950 27950 F DEBUG   :       #04 pc 00042047  /data/app/~~IpN2405MYXlSHrcJwd3tiA==/com.exampleReactNativeApp-PwM3DMOnX0PEvhkK9L6J6Q==/base.apk!librnworklets.so (BuildId: cfa9d690ac39aadc28c7bcb883b822bdba60c09a)
08-14 12:25:29.397 27950 27950 F DEBUG   :       #05 pc 0003a209  /data/app/~~IpN2405MYXlSHrcJwd3tiA==/com.exampleReactNativeApp-PwM3DMOnX0PEvhkK9L6J6Q==/base.apk!libhermes.so (BuildId: f966cb42e732ea0b25a04b1b206486c4c8941c29)
08-14 12:25:29.397 27950 27950 F DEBUG   :       #06 pc 00039fa5  /data/app/~~IpN2405MYXlSHrcJwd3tiA==/com.exampleReactNativeApp-PwM3DMOnX0PEvhkK9L6J6Q==/base.apk!libhermes.so (BuildId: f966cb42e732ea0b25a04b1b206486c4c8941c29)
08-14 12:25:29.397 27950 27950 F DEBUG   :       #07 pc 0003e977  /data/app/~~IpN2405MYXlSHrcJwd3tiA==/com.exampleReactNativeApp-PwM3DMOnX0PEvhkK9L6J6Q==/base.apk!libhermes.so (BuildId: f966cb42e732ea0b25a04b1b206486c4c8941c29)
08-14 12:25:29.397 27950 27950 F DEBUG   :       #08 pc 0004a32f  /data/app/~~IpN2405MYXlSHrcJwd3tiA==/com.exampleReactNativeApp-PwM3DMOnX0PEvhkK9L6J6Q==/base.apk!libhermes.so (BuildId: f966cb42e732ea0b25a04b1b206486c4c8941c29)
08-14 12:25:29.397 27950 27950 F DEBUG   :       #09 pc 0004b7a5  /data/app/~~IpN2405MYXlSHrcJwd3tiA==/com.exampleReactNativeApp-PwM3DMOnX0PEvhkK9L6J6Q==/base.apk!libhermes.so (BuildId: f966cb42e732ea0b25a04b1b206486c4c8941c29)
08-14 12:25:29.397 27950 27950 F DEBUG   :       #10 pc 0004af43  /data/app/~~IpN2405MYXlSHrcJwd3tiA==/com.exampleReactNativeApp-PwM3DMOnX0PEvhkK9L6J6Q==/base.apk!libhermes.so (BuildId: f966cb42e732ea0b25a04b1b206486c4c8941c29)
08-14 12:25:29.397 27950 27950 F DEBUG   :       #11 pc 00050c19  /data/app/~~IpN2405MYXlSHrcJwd3tiA==/com.exampleReactNativeApp-PwM3DMOnX0PEvhkK9L6J6Q==/base.apk!libhermes.so (BuildId: f966cb42e732ea0b25a04b1b206486c4c8941c29)
08-14 12:25:29.397 27950 27950 F DEBUG   :       #12 pc 0003ea61  /data/app/~~IpN2405MYXlSHrcJwd3tiA==/com.exampleReactNativeApp-PwM3DMOnX0PEvhkK9L6J6Q==/base.apk!libhermes.so (BuildId: f966cb42e732ea0b25a04b1b206486c4c8941c29)
08-14 12:25:29.397 27950 27950 F DEBUG   :       #13 pc 00034dd5  /data/app/~~IpN2405MYXlSHrcJwd3tiA==/com.exampleReactNativeApp-PwM3DMOnX0PEvhkK9L6J6Q==/base.apk!libhermes.so (BuildId: f966cb42e732ea0b25a04b1b206486c4c8941c29)
08-14 12:25:29.397 27950 27950 F DEBUG   :       #14 pc 0003068b  /data/app/~~IpN2405MYXlSHrcJwd3tiA==/com.exampleReactNativeApp-PwM3DMOnX0PEvhkK9L6J6Q==/base.apk!libVisionCamera.so (RNWorklet::JsiWor [... cut ...]
08-14 12:25:29.398 27950 27950 F DEBUG   :       #83 pc 00062e13  /apex/com.android.runtime/lib/bionic/libc.so (__start_thread+30) (BuildId: 0b1943cebf917c185cd30ce9e6b378b9)
```

If we modify `runAsync.ts` and replace `const asyncContext = Worklets.createContext('VisionCamera.async')` with `const asyncContext = Worklets.defaultContext` as suggested by OP, on Android side frame processor seems to be stable. (Did not test with iOS since we only use runAsync to avoid stutter on Android).

I would suggest the forementioned modification as one viable workaound for anyone dire need.

Package versions:
react-native-vision-camera: 4.5.0
react-native-worklets-core: 1.3.3

mrEuler, 2024-08-14:
I only found a workaround to change the context of runAsync execution from .createContext(...) to just .defaultContext;
Because I had also issue, that worklet wasn't working on ios in RELEASE mode.

pepperav, 2024-08-23:
> We also experienced the same issue, with `runAsync` in frame processor the app crashes after a few frames on Android. While on iOS the frame processor stopped executing in less than 1 sec.
> 
> Below is the crash scene from Android.
> 
> ```
> 08-14 12:25:29.397 27950 27950 F DEBUG   : *** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
> 08-14 12:25:29.397 27950 27950 F DEBUG   : Build fingerprint: 'HUAWEI/CTRN-AL20/HWCTRN-Q:12/HUAWEICTR-AL20/104.2.0.178C00:user/release-keys'
> 08-14 12:25:29.397 27950 27950 F DEBUG   : Revision: '0'
> 08-14 12:25:29.397 27950 27950 F DEBUG   : ABI: 'arm'
> 08-14 12:25:29.397 27950 27950 F DEBUG   : Timestamp: 2024-08-14 12:25:28.914062761+0800
> 08-14 12:25:29.397 27950 27950 F DEBUG   : Process uptime: 0s
> 08-14 12:25:29.397 27950 27950 F DEBUG   : Cmdline: com.exampleReactNativeApp
> 08-14 12:25:29.397 27950 27950 F DEBUG   : pid: 27635, tid: 27682, name: ionCamera.video  >>> com.exampleReactNativeApp <<<
> 08-14 12:25:29.397 27950 27950 F DEBUG   : uid: 10213
> 08-14 12:25:29.397 27950 27950 F DEBUG   : signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x6e2f6574
> 08-14 12:25:29.397 27950 27950 F DEBUG   :     r0  6e2f6574  r1  69626f4d  r2  c253ace8  r3  ffffffff
> 08-14 12:25:29.397 27950 27950 F DEBUG   :     r4  efbe4f10  r5  efbe4f10  r6  b86254a0  r7  c253ac58
> 08-14 12:25:29.397 27950 27950 F DEBUG   :     r8  d2b64a40  r9  c25fac84  r10 b8624dd0  r11 b8625224
> 08-14 12:25:29.397 27950 27950 F DEBUG   :     ip  c25fbc04  sp  c253ac08  lr  c25e43f1  pc  c25d9b3a
> 08-14 12:25:29.397 27950 27950 F DEBUG   : backtrace:
> 08-14 12:25:29.397 27950 27950 F DEBUG   :       #00 pc 00039b3a  /data/app/~~IpN2405MYXlSHrcJwd3tiA==/com.exampleReactNativeApp-PwM3DMOnX0PEvhkK9L6J6Q==/base.apk!librnworklets.so (RNWorklet::JsiWorkletContext::invokeOnWorkletThread(std::__ndk1::function<void (RNWorklet::JsiWorkletContext*, facebook::jsi::Runtime&)>&&)+78) (BuildId: cfa9d690ac39aadc28c7bcb883b822bdba60c09a)
> 08-14 12:25:29.397 27950 27950 F DEBUG   :       #01 pc 000443ed  /data/app/~~IpN2405MYXlSHrcJwd3tiA==/com.exampleReactNativeApp-PwM3DMOnX0PEvhkK9L6J6Q==/base.apk!librnworklets.so (BuildId: cfa9d690ac39aadc28c7bcb883b822bdba60c09a)
> 08-14 12:25:29.397 27950 27950 F DEBUG   :       #02 pc 0004b473  /data/app/~~IpN2405MYXlSHrcJwd3tiA==/com.exampleReactNativeApp-PwM3DMOnX0PEvhkK9L6J6Q==/base.apk!librnworklets.so (RNWorklet::JsiPromiseWrapper::runComputation(facebook::jsi::Runtime&, std::__ndk1::function<void (facebook::jsi::Runtime&, std::__ndk1::shared_ptr<RNWorklet::PromiseParameter>)>)+90) (BuildId: cfa9d690ac39aadc28c7bcb883b822bdba60c09a)
> 08-14 12:25:29.397 27950 27950 F DEBUG   :       #03 pc 0004b2a1  /data/app/~~IpN2405MYXlSHrcJwd3tiA==/com.exampleReactNativeApp-PwM3DMOnX0PEvhkK9L6J6Q==/base.apk!librnworklets.so (RNWorklet::JsiPromiseWrapper::createPromiseWrapper(facebook::jsi::Runtime&, std::__ndk1::function<void (facebook::jsi::Runtime&, std::__ndk1::shared_ptr<RNWorklet::PromiseParameter>)>)+88) (BuildId: cfa9d690ac39aadc28c7bcb883b822bdba60c09a)
> 08-14 12:25:29.397 27950 27950 F DEBUG   :       #04 pc 00042047  /data/app/~~IpN2405MYXlSHrcJwd3tiA==/com.exampleReactNativeApp-PwM3DMOnX0PEvhkK9L6J6Q==/base.apk!librnworklets.so (BuildId: cfa9d690ac39aadc28c7bcb883b822bdba60c09a)
> 08-14 12:25:29.397 27950 27950 F DEBUG   :       #05 pc 0003a209  /data/app/~~IpN2405MYXlSHrcJwd3tiA==/com.exampleReactNativeApp-PwM3DMOnX0PEvhkK9L6J6Q==/base.apk!libhermes.so (BuildId: f966cb42e732ea0b25a04b1b206486c4c8941c29)
> 08-14 12:25:29.397 27950 27950 F DEBUG   :       #06 pc 00039fa5  /data/app/~~IpN2405MYXlSHrcJwd3tiA==/com.exampleReactNativeApp-PwM3DMOnX0PEvhkK9L6J6Q==/base.apk!libhermes.so (BuildId: f966cb42e732ea0b25a04b1b206486c4c8941c29)
> 08-14 12:25:29.397 27950 27950 F DEBUG   :       #07 pc 0003e977  /data/app/~~IpN2405MYXlSHrcJwd3tiA==/com.exampleReactNativeApp-PwM3DMOnX0PEvhkK9L6J6Q==/base.apk!libhermes.so (BuildId: f966cb42e732ea0b25a04b1b206486c4c8941c29)
> 08-14 12:25:29.397 27950 27950 F DEBUG   :       #08 pc 0004a32f  /data/app/~~IpN2405MYXlSHrcJwd3tiA==/com.exampleReactNativeApp-PwM3DMOnX0PEvhkK9L6J6Q==/base.apk!libhermes.so (BuildId: f966cb42e732ea0b25a04b1b206486c4c8941c29)
> 08-14 12:25:29.397 27950 27950 F DEBUG   :       #09 pc 0004b7a5  /data/app/~~IpN2405MYXlSHrcJwd3tiA==/com.exampleReactNativeApp-PwM3DMOnX0PEvhkK9L6J6Q==/base.apk!libhermes.so (BuildId: f966cb42e732ea0b25a04b1b206486c4c8941c29)
> 08-14 12:25:29.397 27950 27950 F DEBUG   :       #10 pc 0004af43  /data/app/~~IpN2405MYXlSHrcJwd3tiA==/com.exampleReactNativeApp-PwM3DMOnX0PEvhkK9L6J6Q==/base.apk!libhermes.so (BuildId: f966cb42e732ea0b25a04b1b206486c4c8941c29)
> 08-14 12:25:29.397 27950 27950 F DEBUG   :       #11 pc 00050c19  /data/app/~~IpN2405MYXlSHrcJwd3tiA==/com.exampleReactNativeApp-PwM3DMOnX0PEvhkK9L6J6Q==/base.apk!libhermes.so (BuildId: f966cb42e732ea0b25a04b1b206486c4c8941c29)
> 08-14 12:25:29.397 27950 27950 F DEBUG   :       #12 pc 0003ea61  /data/app/~~IpN2405MYXlSHrcJwd3tiA==/com.exampleReactNativeApp-PwM3DMOnX0PEvhkK9L6J6Q==/base.apk!libhermes.so (BuildId: f966cb42e732ea0b25a04b1b206486c4c8941c29)
> 08-14 12:25:29.397 27950 27950 F DEBUG   :       #13 pc 00034dd5  /data/app/~~IpN2405MYXlSHrcJwd3tiA==/com.exampleReactNativeApp-PwM3DMOnX0PEvhkK9L6J6Q==/base.apk!libhermes.so (BuildId: f966cb42e732ea0b25a04b1b206486c4c8941c29)
[... cut ...]
> If we modify `runAsync.ts` and replace `const asyncContext = Worklets.createContext('VisionCamera.async')` with `const asyncContext = Worklets.defaultContext` as suggested by OP, on Android side frame processor seems to be stable. (Did not test with iOS since we only use runAsync to avoid stutter on Android).
> 
> I would suggest the forementioned modification as one viable workaound for anyone dire need.
> 
> By the look of things this should be a `react-native-worklets-core` bug.
> 
> Package versions: react-native-vision-camera: 4.5.0 react-native-worklets-core: 1.3.3

I've tested both iOS and Android and with this _workaround_ it works!

developerdanx, 2024-09-09:
Same issue here with package versions:

```
 react-native-vision-camera: 4.5.2
 react-native-worklets-core: 1.3.3
```

valentinchelle, 2024-11-03:
any way to fix this crash?

mattbell87, 2024-11-18:
Had this issue as well with up to date packages. Ended up creating a little shell script to patch the context until it's fixed.

Here it is if it helps anyone, hopefully we don't need it for long.

```sh
#!/bin/sh
set -e 

# Patch node_modules/react-native-vision-camera/src/frame-processors/runAsync.ts so that it can be used in a release build. 

pushd node_modules/react-native-vision-camera/src/frame-processors

sed -i 's/const asyncContext = Worklets.createContext('\''VisionCamera.async'\'')/const asyncContext = Worklets.defaultContext/g' runAsync.ts

popd

echo "Done" 
```

spsaucier, 2024-11-26:
I created a patch to change the context of runAsync execution from `.createContext(...)` to just `.defaultContext`, but I'm still getting the error periodically - seemingly just as often.

K4GOS, 2024-12-01:
I can confirm that runAsync works in dev but not in release. Replacing `const asyncContext = Worklets.createContext(''VisionCamera.async'')` with `const asyncContext = Worklets.defaultContext` did not help either.

ghashi, 2025-02-06:
For me it helped replacing `const asyncContext = Worklets.createContext(''VisionCamera.async'')` with `const asyncContext = Worklets.defaultContext`. It works on both iOS and Android

Here's my patch

react-native-vision-camera+4.6.3.patch
```
diff --git a/node_modules/react-native-vision-camera/src/frame-processors/runAsync.ts b/node_modules/react-native-vision-camera/src/frame-processors/runAsync.ts
index e98fab5..1c84bdd 100644
--- a/node_modules/react-native-vision-camera/src/frame-processors/runAsync.ts
+++ b/node_modules/react-native-vision-camera/src/frame-processors/runAsync.ts
@@ -16,7 +16,8 @@ try {
   const Worklets = WorkletsProxy.Worklets
   isAsyncContextBusy = Worklets.createSharedValue(false)
 
-  const asyncContext = Worklets.createContext('VisionCamera.async')
+  // https://github.com/mrousavy/react-native-vision-camera/issues/2820 
+  const asyncContext = Worklets.defaultContext
   runOnAsyncContext = asyncContext.createRunAsync((frame: Frame, func: () => void) => {
     'worklet'
     try {
```

versions of some packages:
```
    "react": "18.2.0",
    "react-native": "0.74.5",
    "react-native-worklets-core": "1.5.0",
    "react-native-vision-camera": "4.6.3",
```

Before the patch I was seeing the following error (on Sentry) on Android:

```
RNWorklet::JsiWorkletContext::invokeOnWorkletThread

SIGSEGV
Segfault
```

And I was seeing the following error on iOS:
```
WatchdogTermination
The OS watchdog terminated your app, possibly because it overused RAM.
```

adampolczynski, 2025-04-14:
did the same as @ghashi , works as expected on both platforms

jslok, 2025-04-16:
runAsync will work for me if I only have some simple code within it like a console.log. If I have a resize plugin then it will crash after a few seconds. I tried the defaultContext fix but it does not work for me. Old and new architecture, with/without reanimated, debug/release doesn't make a difference.

rn 0.75.5
rnvc 4.6.4
rnwc 1.5.0

```
04-15 18:07:03.057   859  1566 I G3ACoordinator: rls_manager.cc:335: RlsManager::TofUnregister successful.
04-15 18:07:03.057   859  1568 I Lyric   : focus_lens_actuator_driver_impl_base.cc:972: act-rear-normal: Total events dequeued: 96, failed events: 0
04-15 18:07:03.058   859  1568 I Lyric   : ois_driver_proxy.cc:106: ois-rear-normal sampling intervals: (count=187, min=1.59351ms, max=7.49381ms, mean=4.98185ms, stddev=0.464247ms, skew=-1.99378, kurtosis=26.6122)
04-15 18:07:03.058   859  1568 I Lyric   : ois_driver.cc:461: ois-rear-normal Total event deuqued: 95, failed event: 0
04-15 18:07:03.058   885  1853 I goog_range-sensor: Enable: 0
04-15 18:07:03.059   885  1853 I rlsservice: DisableWork mode 3, enable count 0, handle:e0bb886, active_period:30, sensor type:1
04-15 18:07:03.061   560   560 D CompositionEngine: Layer: com.visioncamerascanner/com.visioncamerascanner.MainActivity#92337 had an out of bounds transparent region
04-15 18:07:03.061   560   560 D Region  :   Region transparentRegionHint (this=0xb40000724bf0b250, count=1)
04-15 18:07:03.061   560   560 D Region  :     [-61, 136,  -2, 496]
04-15 18:07:03.062  1149  1149 D AOC     : A3:MSG: lsm6dsv_device.cc, 732: USF: Lsm6dsvDevice: Sensor 4: Enabled 0, odr_index 0, period_ns 0.
04-15 18:07:03.063  1149  1149 D AOC     : A3:MSG: lsm6dsv_fifo.cc, 755: USF: Lsm6dsvFifo: ODR: XL/G/T/TS: 60/60/15/7 WM: 6 Intr: 24.
04-15 18:07:03.064  1149  1149 D AOC     : A3:MSG: lsm6dsv_device.cc, 641: USF: Lsm6dsvDevice: Fifo was disabled for 1072591ns.
04-15 18:07:03.065  1149  1149 D AOC     : A3:MSG: usf_sensor.cc, 1713: USF: Received stop sampling request for LSM6DSV Stationary Detect (active:-1).
04-15 18:07:03.065  1149  1149 D AOC     : A3:MSG: usf_sensor.cc, 2169: USF: Sampling configuration 69461 not found.
04-15 18:07:03.066  1149  1149 D AOC     : A3:MSG: usf_sensor_client.cc, 161: USF: Received kErrNotFound when disabling one shot sensor.
04-15 18:07:03.066  1149  1149 D AOC     : A3:MSG: usf_sensor.cc, 1705: USF: Received start sampling request for LSM6DSV Motion Detect (active:1).
04-15 18:07:03.067  1149  1149 D AOC     : A3:MSG: lsm6dsv_device.cc, 732: USF: Lsm6dsvDevice: Sensor 3: Enabled 1, odr_index 1, period_ns 500000000.
04-15 18:07:03.068  1149  1149 D AOC     : A3:MSG: lsm6dsv_fifo.cc, 755: USF: Lsm6dsvFifo: ODR: XL/G/T/TS: 60/60/15/7 WM: 6 Intr: 24.
04-15 18:07:03.068   859  1568 W Lyric   : device_context.cc:362: NotifyCaptureSessionClose: No OisController found for physical camera RearWide. Status: NOT_FOUND: Failed to find controller by type and camera id(RearWide); Controller with ID: optical image stabilization controller for camera: RearWide not present in device_context
04-15 18:07:03.068   859  1918 I StreamBufferCacheManager: ~StreamBufferCacheManager: Destroying stream buffer cache manager.
04-15 18:07:03.068   859 15467 I StreamBufferCacheManager: WorkloadThreadLoop: Exiting stream buffer cache manager workload thread.
04-15 18:07:03.069   859  1918 I Lyric   : camera_latency_analyzer.cc:100: [CLA] Cam0-Close
04-15 18:07:03.069   859  1918 I Lyric   : camera_latency_analyzer.cc:102: [CLA] Flush latency: 16.1079
04-15 18:07:03.069   859  1918 I Lyric   : camera_latency_analyzer.cc:102: [CLA] <-- IDLE --> latency: 3.88924
04-15 18:07:03.069   859  1918 I Lyric   : camera_latency_analyzer.cc:102: [CLA] Close latency: 88.0517
04-15 18:07:03.069   859  1918 I Lyric   : camera_latency_analyzer.cc:102: [CLA] Overall latency: 108.057
04-15 18:07:03.069   859  1918 I Lyric   : camera_latency_analyzer.cc:102: [CLA] HAL Total latency: 104.156
04-15 18:07:03.069  1074  1173 W Camera3-OutputStream: disconnectLocked: While disconnecting stream 0 from native window, the native window died from under us
04-15 18:07:03.069  1074  1173 I CameraLatencyHistogram: Stream 0 dequeueBuffer latency histogram (39) samples:
04-15 18:07:03.069  1074  1173 I CameraLatencyHistogram:         5     10     15     20     25     30     35     40     45    inf (max ms)
04-15 18:07:03.069  1074  1173 I CameraLatencyHistogram:      92.31   0.00   2.56   0.00   0.00   0.00   0.00   2.56   0.00   2.56 (%)
04-15 18:07:03.069  1074  1173 W Camera3-OutputStream: disconnectLocked: While disconnecting stream 1 from native window, the native window died from under us
04-15 18:07:03.069  1074  1173 I CameraLatencyHistogram: Stream 1 dequeueBuffer latency histogram (39) samples:
04-15 18:07:03.069  1074  1173 I CameraLatencyHistogram:         5     10     15     20     25     30     35     40     45    inf (max ms)
04-15 18:07:03.069  1074  1173 I CameraLatencyHistogram:      100.00   0.00   0.00   0.00   0.00   0.00   0.00   0.00   0.00   0.00 (%)
04-15 18:07:03.069  1149  1149 D AOC     : A3:MSG: lsm6dsv_device.cc, 641: USF: Lsm6dsvDevice: Fifo was disabled for 1012004ns.
04-15 18:07:03.069  1074  1173 E IPCThreadState: Binder transaction failure. id: 153439511, BR_*: 29189, error: -22 (Invalid argument)
04-15 18:07:03.069  1074  1173 W Camera3-OutputStream: disconnectLocked: While disconnecting stream 2 from native window, the native window died from under us
[... cut ...]
04-15 18:07:03.261  1074  2317 D Camera2ClientBase: Camera 0: Shutting down
04-15 18:07:03.261  1074  2317 I Camera3-Device: disconnectImpl: E
04-15 18:07:03.261  1074  2317 I Camera2ClientBase: ~Camera2ClientBase: Client object's dtor for Camera Id 0 completed. Client was: com.visioncamerascanner (PID 15313, UID 10394)
04-15 18:07:03.262  1074  2317 I Camera3-Device: disconnectImpl: E
04-15 18:07:03.262  1074  2317 I Camera3DeviceInjectionMethods: injectionDisconnectImpl: Injection camera disconnect
```

danielkasprzak, 2025-08-14:
This issue is still a thing.

Versions
```
"react": "19.0.0", 
"react-native": "0.78.0", "react-native-worklets-core": "1.6.0", "react-native-vision-camera": "4.7.1"
```

App works perfectly in Debug mode, but in Release it crashes after launch. What helped is the @ghashi solution.

fracico-tech, 2026-01-31:
Crash still happening with:
```
"react-native-vision-camera": "4.7.3",
"react-native-worklets-core": "1.6.2",
"react-native": "0.81.5",
```
No reanimated package installed. Works in debug, not in release.

Logcat output:
```
2026-01-31 07:28:50.726 23902-24291 FrameProce...inRegistry com.parkingless.production           I  Looking up Frame Processor Plugin "scanOCR"...
2026-01-31 07:28:50.726 23902-24291 FrameProce...inRegistry com.parkingless.production           I  Frame Processor Plugin "scanOCR" found! Initializing...
2026-01-31 07:28:50.746 23902-23902 CameraSession           com.parkingless.production           I  Camera Lifecycle changed to CREATED!
2026-01-31 07:28:50.747 23902-23902 CameraView              com.parkingless.production           I  Updating CameraSession...
2026-01-31 07:28:50.750 23902-23902 CameraView              com.parkingless.production           I  Updating CameraSession...
2026-01-31 07:28:50.750 23902-23902 CameraView              com.parkingless.production           I  Updating CameraSession...
2026-01-31 07:28:50.750 23902-23902 CameraSession           com.parkingless.production           I  configure { ... }: Waiting for lock...
2026-01-31 07:28:50.750 23902-23902 CameraView              com.parkingless.production           I  A new configure { ... } call arrived, aborting this one...
2026-01-31 07:28:50.751 23902-23902 CameraSession           com.parkingless.production           I  configure { ... }: Waiting for lock...
2026-01-31 07:28:50.751 23902-23902 CameraSession           com.parkingless.production           I  configure { ... }: Updating CameraSession Configuration... Difference(deviceChanged=true, outputsChanged=true, sidePropsChanged=true, isActiveChanged=true, orientationChanged=true, locationChanged=true)
2026-01-31 07:28:50.751 23902-23902 CameraSession           com.parkingless.production           I  Creating new Outputs for Camera #0...
2026-01-31 07:28:50.751 23902-23902 CameraSession           com.parkingless.production           I  Using FPS Range: null
2026-01-31 07:28:50.751 23902-23902 CameraSession           com.parkingless.production           I  Creating Preview output...
2026-01-31 07:28:50.753 23902-23902 CameraSession           com.parkingless.production           I  Creating Video output...
2026-01-31 07:28:50.753 23902-23902 CameraSession           com.parkingless.production           I  Creating new Recorder...
2026-01-31 07:28:50.754 23902-23902 Recorder                com.parkingless.production           D  mRequiredFreeStorageBytes = 50 MB
2026-01-31 07:28:50.755 23902-23902 CameraSession           com.parkingless.production           I  Creating YUV Frame Processor output...
2026-01-31 07:28:50.756 23902-23902 CameraSession           com.parkingless.production           I  Successfully created new Outputs for Camera #0!
2026-01-31 07:28:50.757 23902-23902 CameraSession           com.parkingless.production           I  Binding Camera #0...
2026-01-31 07:28:50.757 23902-23902 CameraSession           com.parkingless.production           I  Binding 3 use-cases...
2026-01-31 07:28:50.759 23902-23902 DeviceQuirks            com.parkingless.production           D  core DeviceQuirks = SurfaceOrderQuirk
2026-01-31 07:28:50.765 23902-23902 CapabilitiesByQuality   com.parkingless.production           D  profiles = ImmutableEncoderProfilesProxy{defaultDurationSeconds=30, recommendedFileFormat=2, audioProfiles=[AudioProfileProxy{codec=3, mediaType=audio/mp4a-latm, bitrate=192000, sampleRate=48000, channels=2, profile=2}], videoProfiles=[VideoProfileProxy{codec=2, mediaType=video/avc, bitrate=20000000, frameRate=30, width=1920, height=1080, profile=-1, bitDepth=8, chromaSubsampling=0, hdrFormat=0}]}
2026-01-31 07:28:50.765 23902-23902 CapabilitiesByQuality   com.parkingless.production           D  profiles = ImmutableEncoderProfilesProxy{defaultDurationSeconds=30, recommendedFileFormat=2, audioProfiles=[AudioProfileProxy{codec=3, mediaType=audio/mp4a-latm, bitrate=192000, sampleRate=48000, channels=2, profile=2}], videoProfiles=[VideoProfileProxy{codec=2, mediaType=video/avc, bitrate=14000000, frameRate=30, width=1280, height=720, profile=-1, bitDepth=8, chromaSubsampling=0, hdrFormat=0}]}
2026-01-31 07:28:50.766 23902-23902 CapabilitiesByQuality   com.parkingless.production           D  profiles = ImmutableEncoderProfilesProxy{defaultDurationSeconds=30, recommendedFileFormat=2, audioProfiles=[AudioProfileProxy{codec=3, mediaType=audio/mp4a-latm, bitrate=192000, sampleRate=48000, channels=2, profile=2}], videoProfiles=[VideoProfileProxy{codec=2, mediaType=video/avc, bitrate=2000000, frameRate=30, width=720, height=480, profile=-1, bitDepth=8, chromaSubsampling=0, hdrFormat=0}]}
2026-01-31 07:28:50.766 23902-23902 QualitySelector         com.parkingless.production           D  supportedQualities = [ConstantQuality{value=6, name=FHD, typicalSizes=[1920x1080]}, ConstantQuality{value=5, name=HD, typicalSizes=[1280x720]}, ConstantQuality{value=4, name=SD, typicalSizes=[720x480, 640x480]}]
2026-01-31 07:28:50.766 23902-23902 VideoCapture            com.parkingless.production           D  Found selectedQualities [ConstantQuality{value=6, name=FHD, typicalSizes=[1920x1080]}, ConstantQuality{value=5, name=HD, typicalSizes=[1280x720]}, ConstantQuality{value=4, name=SD, typicalSizes=[720x480, 640x480]}] by QualitySelector{preferredQualities=[ConstantQuality{value=6, name=FHD, typicalSizes=[1920x1080]}, ConstantQuality{value=5, name=HD, typicalSizes=[12 [... cut ...]
        frameProcessor={frameProcessor}
        device={device}
        isActive={isActive}
      />
      <LicensePlate plate={plate} />
    </>
  ) : (
    <Card onPress={() => Linking.openSettings()}>
      <Card.Title
        title="Camera non disponibile"
        subtitle="Assicurati di aver dato i permessi di accesso alla fotocamera"
      />
      <Card.Content>
        <Text style={styles.settingsTextColor} variant="titleLarge">
          Vai a impostazioni
        </Text>
      </Card.Content>
    </Card>
  );
};
```