margelo/react-native-vision-camera #2824: IllegalArgumentException: getCameraCharacteristics:1278: Unable to retrieve camera characteristics for unknown device 0: No such file or directory (-2)🐛

Sharf8351, 2024-04-30:
### What's happening?

As this issue happened in production. So, I don't have logs for it.

Steps to reproduce:
Open the camera to Scan the QR-Code
Choose allow "only this time"
the camera opened shortly and the app crashed shortly after
if the user wants to scan the code afterwards the camera doesn't open

AN 14 Pixel 7a

### Reproduceable Code

```tsx
const [cameraStatus, setCameraStatus] = useState<CameraStatus>('unknown');
const [cameraProductId, setCameraProductId] = useState<string | undefined>(undefined);

/**
 * Camera
 */
const {hasPermission, requestPermission} = useCameraPermission();
const device = useCameraDevice('back');

useEffect(() => {
    console.log(`Device variant${variant}`);
    (async () => {
        if (hasPermission) {
            console.log('📸 has camera permission');
        } else {
            await requestPermission();
        }
    })();
}, []);

/**
 * Scan barcode and setting up device registration
 * @param event
 */
const codeScanner = useCodeScanner({
    codeTypes: ['data-matrix'],
    onCodeScanned: async codes => {
        if (codes.length > 0 && codes[0].value) {
            if (
                codes[0].value.match('^[A-Z2-7]{10}$') !== null &&
                cameraProductId === undefined
            ) {
                setCameraProductId(codes[0].value);
                await setupDevice(codes[0].value, undefined);
            }
        }
    },
});

<Box style={styles.centerContainer}>
    {cameraStatus === 'active' &&
        (device ? (
            <Camera
                style={StyleSheet.absoluteFill}
                device={device}
                isActive={true}
                codeScanner={codeScanner}
                zoom={1.5}
            />
        ) : (
            <Box />
        ))}

    {cameraStatus === 'unknown' && (
        <Button
            title={t('registerDevice:step.first.button.camera')}
            iconName="camera"
            onPress={() => {
                setCameraStatus('active');
            }}
        />
    )}

    {cameraStatus === 'load' && <CameraActivityIndicator />}
</Box>
```

### Relevant log output

```shell
As this issue happened in production. So, I don't have logs for it.
```

### Camera Device

```json
As this issue happened in production. So, I don't have logs for it.
```

### Device

Google Pixel 7a

### VisionCamera Version

3.5.1

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2024-05-01:
This has been fixed in VisionCamera V4.

closed 2024-05-01 by mrousavy

debdeep-ganguly, 2024-05-30:
@mrousavy Not yet fixed in V4. 

### Version used:
> "react-native-vision-camera": "^4.0.4",

### Sample stacktrace:

 ```
Fatal Exception: com.facebook.react.common.JavascriptException: Error: Exception in HostObject::get for prop 'CameraDevices': java.lang.IllegalArgumentException: getCameraCharacteristics:747: Unable to retrieve camera characteristics for unknown device 3: No such file or directory (-2), js engine: hermes, stack:
/Users/debdeepganguly/Documents/repos/symphony-mobile/node_modules/react-native-vision-camera/src/CameraDevices.ts:4:<global>
/Users/debdeepganguly/Documents/repos/symphony-mobile/node_modules/metro-runtime/src/polyfills/require.js:342:loadModuleImplementation
/Users/debdeepganguly/Documents/repos/symphony-mobile/node_modules/metro-runtime/src/polyfills/require.js:240:guardedLoadModule
/Users/debdeepganguly/Documents/repos/symphony-mobile/node_modules/metro-runtime/src/polyfills/require.js:127:metroRequire
/Users/debdeepganguly/Documents/repos/symphony-mobile/node_modules/react-native-vision-camera/src/Camera.tsx:12:<global>
/Users/debdeepganguly/Documents/repos/symphony-mobile/node_modules/metro-runtime/src/polyfills/require.js:342:loadModuleImplementation
/Users/debdeepganguly/Documents/repos/symphony-mobile/node_modules/metro-runtime/src/polyfills/require.js:240:guardedLoadModule
/Users/debdeepganguly/Documents/repos/symphony-mobile/node_modules/metro-runtime/src/polyfills/require.js:127:metroRequire
/Users/debdeepganguly/Documents/repos/symphony-mobile/node_modules/react-native-vision-camera/src/index.ts:2:<global>
/Users/debdeepganguly/Documents/repos/symphony-mobile/node_modules/metro-runtime/src/polyfills/require.js:342:loadModuleImplementation
/Users/debdeepganguly/Documents/repos/symphony-mobile/node_modules/metro-runtime/src/polyfills/require.js:240:guardedLoadModule
/Users/debdeepganguly/Documents/repos/symphony-mobile/node_modules/metro-runtime/src/polyfills/require.js:127:metroRequire
/Users/debdeepganguly/Documents/repos/symphony-mobile/src/containers/Interview/CameraView.js:4:<global>
/Users/debdeepganguly/Documents/repos/symphony-mobile/node_modules/metro-runtime/src/polyfills/require.js:342:loadModuleImplementation
/Users/debdeepganguly/Documents/repos/symphony-mobile/node_modules/metro-runtime/src/polyfills/require.js:240:guardedLoadModule
/Users/debdeepganguly/Documents/repos/symphony-mobile/node_modules/metro-runtime/src/polyfills/require.js:127:metroRequire
/Users/debdeepganguly/Documents/repos/symphony-mobile/src/containers/Interview/AiInterview.js:11:<global>
/Users/debdeepganguly/Documents/repos/symphony-mobile/node_modules/metro-runtime/src/polyfills/require.js:342:loadModuleImplementation
/Users/debdeepganguly/Documents/repos/symphony-mobile/node_modules/metro-runtime/src/polyfills/require.js:240:guardedLoadModule
/Users/debdeepganguly/Documents/repos/symphony-mobile/node_modules/metro-runtime/src/polyfills/require.js:127:metroRequire
/Users/debdeepganguly/Documents/repos/symphony-mobile/src/navigation/navigators/CandidateNavigators/CommonNavigator.js:36:<global>
/Users/debdeepganguly/Documents/repos/symphony-mobile/node_modules/metro-runtime/src/polyfills/require.js:342:loadModuleImplementation
/Users/debdeepganguly/Documents/repos/symphony-mobile/node_modules/metro-runtime/src/polyfills/require.js:240:guardedLoadModule
/Users/debdeepganguly/Documents/repos/symphony-mobile/node_modules/metro-runtime/src/polyfills/require.js:127:metroRequire
/Users/debdeepganguly/Documents/repos/symphony-mobile/src/navigation/navigators/AppNavigator.js:8:<global>
/Users/debdeepganguly/Documents/repos/symphony-mobile/node_modules/metro-runtime/src/polyfills/require.js:342:loadModuleImplementation
/Users/debdeepganguly/Documents/repos/symphony-mobile/node_modules/metro-runtime/src/polyfills/require.js:240:guardedLoadModule
/Users/debdeepganguly/Documents/repos/symphony-mobile/node_modules/metro-runtime/src/polyfills/require.js:127:metroRequire
/Users/debdeepganguly/Documents/repos/symphony-mobile/App.js:3:<global>
/Users/debdeepganguly/Documents/repos/symphony-mobile/node_modules/metro-runtime/src/polyfills/require.js:342:loadModuleImplementation
/Users/debdeepganguly/Documents/repos/symphony-mobile/node_modules/metro-runtime/src/polyfills/require.js:240:guardedLoadModule
/Users/debdeepganguly/Documents/repos/symphony-mobile/node_modules/metro-runtime/src/polyfills/require.js:127:metroRequire
/Users/debdeepganguly/Documents/repos/symphony-mobile/index.js:9:<global>
/Users/debdeepganguly/Documents/repos/symphony-mobile/node_modules/metro-runtime/src/polyfills/require.js:342:loadModuleImplementation
/Users/debdeepganguly/Documents/repos/symphony-mobile/node_modules/metro-runtime/src/polyfills/require.js:232:guardedLoadModule
/Users/debdeepganguly/Documents/repos/symphony-mobile/node_modules/metro-runtime/src/polyfills/require.js:127:metroRequire
null:null:null

       at com.facebook.react.modules.core.ExceptionsManagerModule.reportException(ExceptionsManagerModule.java:65)
       at java.lang.reflect.Method.invoke(Method.java)
       at com.facebook.react.bridge.JavaMethodWrapper.invoke(JavaMethodWrapper.java:372)
       at com.facebook.react.bridge.JavaModuleWrapper.invoke(JavaModuleWrapper.java:188)
       at com.facebook.jni.NativeRunnable.run(NativeRunnable.java)
       at android.os.Handler.handleCallback(Handler.java:938)
       at android.os.Handler.dispatchMessage(Handler.java:99)
       at com.facebook.react.bridge.queue.MessageQueueThreadHandler.dispatchMessage(MessageQueueThreadHandler.java:27)
       at android.os.Looper.loopOnce(Looper.java:201)
       at android.os.Looper.loop(Looper.java:288)
       at com.facebook.react.bridge.queue.MessageQueueThreadImpl$4.run(MessageQueueThreadImpl.java:228)
       at java.lang.Thread.run(Thread.java:1012)
```

leileiChenGentleman, 2024-11-05:
getCameraCharacteristics:862: Unable to retrieve camera characteristics for unknown device 2: No such file or directory (-2)

androidx.camera.core.CameraX.void lambda$initAndRetryRecursively$2(android.content.Context,java.util.concurrent.Executor,androidx.concurrent.futures.Callback

androidx.camera.core.InitializationException:java.lang.IllegalArgumentException: getCameraCharacteristics:862: Unable to retrieve camera characteristics for unknown device 2: No such file or directory (-2)
2
androidx.camera.core.CameraX.void lambda$initAndRetryRecursively$2(android.content.Context,java.util.concurrent.Executor,androidx.concurrent.futures.CallbackToFutureAdapter$Completer,long)(CameraX.java:369)
3
......
4
Caused by:
5
android.os.ServiceSpecificException:getCameraCharacteristics:862: Unable to retrieve camera characteristics for unknown device 2: No such file or directory (-2)
6
android.os.Parcel.createExceptionOrNull(Parcel.java:2457)
7
android.os.Parcel.createException(Parcel.java:2427)
8
android.os.Parcel.readException(Parcel.java:2410)
9
android.os.Parcel.readException(Parcel.java:2352)
10
android.hardware.ICameraService$Stub$Proxy.getCameraCharacteristics(ICameraService.java:971)
11
android.hardware.camera2.CameraManager.getCameraCharacteristics(CameraManager.java:489)
12
androidx.camera.camera2.internal.compat.CameraManagerCompatApi29Impl.android.hardware.camera2.CameraCharacteristics getCameraCharacteristics(java.lang.String)(CameraManagerCompatApi29Impl.java:57)
13
androidx.camera.camera2.internal.compat.CameraManagerCompat.androidx.camera.camera2.internal.compat.CameraCharacteristicsCompat getCameraCharacteristicsCompat(java.lang.String)(CameraManagerCompat.java:189)
14
androidx.camera.camera2.internal.Camera2CameraFactory.boolean isBackwardCompatible(java.lang.String)(Camera2CameraFactory.java:171)
15
androidx.camera.camera2.internal.Camera2CameraFactory.java.util.List getBackwardCompatibleCameraIds(java.util.List)(Camera2CameraFactory.java:149)
16
androidx.camera.camera2.internal.Camera2CameraFactory.void <init>(android.content.Context,androidx.camera.core.impl.CameraThreadConfig,androidx.camera.core.CameraSelector,long)(Camera2CameraFactory.java:80)
17
androidx.camera.camera2.Camera2Config$$InternalSyntheticLambda$1$b737bf8082c5f7ea267385a154eefca2640a55b974e0d2599d3c3cd43ba9dca3$0.androidx.camera.core.impl.CameraFactory newInstance(android.content.Context,androidx.camera.core.impl.CameraThreadConfig,androidx.camera.core.CameraSelector,long)(Camera2Config.java:0)
18
androidx.camera.core.CameraX.void lambda$initAndRetryRecursively$2(android.content.Context,java.util.concurrent.Executor,androidx.concurrent.futures.CallbackToFutureAdapter$Completer,long)(CameraX.java:306)
19
androidx.camera.core.CameraX.void $r8$lambda$dnuYhxGsACX_lH-9SVlUVTdtTJg(androidx.camera.core.CameraX,android.content.Context,java.util.concurrent.Executor,androidx.concurrent.futures.CallbackToFutureAdapter$Completer,long)(CameraX.java:0)
20
androidx.camera.core.CameraX$$InternalSyntheticLambda$4$845f10987ef598ac292769067d92d63fd4aa661f29de3bb3ed47d03f5eaa76ed$0.void run()(CameraX.java:0)
21
java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
22
java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
23
java.lang.Thread.run(Thread.java:933)

react-native-vision-camera version 4.0.0