margelo/react-native-vision-camera #2826: 🐛  Android ImageAnalysis setBackpressureStrategy STRATEGY_BLOCK_PRODUCER causing lag with OCR libraries

kevinranks, 2024-04-30:
### What's happening?

With the back pressure strategy set to `STRATEGY_BLOCK_PRODUCER` instead of `STRATEGY_KEEP_ONLY_LATEST` I'm experiencing camera lag while processing OCR unless I use `runAsync()`. I've confirmed that changing the strategy resolves the issue and `runAsync()` isn't required. 

I previously opened issues in these two OCR repos prior to learning with Ismael on Discord that the issue was indeed the back pressure strategy:

- https://github.com/gev2002/react-native-vision-camera-v3-text-recognition
- https://github.com/ismaelsousa/vision-camera-ocr/issues/9

### Reproduceable Code

```tsx
const frameProcessor = useFrameProcessor(frame => {
    'worklet';

    runAtTargetFps(1, () => {
      'worklet';
      const data = scanOCR(frame);
      //console.log('🚀 ~ frameProcessor ~ data:', JSON.stringify(data, null, 2));
    });

    // setOcrJS({...data});
  }, []);
```

### Relevant log output

```shell
04-30 10:25:53.855  2370  4461 I Camera3-Status: markComponent: Component 3 is now idle - streamId = 0, format = 0x22(IMPL_DEF), width = 1440, height = 1080, option = 0x0, type = OUTPUT
04-30 10:25:53.855  2370  4461 I Camera3-IOStreamBase: FpsStats: stream 0, requested=37, received=37, valid=37, invalid=0
04-30 10:25:53.877 18968 19352 W ImageReader_JNI: Unable to acquire a buffer item, very likely client tried to acquire more than maxImages buffers
04-30 10:25:53.878  1898  3597 E GME     : [ERROR][ChiNode] camxchinodegme.cpp:12842 CheckRemosaicCropVideo() MultiCameraSATNoGPUQuad0_com.qti.node.gme3_cam2 Zoom State vendor tag not present
04-30 10:25:53.878 18968 19352 E ImageAnalysisAnalyzer: Failed to acquire image.
04-30 10:25:53.878 18968 19352 E ImageAnalysisAnalyzer: java.lang.IllegalStateException: maxImages (6) has already been acquired, call #close before acquiring more.
04-30 10:25:53.878 18968 19352 E ImageAnalysisAnalyzer: 	at android.media.ImageReader.acquireNextImage(ImageReader.java:661)
04-30 10:25:53.878 18968 19352 E ImageAnalysisAnalyzer: 	at androidx.camera.core.AndroidImageReaderProxy.acquireNextImage(AndroidImageReaderProxy.java:88)
04-30 10:25:53.878 18968 19352 E ImageAnalysisAnalyzer: 	at androidx.camera.core.SafeCloseImageReaderProxy.acquireNextImage(SafeCloseImageReaderProxy.java:88)
04-30 10:25:53.878 18968 19352 E ImageAnalysisAnalyzer: 	at androidx.camera.core.ImageAnalysisBlockingAnalyzer.acquireImage(ImageAnalysisBlockingAnalyzer.java:41)
04-30 10:25:53.878 18968 19352 E ImageAnalysisAnalyzer: 	at androidx.camera.core.ImageAnalysisAbstractAnalyzer.onImageAvailable(ImageAnalysisAbstractAnalyzer.java:128)
04-30 10:25:53.878 18968 19352 E ImageAnalysisAnalyzer: 	at androidx.camera.core.SafeCloseImageReaderProxy.lambda$setOnImageAvailableListener$1$androidx-camera-core-SafeCloseImageReaderProxy(SafeCloseImageReaderProxy.java:213)
04-30 10:25:53.878 18968 19352 E ImageAnalysisAnalyzer: 	at androidx.camera.core.SafeCloseImageReaderProxy$$ExternalSyntheticLambda1.onImageAvailable(Unknown Source:4)
04-30 10:25:53.878 18968 19352 E ImageAnalysisAnalyzer: 	at androidx.camera.core.AndroidImageReaderProxy.lambda$setOnImageAvailableListener$0$androidx-camera-core-AndroidImageReaderProxy(AndroidImageReaderProxy.java:169)
04-30 10:25:53.878 18968 19352 E ImageAnalysisAnalyzer: 	at androidx.camera.core.AndroidImageReaderProxy$$ExternalSyntheticLambda0.run(Unknown Source:4)
04-30 10:25:53.878 18968 19352 E ImageAnalysisAnalyzer: 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1145)
04-30 10:25:53.878 18968 19352 E ImageAnalysisAnalyzer: 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:644)
04-30 10:25:53.878 18968 19352 E ImageAnalysisAnalyzer: 	at java.lang.Thread.run(Thread.java:1012)
04-30 10:25:53.912  1898  3597 E GME     : [ERROR][ChiNode] camxchinodegme.cpp:12842 CheckRemosaicCropVideo() MultiCameraSATNoGPUQuad0_com.qti.node.gme3_cam2 Zoom State vendor tag not present
04-30 10:25:53.912  2370  4461 I Camera3-Status: markComponent: Component 4 is now idle - streamId = 1, format = 0x23(YUV_420_888), width = 3840, height = 2160, option = 0x0, type = OUTPUT
04-30 10:25:53.912  2370  4461 I Camera3-IOStreamBase: FpsStats: stream 1, requested=38, received=38, valid=38, invalid=0
04-30 10:25:53.912 18968 19352 W ImageReader_JNI: Unable to acquire a buffer item, very likely client tried to acquire more than maxImages buffers
04-30 10:25:53.914 18968 19352 E ImageAnalysisAnalyzer: Failed to acquire image.
04-30 10:25:53.914 18968 19352 E ImageAnalysisAnalyzer: java.lang.IllegalStateException: maxImages (6) has already been acquired, call #close before acquiring more.
04-30 10:25:53.914 18968 19352 E ImageAnalysisAnalyzer: 	at android.media.ImageReader.acquireNextImage(ImageReader.java:661)
04-30 10:25:53.914 18968 19352 E ImageAnalysisAnalyzer: 	at androidx.camera.core.AndroidImageReaderProxy.acquireNextImage(AndroidImageReaderProxy.java:88)
04-30 10:25:53.914 18968 19352 E ImageAnalysisAnalyzer: 	at androidx.camera.core.SafeCloseImageReaderProxy.acquireNextImage(SafeCloseImageReaderProxy.java:88)
04-30 10:25:53.914 18968 19352 E ImageAnalysisAnalyzer: 	at androidx.camera.core.ImageAnalysisBlockingAnalyzer.acquireImage(ImageAnalysisBlockingAnalyzer.java:41)
04-30 10:25:53.914 18968 19352 E ImageAnalysisAnalyzer: 	at androidx.camera.core.ImageAnalysisAbstractAnalyzer.onImageAvailable(ImageAnalysisAbstractAnalyzer.java:128)
04-30 10:25:53.914 18968 19352 E ImageAnalysisAnalyzer: 	at androidx.camera.core.SafeCloseImageReaderProxy.lambda$setOnImageAvailableListener$1$androidx-camera-core-SafeCloseImageReaderProxy(SafeCloseImageReaderProxy.java:213)
04-30 10:25:53.914 18968 19352 E ImageAnalysisAnalyzer: 	at androidx.camera.core.SafeCloseImageReaderProxy$$ExternalSyntheticLambda1.onImageAvailable(Unknown Source:4)
04-30 10:25:53.914 18968 19352 E ImageAnalysisAnalyzer: 	at androidx.camera.core.AndroidImageReaderProxy.lambda$setOnImageAvailableListener$0$androidx-camera-core-AndroidImageReaderProxy(AndroidImageReaderProxy.java:169)
04-30 10:25:53.914 18968 19352 E ImageAnalysisAnalyzer: 	at androidx.camera.core.AndroidImageReaderProxy$$ExternalSyntheticLambda0.run(Unknown Source:4)
04-30 10:25:53.914 18968 19352 E ImageAnalysisAnalyzer: 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1145)
04-30 10:25:53.914 18968 19352 E ImageAnalysisAnalyzer: 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:644)
04-30 10:25:53.914 18968 19352 E ImageAnalysisAnalyzer: 	at java.lang.Thread.run(Thread.java:1012)
04-30 10:25:53.929  1898  3600 E ChiX    : [ERROR][SBM    ] chxstreambuffermanager.cpp:1965 GetStreamBuffer() Can't get buffer for stream 0xb40000702978a918 : (0, 6, 0, 0, 8) : (avail, pending, used, psStatus, totalUnique), requestId 38, bufferMgrName MultiCameraSATNoGPUQuad0_IPE10_cam2_OutputPortId8_Display
04-30 10:25:53.929  1898  3600 E CamX    : [ERROR][MEMMGR ] camximagebuffer.cpp:1752 BindImportedBuffer() [MultiCameraSATNoGPUQuad0_IPE10_cam2_OutputPortId8_Display] Failed to get buffer from stream buffer manager
04-30 10:25:53.929  1898  3600 E CamX    : [ERROR][MEMMGR ] camximagebuffer.cpp:1823 BindImportedBuffer() [MultiCameraSATNoGPUQuad0_IPE10_cam2_OutputPortId8_Display][0xb400007179f5b590] Invalid Gralloc buffer [0x0] to import
04-30 10:25:53.930  1898  2342 E CamX    : [ERROR][CORE   ] camxnodelegacy.cpp:8211 CSLFenceCallback() Node::MultiCameraSATNoGPUQuad0_IPE10_cam2 : Type 65538 portId=8 Fence 45 signaled with failure 2 in node fence handler FOR 38
04-30 10:25:53.932  1898  3600 E CamX    : [ERROR][CORE   ] camxpipeline.cpp:5354 SinkPortFenceErrorSignaled() MultiCameraSATNoGPUQuad0_0_cam_2 Reporting stream done for stream=0, sequenceId=37 Buffer type=0 handle=0x0
04-30 10:25:53.932  1898  3600 E CamX    : [ERROR][CORE   ] camxsession.cpp:8659 InjectResult() Session 0xb40000720a24ab90 [MultiCameraSATNoGPUQuad0] Reporting a buffer error to the framework for streamIndex 0 SeqId: 37 <-> ReqId: 38
[... 96 lines cut ...]
  "hasTorch": true,
  "position": "back",
  "id": "0"
}
```

### Device

Samsung Galaxy 24 Ultra

### VisionCamera Version

4.0.1

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2024-05-01:
> I'm experiencing camera lag

What does that mean? What is camera lag? Is the FPS slow? Is it completely freezing after 6 images? Is it stuttering behind? Is it sometimes smooth sometimes stuttering? Post a video so I can see what you are talking about.

`STRATEGY_BLOCK_PRODUCER` is used intentionally, this is the behaviour a Frame Processor should have. This is the same strategy that is also used on iOS.
It's what I'm trying to explain in the Frame Processor documentation page - frames stream in as often as they can (with `fps={30}`, 30 times a second, or every 33ms), and if your Frame Processor is not fast enough to finish executing before a new frame arrives, it might drop a new Frame.

`STRATEGY_BLOCK_PRODUCER` basically means that we push Frames into a queue, and if the queue is full we drop all Frames until a slot is free in the queue.

So if your queue is too slow, you need to do one of two things;
1. Optimize your Frame Processor to make sure it finishes within 33ms (make `scanOCR` faster)
2. Use `runAsync` to run something that takes longer asynchronously. If your FP is called 30 times a second and you use `runAsync` which takes longer than 33ms to execute, it will skip running that runAsync code until the Async Context is free again, **which is exactly the same behaviour as using `STRATEGY_KEEP_ONLY_LATEST`**.
That's what the `enableFpsGraph={true}` should help you with. If you easily hit 30 or 60 there, you're good.

With the introduction of `runAsync` we have the power of using `STRATEGY_BLOCK_PRODUCER` by default, and `STRATEGY_KEEP_ONLY_LATEST` being opt in (aka using `runAsync`). 

What `STRATEGY_KEEP_ONLY_LATEST` does is essentially just not calling the `ImageAnalysis` function if it is still busy. There is no magic behind it - it's simply dropping Frames and allowing the Preview to run freely, we can do the same thing with `runAsync`.

---

So I'm not sure if I understood your issue correctly, so curious to hear your thoughts on this after the above explanation as to why I think `STRATEGY_KEEP_ONLY_LATEST` is a bad idea. :)

mrousavy, 2024-05-01:
As per the documentation of [`STRATEGY_KEEP_ONLY_LATEST`](https://developer.android.com/reference/androidx/camera/core/ImageAnalysis#STRATEGY_KEEP_ONLY_LATEST):
[image]

If only one image is delivered at a time, `runAsync` would effectively achieve nothing anymore, since no Frames can come in in parallel.

ismaelsousa, 2024-05-01:
@mrousavy here are some videos and also two ways of handling this https://github.com/ismaelsousa/vision-camera-ocr/issues/9

mrousavy, 2024-05-02:
Replied over in that issue. This is expected behaviour.

closed 2024-05-02 by mrousavy