margelo/react-native-vision-camera #2827: [V4] [iOS] Error: react-native-worklets-core is not installed! 🐛

Tymofiev, 2024-05-01:
### What's happening?

When app starts on iOS without `react-native-worklets-core` installed `Error: react-native-worklets-core is not installed!, js engine: hermes` error is thrown. Following https://www.react-native-vision-camera.com/docs/guides/frame-processors#disabling-frame-processors guide to explicitly disabke frame processors makes no change.

React Native version: `0.71.15`.

PS: This works fine on example app after uninstalling `react-native-worklets-core` dependency and removing `useFrameProcessor` from camera page.

### Reproduceable Code

```tsx
N/A
```

### Relevant log output

```shell
BUNDLE  ./index.js 
 ERROR  Error: react-native-worklets-core is not installed!, js engine: hermes
 LOG  Running "AppName" with {"rootTag":61,"initialProps":{}}
 ERROR  Invariant Violation: "AppName" has not been registered. This can happen if:
* Metro (the local dev server) is run from the wrong folder. Check if Metro is running, stop it and restart it in the current project.
* A module failed to load due to an error and `AppRegistry.registerComponent` wasn't called., js engine: hermes
```

### Camera Device

```json
N/A
```

### Device

iPhone 15 (iOS 17.4) Simulator

### VisionCamera Version

4.0.1

### Can you reproduce this issue in the VisionCamera Example app?

No, I cannot reproduce the issue in the Example app

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2024-05-01:
1. Why did you not post your react-native-worklets-core version?
2. You need react-native-worklets-core 1.x.x for V4, and 0.x.x for V3 (you are on V4, so react-native-worklets-core 1.x.x)
3. Might also be a good idea to upgrade your RN version to 0.72 or 0.73, but 0.71 should be good.

closed 2024-05-01 by mrousavy

Tymofiev, 2024-05-01:
@mrousavy I did not add `react-native-worklets-core` version because I am not using any frame processors and don't have it installed, is that dependency required anyway?

The RN upgrade is coming soon, but I thought it might be helpful information if it's not working with older versions for some reason...

mrousavy, 2024-05-01:
Ah okay, well I didn't know if you are using FPs or not because you chose to ignore the required "reproduceable code" section: 
[image]

Tymofiev, 2024-05-01:
@mrousavy I am so sorry for that! I just wasn't sure what to put in there since it's pretty basic rendering of Camera component with just a few props. Updated the description.

lsps9150414, 2024-05-16:
Hey I am running into the same situation seeing the `Error: react-native-worklets-core is not installed!` following the Expo installation steps on https://react-native-vision-camera.com/docs/guides

I'm simply trying to render the camera without using the processor so I don't think I need to install `react-native-worklets-core` neither does the installation guide mentioned it.

I'm confused, so is `react-native-worklets-core` required anyway?

Tymofiev, 2024-05-16:
@Sharf8351 I was just about to test in on the new version of RN... but if you are saying it still requires `react-native-worklets-core` then I don't know how to fix it right now other than adding that package...

mrousavy, 2024-05-16:
It does not require react-native-worklets-core - try latest VisionCamera. That builds perfectly fine without Worklets.

Tymofiev, 2024-05-16:
Just tested and seems to work without `react-native-worklets-core` on `"react-native": "0.73.6"` with `"react-native-vision-camera": "4.0.4"`.