margelo/react-native-vision-camera #2834: Android build fails - Task configureCMakeDebug[arm64-v8a] - due linkage with react-native-reanimated 🔧

mirote3, 2024-05-02:
### How were you trying to build the app?

npm run bundle && react-native run-android --variant=devDebug

### Full build logs

```tsx
> Task :react-native-vision-camera:configureCMakeDebug[arm64-v8a] FAILED
C/C++: debug|arm64-v8a :CMake Warning at /Users/rotemmizrahi/Library/Android/sdk/ndk/24.0.8215888/build/cmake/android-legacy.toolchain.cmake:415 (message):
C/C++: debug|arm64-v8a :  An old version of CMake is being used that cannot automatically detect
C/C++: debug|arm64-v8a :  compiler attributes.  Compiler identification is being bypassed.  Some
C/C++: debug|arm64-v8a :  values may be wrong or missing.  Update to CMake 3.19 or newer to use
C/C++: debug|arm64-v8a :  CMake's built-in compiler identification.
C/C++: debug|arm64-v8a :Call Stack (most recent call first):
C/C++: debug|arm64-v8a :  /Users/rotemmizrahi/Library/Android/sdk/ndk/24.0.8215888/build/cmake/android.toolchain.cmake:54 (include)
C/C++: debug|arm64-v8a :  /Users/rotemmizrahi/Library/Android/sdk/cmake/3.18.1/share/cmake-3.18/Modules/CMakeDetermineSystem.cmake:93 (include)
C/C++: debug|arm64-v8a :  CMakeLists.txt:1 (project)
C/C++: debug|arm64-v8a :CMake Warning at /Users/rotemmizrahi/Library/Android/sdk/ndk/24.0.8215888/build/cmake/android-legacy.toolchain.cmake:415 (message):
C/C++: debug|arm64-v8a :  An old version of CMake is being used that cannot automatically detect
C/C++: debug|arm64-v8a :  compiler attributes.  Compiler identification is being bypassed.  Some
C/C++: debug|arm64-v8a :  values may be wrong or missing.  Update to CMake 3.19 or newer to use
C/C++: debug|arm64-v8a :  CMake's built-in compiler identification.
C/C++: debug|arm64-v8a :Call Stack (most recent call first):
C/C++: debug|arm64-v8a :  /Users/rotemmizrahi/Library/Android/sdk/ndk/24.0.8215888/build/cmake/android.toolchain.cmake:54 (include)
C/C++: debug|arm64-v8a :  /Users/rotemmizrahi/Desktop/Development/IDF-Zuzu/ZUZU-Front-Native/node_modules/react-native-vision-camera/android/.cxx/Debug/e54361o3/arm64-v8a/CMakeFiles/3.18.1-g262b901/CMakeSystem.cmake:6 (include)
C/C++: debug|arm64-v8a :  /Users/rotemmizrahi/Desktop/Development/IDF-Zuzu/ZUZU-Front-Native/node_modules/react-native-vision-camera/android/.cxx/Debug/e54361o3/arm64-v8a/CMakeFiles/CMakeTmp/CMakeLists.txt:2 (project)
C/C++: debug|arm64-v8a :CMake Warning at /Users/rotemmizrahi/Library/Android/sdk/ndk/24.0.8215888/build/cmake/android-legacy.toolchain.cmake:415 (message):
C/C++: debug|arm64-v8a :  An old version of CMake is being used that cannot automatically detect
C/C++: debug|arm64-v8a :  compiler attributes.  Compiler identification is being bypassed.  Some
C/C++: debug|arm64-v8a :  values may be wrong or missing.  Update to CMake 3.19 or newer to use
C/C++: debug|arm64-v8a :  CMake's built-in compiler identification.
C/C++: debug|arm64-v8a :Call Stack (most recent call first):
C/C++: debug|arm64-v8a :  /Users/rotemmizrahi/Library/Android/sdk/ndk/24.0.8215888/build/cmake/android.toolchain.cmake:54 (include)
C/C++: debug|arm64-v8a :  /Users/rotemmizrahi/Desktop/Development/IDF-Zuzu/ZUZU-Front-Native/node_modules/react-native-vision-camera/android/.cxx/Debug/e54361o3/arm64-v8a/CMakeFiles/3.18.1-g262b901/CMakeSystem.cmake:6 (include)
C/C++: debug|arm64-v8a :  /Users/rotemmizrahi/Desktop/Development/IDF-Zuzu/ZUZU-Front-Native/node_modules/react-native-vision-camera/android/.cxx/Debug/e54361o3/arm64-v8a/CMakeFiles/CMakeTmp/CMakeLists.txt:2 (project)
C/C++: debug|arm64-v8a :CMake Error at CMakeLists.txt:208 (find_package):
C/C++: debug|arm64-v8a :  Could not find a package configuration file provided by
C/C++: debug|arm64-v8a :  "react-native-reanimated" with any of the following names:
C/C++: debug|arm64-v8a :    react-native-reanimatedConfig.cmake
C/C++: debug|arm64-v8a :    react-native-reanimated-config.cmake
C/C++: debug|arm64-v8a :  Add the installation prefix of "react-native-reanimated" to
C/C++: debug|arm64-v8a :  CMAKE_PREFIX_PATH or set "react-native-reanimated_DIR" to a directory
C/C++: debug|arm64-v8a :  containing one of the above files.  If "react-native-reanimated" provides a
C/C++: debug|arm64-v8a :  separate development package or SDK, be sure it has been installed.
w: Detected multiple Kotlin daemon sessions at build/kotlin/sessions

Deprecated Gradle features were used in this build, making it incompatible with Gradle 8.0.

You can use '--warning-mode all' to show the individual deprecation warnings and determine if they come from your own scripts or plugins.

See https://docs.gradle.org/7.3.3/userguide/command_line_interface.html#sec:command_line_warnings
641 actionable tasks: 635 executed, 6 up-to-date
Note: Some input files use or override a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
Note: /Users/rotemmizrahi/Desktop/Development/IDF-Zuzu/ZUZU-Front-Native/node_modules/lottie-react-native/src/android/src/main/java/com/airbnb/android/react/lottie/LottieAnimationViewManager.java uses or overrides a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
Note: /Users/rotemmizrahi/Desktop/Development/IDF-Zuzu/ZUZU-Front-Native/node_modules/lottie-react-native/src/android/src/main/java/com/airbnb/android/react/lottie/LottieAnimationViewManager.java uses unchecked or unsafe operations.
Note: Recompile with -Xlint:unchecked for details.
Note: /Users/rotemmizrahi/Desktop/Development/IDF-Zuzu/ZUZU-Front-Native/node_modules/mixpanel-react-native/android/src/main/java/com/mixpanel/reactnative/MixpanelReactNativeModule.java uses unchecked or unsafe operations.
Note: Recompile with -Xlint:unchecked for details.
Note: /Users/rotemmizrahi/Desktop/Development/IDF-Zuzu/ZUZU-Front-Native/node_modules/@react-native-async-storage/async-storage/android/src/main/java/com/reactnativecommunity/asyncstorage/AsyncStorageModule.java uses or overrides a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
Note: /Users/rotemmizrahi/Desktop/Development/IDF-Zuzu/ZUZU-Front-Native/node_modules/@react-native-async-storage/async-storage/android/src/javaPackage/java/com/reactnativecommunity/asyncstorage/AsyncStoragePackage.java uses unchecked or unsafe operations.
Note: Recompile with -Xlint:unchecked for details.
Note: Some input files use or override a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
Note: Some input files use or override a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
Note: /Users/rotemmizrahi/Desktop/Development/IDF-Zuzu/ZUZU-Front-Native/node_modules/react-native-camera/android/src/main/java/com/google/android/cameraview/Camera2.java uses unchecked or unsafe operations.
Note: Recompile with -Xlint:unchecked for details.
[... 329 lines cut ...]
```

### VisionCamera Version

2.16.7

### Target platforms

Android

### Operating system

MacOS

### Can you build the VisionCamera Example app?

Yes, I can successfully build the Example app here

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2024-05-02:
> VisionCamera Version
>
> 2.16.7

There's no public support for VisionCamera V2 anymore. We are already at V4.

If you need to stay on V2 for any reasons, I can maybe help you get that fixed through my consultancy services.

closed 2024-05-02 by mrousavy

mirote3, 2024-05-02:
Thanks @mrousavy 

> There's no public support for VisionCamera V2 anymore. We are already at V4.

Is it supports react-native@0.68 ? or I need to upgrade it also?

mrousavy, 2024-05-02:
No, you need react-native 0.71