margelo/react-native-vision-camera #2835: Black screen with useSkiaProcessor 🐛

DeanGracey, 2024-05-02:
### What's happening?

Thanks for a great library! I've been using it nicely until trying to use frame processors and Skia drawing. Everything works well when I use useFrameProcessor, but when I use useSkiaFrameProcessor I just get a black screen. I also use recording in my app and when I record, I can see the video is recorded properly, so it's just the display that's black.

Perhaps I'm misunderstanding how the display is supposed to work, I would like the usual camera to be displayed and I would like to draw lines over it. 

### Reproduceable Code

```tsx
const plugin = useTensorflowModel(
    require("../../assets/pose-detection-fast.tflite")
  );

  const paint = Skia.Paint();
  paint.setStrokeWidth(5);
  paint.setColor(Skia.Color("white"));

  const frameProcessor = useSkiaFrameProcessor(
    (frame) => {
      "worklet";
      if (!plugin.model) return;
      const resized = resize(frame, 192, 192);
      const outputs = plugin.model.runSync([resized]);
      const output = outputs[0]; // one output tensor

      const frameWidth = frame.width;
      const frameHeight = frame.height;

      // draw each line on the skeleton
      for (let i = 0; i < lines.length; i += 2) {
        const from = lines[i];
        const to = lines[i + 1];

        const confidence = output[from * 3 + 2];
        if (confidence > 0.5) {
          // frame.drawLine(
          //   (output[from * 3 + 1] as number) * frameWidth,
          //   (output[from * 3] as number) * frameHeight,
          //   (output[to * 3 + 1] as number) * frameWidth,
          //   (output[to * 3] as number) * frameHeight,
          //   paint
          // );
        }
      }
    },
    [plugin, plugin.model, paint]
  );

  let device: CameraDevice | undefined = undefined;
  device = useCameraDevice("front");

<VisionCamera
            style={StyleSheet.absoluteFill}
            device={device}
            isActive={true}
            video={true}
            ref={cameraRef}
            frameProcessor={frameProcessor}
          />
```

### Relevant log output

```shell
LOG  Loading react-native-worklets-core...
 LOG  Worklets loaded successfully
 LOG  Installing bindings...
 LOG  Successfully installed!
 LOG  Loading Tensorflow Lite Model 15
 LOG  Resolved Model path: http://localhost:8081/assets/?unstable_path=.%2Fassets/pose-detection-fast.tflite?platform=android&hash=361a28992b649751d6d66cd8cfb1e2a6
[... 57 lines cut ...]
  "hasTorch": false,
  "position": "front",
  "id": "1"
}
```

### Device

Xiaomi 12T

### VisionCamera Version

4.0.1

### Can you reproduce this issue in the VisionCamera Example app?

No, I cannot reproduce the issue in the Example app

### Additional information

- [X] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

DeanGracey, 2024-05-02:
I've just discovered that with useSkiaFrameProcessor you have to call frame.render() 😅 

Thanks!

closed 2024-05-02 by DeanGracey

minhaz19, 2024-08-18:
@DeanGracey but the front camera is not working as expected.