margelo/react-native-vision-camera #2837: IOS 12 does not work with version 4.0.0? (according to the guide, it should)

zpinto123, 2024-05-02:
### Question

Hi,

According to the guides, the recent 4.0.0 version requires IOS 12, but react-native-skia does not support IOS 12. I have to up my IOS version to 13.

Is the guide incorrect, or am I missing something?

### What I tried

My current react-native-skia version is 1.2.3

### VisionCamera Version

4.0.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22💭+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2024-05-02:
This is not related to VisionCamera. VisionCamera works on iOS 12+, if Skia requires 13+ then upgrade to 13.

closed 2024-05-02 by mrousavy

zpinto123, 2024-05-02:
> This is not related to VisionCamera. VisionCamera works on iOS 12+, if Skia requires 13+ then upgrade to 13.

But react-native-skia is required by vision camera, right? That makes it impossible to use IOS 12 if react-native-skia requires IOS 13.