margelo/react-native-vision-camera #2838: [v4][Android] 🐛 - Camera frame processor FPS and torch do not behave as expected.

wuguishifu, 2024-05-02:
### What's happening?

I am using a frame processor on a Pixel 6a/Note 9. I have simplified the frame processor to just not actually perform any calculations. The camera normally runs at 30 FPS according to the graph, but when I cover the camera (with a cloth, my finger, putting it on a table, etc.) the FPS drops to 15. I have tried to disable as much auto stuff as possible on the camera.

Another issue is that the torch does not turn on, even when specified "on" in the camera's props. This happens regardless of the other props. If I edit the file and hot update my app while it's running (in dev mode) it turns on the flashlight. However, if I close and reopen the app, the flashlight once again does not turn on.

These systems worked in v3.9.2 on Android and are currently working in v4 on iOS (iPhone 15pm, iOS 17.x and iPhone 11, iOS 17.x), so I believe these issues are exclusive to Android and v4.

### Reproduceable Code

```tsx
const device = useCameraDevice('back');

const frameProcessor = useFrameProcessor(() => {
  'worklet'
}, []);

const format = useMemo(() => {
  return getCameraFormat(device, [{
    fps: 30
  }]);
}, [device]);

return (
  <Camera
    isActive
    style={StyleSheet.absoluteFill}
    device={device}
    torch='on'
    frameProcessor={frameProcessor}
    enableFpsGraph
    pixelFormat='yuv'
    fps={30}
    format={format}
  />
);
```

### Relevant log output

```shell
N/A
```

### Camera Device

```json
{
  "formats": [],
  "sensorOrientation": "landscape-left",
  "hardwareLevel": "full",
  "maxZoom": 8,
  "minZoom": 1,
  "maxExposure": 20,
  "supportsLowLightBoost": false,
  "neutralZoom": 1,
  "physicalDevices": [
    "wide-angle-camera"
  ],
  "supportsFocus": true,
  "supportsRawCapture": false,
  "isMultiCam": false,
  "minFocusDistance": 10,
  "minExposure": -20,
  "name": "0 (BACK) androidx.camera.camera2",
  "hasFlash": true,
  "hasTorch": true,
  "position": "back",
  "id": "0"
}
```

### Device

Pixel 6A (Android 14), Note 9 (Android 10)

### VisionCamera Version

4.0.1

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [X] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2024-05-03:
> The camera normally runs at 30 FPS according to the graph, but when I cover the camera (with a cloth, my finger, putting it on a table, etc.) the FPS drops to 15

This is intended behaviour on Android, devices automatically throttle their FPS to allow more light to reach the sensor.

Also please leave the issue title as prefered in the template.

mrousavy, 2024-05-03:
> Relevant log output
> 
> N/A

This is literally the only thing I would've needed to investigate the torch issue.

closed 2024-05-03 by mrousavy

mrousavy, 2024-05-03:
Yes you can reopen once you have logs

wuguishifu, 2024-05-03:
Sorry, one last thing. Is the throttling intended behavior even when the lowLightBoost prop is set to false? It was my understanding that the throttling would only happen when that was set to true.

Also, I just wanted to say that this is one of the best react-native libraries IMO and I'm sure everyone appreciates the amount of work you put into maintaining and constantly updating and fixing things!

mrousavy, 2024-05-03:
> Is the throttling intended behavior even when the lowLightBoost prop is set to false? It was my understanding that the throttling would only happen when that was set to true.

Yea it was usually only enabled when `lowLightBoost` is enabled, but since I migrated from my custom Camera2 implementation to CameraX, this now is no longer configurable. So I guess we just have to accept it, maybe I'll create a feature request in the Google issue tracker but they only have `setTargetFps(...)`, no min or max.

mrousavy, 2024-05-03:
> Also, I just wanted to say that this is one of the best react-native libraries IMO and I'm sure everyone appreciates the amount of work you put into maintaining and constantly updating and fixing things!

Thank you so much, much appreciated!

Hope you don't take offense in me closing issues without logs, but otherwise this just gets way too messy :)

wuguishifu, 2024-05-29:
Hi Marc, I wanted to touch back on this issue. It seems as though CameraX doesn't let you change the low light boost option manually because it's wrapping the Camera 2 hardware API (?) or something like that. Either way, I was able to get it to behave correctly by adding this code to the frame processor output configuration in https://github.com/mrousavy/react-native-vision-camera/blob/f8b40fd7c7df33d9436ce62cdb2eaa01a652c238/package/android/src/main/java/com/mrousavy/camera/core/CameraSession.kt#L347-L358

```kt
val analyzer = ImageAnalysisBuilder().also { analysis ->
  if (fpsRange != null) {
    val camera2Interop = Camera2Interop.Extender(analysis)
    camera2Interop.setCaptureRequestOption(
      CaptureRequest.CONTROL_AE_TARGET_FPS_RANGE,
      fpsRange
    )
  }

  // ...
}
```

I also needed to add the opt in annotation for the interop:
```kt
@OptIn(ExperimentalGetImage::class, ExperimentalCamera2Interop::class)
@SupressLint("RestrictedApi")
@Supress("LiftReturnOrAssignment")
private fun configureOutputs(configuration: CameraConfiguration) {
  // ...
}
```

I'm not sure of the stability but so far I haven't experienced any issues. As far as I'm aware, this should only affect the FPS range of the auto exposure which is what's causing the low light boost to decrease the FPS even when set to false in the `<Camera />` component.

I've also tested with setting low light boost to true and it works as expected because I'm just using the fpsRange value that has the lower range as `fps / 2` when low light boost is set to true.

I don't plan on submitting a PR because I don't know enough about this library to holistically know what this change would affect, but for my purposes it seems to work. If you think this change would be valuable let me know and I'd be happy to submit a PR -- I definitely want to contribute to this library in some way after all the time and energy it has saved me!

mrousavy, 2024-05-31:
thanks for testing & confirming that btw

mrousavy, 2024-05-31:
and thanks for your sponsorship! ❤️

wuguishifu, 2024-05-31:
I'm glad it was able to help. Thanks for taking the time to look through it and for all the work you're doing on this library!