margelo/react-native-vision-camera #2843: 🐛  Photo is not what is in the preview

inssein, 2024-05-06:
### What's happening?

I just upgraded from 2.x to 4.x to test the new release (not upgrading to new architecture yet). 

My camera app intentionally has a view with a 4:3 aspect ratio, but when I execute `takePhoto`, I seem to get a portrait photo back.

### Reproduceable Code

```tsx
takePhoto({
  flash: "auto",
})
```
```

### Relevant log output

```shell
No relevant log output.
```

### Camera Device

```json
All devices on my iPhone 13 Pro Max
```

### Device

iPhone 13 Pro Max

### VisionCamera Version

4.0.1

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

inssein, 2024-05-06:
I tried scanning through the codebase and there doesn't seem to be any resizing happening - was this just a feature of 2.x that is no longer carried over?

zzz08900, 2024-05-06:
Would mind clarify what is actually happening (and what is wrong there)?
Is the photo rotated by 90 degrees or?

Are you sure you are selecting the right `format` for your desired output?

mrousavy, 2024-05-06:
Hey- 

1. There's no log output here, which is exactly what I need for these kinds of issues. I don't know how else to explain it than to give Xcode/Android Studio run instructions. 
    [image]
2. You didn't try in the example app.

Are you using `useCameraFormat`? That's what selects a format for you.

closed 2024-05-06 by mrousavy

inssein, 2024-05-06:
@mrousavy: I added logs via editing, you may have been looking at the email. That said, I'm not quite sure it's the logs you want. You are right about the example app, I can give it a shot later today.

I also haven't tried `useCameraFormat` yet, because I wasn't sure that was the problem.

If I had to re-describe the problem: I set up the camera with a 4:3 aspect ratio "preview", but when I take the photo, I get something as if the preview was my whole camera screen.

Here is what the preview looks like:
[image]

Here is what the photo ends up looking like:
[image]

mrousavy, 2024-05-06:
Those are not native logs - I need the **native** logs (from Xcode or Android Studio) - those will tell me what format actually gets selected.

But yes I can answer that question right away then; 

>  I also haven't tried useCameraFormat yet, because I wasn't sure that was the problem.
> 
> If I had to re-describe the problem: I set up the camera with a 4:3 aspect ratio "preview", but when I take the photo, I get something as if the preview was my whole camera screen.

The preview view has nothing to do with the photo output - previously in V2 it might have had something to do with it, but now this is fully decoupled (preview can even be disabled with `preview={false}`). So you want to use `useCameraFormat(...)` to select a format that has the photo resolution you desire.

Maybe try something like:

```ts
const format = useCameraFormat(device, [
  { photoAspectRatio: 4 / 3 }, // or 3 / 4, can't remember which way, sorry
  { photoResolution: 'max' }
])
```

mrousavy, 2024-05-06:
btw nice mic, is that the blue jeti? I have the same

inssein, 2024-05-06:
Ah okay, in V2 the preview is exactly what you got when you took the photo, so those two being disconnected concepts is exactly the issue. I tried your tip with the `photoAspectRatio`, and while it's pretty close, I seem to be getting extra content on all 4 sides (compared to the preview).

Any thoughts on how I'd go about ensuring that the bounds of the preview are exactly the same as the photo I take?

And yeah, that's a Blue Yeti. We had some in the office before the pandemic and I was able to take one home. I hate wearing headphones all day, and the mic does a surprising job of not picking up the speaker output to cancel out echo.

mrousavy, 2024-05-06:
> ensuring that the bounds of the preview are exactly the same as the photo I take?

Well yea, make sure your preview is exactly the same aspect ratio as one of the photo sizes.

The `photoWidth`/`photoHeight` (or `photoSize`) values are fixed and given by the Camera hardware. It cannot be flexed/changed on the fly - so you always shoot in those specific formats. V2 wasn't exactly the same as the preview, it might've been a good coincidence that the aspect ratio exactly matched though.

But what you wanna do is figure out which format you want to use, then resize your preview accordingly. or just account for slight aspect ratio mismatches, which is what pretty much every camera app does.

Zoxive, 2024-05-06:
I have a similar issue after upgrading to v4 from v3.

I have my format set to ``` "photoHeight": 1512, "photoWidth": 2688, ``` making it 16:9.

I have a react native view set with its aspect ratio of 9 / 16 for the Preview.

But my preview (set with resizeMode='contain')  has blackbars in v4, and when the photo is taken i see extra padding just like what @inssein is noticing.

(This on a pixel 6 pro,  "react-native-vision-camera": "^4.0.3")

mrousavy, 2024-05-08:
Preview uses a video stream. So video resolution is what counts.

carlezzo, 2025-02-05:
I had a similar problem. After spending one day trying to force photoWidth, photoAspectRatio and nothing work in different devices, I figured out that I had set the prop fps={format?.maxFps} to the Camera component, so, remove it and try it again!