margelo/react-native-vision-camera #2844: 🐛 On iOS 15, takePhoto produces a very low quality photo

xulihang, 2024-05-06:
### What's happening?

Use `takePhoto` produces a very low quality photo.
[image]

### Reproduceable Code

```tsx
const cameraFormat = useCameraFormat(device, [
    { videoResolution: { width: 1920, height: 1080 } },
    { photoResolution: "max"},
    { fps: 60 }
  ])
```

### Relevant log output

```shell
{"height": 2304, "isMirrored": false, "isRawPhoto": false, "metadata": {"DPIHeight": 72, "DPIWidth": 72, "Orientation": 6, "{Exif}": {"ApertureValue": 2.2750070474998703, "BrightnessValue": 4.232735648449586, "ColorSpace": 1, "DateTimeDigitized": "2024:05:06 10:18:42", "DateTimeOriginal": "2024:05:06 10:18:42", "ExifVersion": "0232", "ExposureBiasValue": 1, "ExposureMode": 0, "ExposureProgram": 2, "ExposureTime": 0.01, "FNumber": 2.2, "Flash": 16, "FocalLenIn35mmFilm": 32, "FocalLength": 4.15, "ISOSpeedRatings": [Array], "LensMake": "Apple", "LensModel": "iPhone SE (1st generation) back camera 4.15mm f/2.2", "LensSpecification": [Array], "MeteringMode": 5, "OffsetTime": "+08:00", "OffsetTimeDigitized": "+08:00", "OffsetTimeOriginal": "+08:00", "PixelXDimension": 4096, "PixelYDimension": 2304, "SceneType": 1, "SensingMethod": 2, "ShutterSpeedValue": 6.645155199953828, "SubjectArea": [Array], "SubsecTimeDigitized": "604", "SubsecTimeOriginal": "604", "WhiteBalance": 0}, "{MakerApple}": {"1": 14, "14": 0, "20": 1, "23": 0, "25": 0, "3": [Object], "31": 0, "32": "6F50671D-4176-4F87-BC2A-567FA9C0637E", "37": 0, "38": 0, "39": 0, "4": 1, "43": "91A94A90-0DAD-456B-83A9-9B0E3E3BBA67", "47": 0, "5": 498, "54": 859, "59": 0, "6": 499, "60": 0, "65": false, "7": 1, "74": 2, "8": [Array]}, "{TIFF}": {"DateTime": "2024:05:06 10:18:42", "HostComputer": "iPhone SE (1st generation)", "Make": "Apple", "Model": "iPhone SE (1st generation)", "ResolutionUnit": 2, "Software": "15.1", "XResolution": 72, "YResolution": 72}}, "orientation": "landscape-right", "path": "file:///private/var/mobile/Containers/Data/Application/9D4FAB56-CD7E-4E3A-8932-7D85A8444533/tmp/CF7501B6-AD9B-4B31-BB8D-A3B4809B8F49.jpeg", "thumbnail": null, "width": 4096}
```

### Camera Device

```json
{
  "hasTorch": true,
  "neutralZoom": 1,
  "supportsFocus": true,
  "physicalDevices": [
    "wide-angle-camera"
  ],
  "supportsRawCapture": false,
  "isMultiCam": false,
  "sensorOrientation": "landscape-right",
  "maxExposure": 8,
  "supportsLowLightBoost": false,
  "minZoom": 1,
  "position": "back",
  "hardwareLevel": "full",
  "maxZoom": 16,
  "minFocusDistance": 10,
  "name": "Back Camera",
  "minExposure": -8,
  "hasFlash": true,
  "formats": [],
  "id": "com.apple.avfoundation.avcapturedevice.built-in_video:0"
}
```

### Device

iPhone SE 1st iOS 15.1

### VisionCamera Version

4.0.1

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2024-05-06:
Again, I'm sorry - there's not a lot I can do with that information.

I don't have any native log output, I don't have a reproduction example, and you didn't try in the example app.

xulihang, 2024-05-06:
I don't think you actually need the log.

In vision camea v3, it can take a high-res photo with `enableHighQualityPhotos={true}`. 

But it is removed in v4 and in the docs, there is not a migration guide or mention of how to taking a high-res photo. I think the docs and the default behavior of taking a photo on iOS should be improved.

mrousavy, 2024-05-06:
Ah yes, `enableHighQualityPhotos` is gone because we don't need it anymore. We always just set the value to the one supplied in `format.photoWidth`/`format.photoHeight`.

But actually now that I think about this, I think this is only available on iOS 16+...

I'm not sure if I want to add backwards compatible support for iOS 15, or just tell user's to upgrade to iOS 16 😅

tony-xlh, 2024-05-07:
My device iPhone SE is old and cannot upgrade to iOS 16.

Compatible support for old iOS versions is often needed by our company's SDK's customers.

I checked your code and it does have `isHighResolutionPhotoEnabled`. Trying to figure out why it does not work...

https://github.com/mrousavy/react-native-vision-camera/blob/cbfa0c6e65b376f08e96681d3de08486560b4a4c/package/ios/Core/CameraSession%2BPhoto.swift#L44-L49

xulihang, 2024-05-07:
My device iPhone SE is old and cannot upgrade to iOS 16.

Compatible support for old iOS versions is often needed by our company's SDK's customers.

I checked your code and it does have `isHighResolutionPhotoEnabled`. Trying to figure out why it does not work...

https://github.com/mrousavy/react-native-vision-camera/blob/cbfa0c6e65b376f08e96681d3de08486560b4a4c/package/ios/Core/CameraSession%2BPhoto.swift#L44-L49

frenbergFNX, 2024-06-04:
I had the same or similar result (over exposed and blurry) on iPhone 15 Pro running iOS 17.5.1 today, happened a couple of times when working on a different issue, can't really reproduce yet but I dont think this is an ios 15 only issue.

xulihang, 2024-07-04:
I've figured out the reason in my case.

I disabled frameprocessor before the taking photo action which causes this problem.

```js
frameProcessor={taken ? undefined: frameProcessor}
```

closed 2024-07-04 by xulihang