margelo/react-native-vision-camera #2845: 💭 Overlay using SVG does not match the camera preview on Android

xulihang, 2024-05-06:
### Question

Before vision camera v4, I can use react-native-svg to draw the overlays.

I set the `viewBox` to the size of the frame and set its `preserveAspectRatio` to `xMidYMid slice`

But using vision camera v4, the overlay cannot match the preview perfectly on Android.

[image]

I know I can now use Skia to draw to the frames. But as my old examples are all using react-native-svg, I am still using it to draw the overlay. I would like to know how to match the two correctly.

### VisionCamera Version

4.0.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22💭+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

rahuldasgupta, 2024-05-06:
Hi, are you able to generate builds with Expo? I'm using:
"expo": "~48.0.21",
"react-native-vision-camera": "^4.0.1"

but when I run `npx expo run:android`, it shows the error tagged below
[image]

xulihang, 2024-05-06:
> Hi, are you able to generate builds with Expo? I'm using: "expo": "~48.0.21", "react-native-vision-camera": "^4.0.1"
>
> but when I run `npx expo run:android`, it shows the error tagged below [...]

Check out this: https://github.com/expo/expo/issues/19534#issuecomment-1277597917

closed 2024-05-06 by mrousavy

xulihang, 2024-05-06:
I think there is some difference between v2-3 and v4 on how the camera preview is sized. I am just lazy since if you know then I don't have to figure this out myself.

mrousavy, 2024-05-06:
Yea no I'd have to play around with this myself as well.

Skia Frame Processors use exactly the Frame as a coordinate system, but if you want to draw something onto the screen you need to calculate the coordinates manually - which can be a bit tricky.

_Maybe_ I can add a helper method for that, but I think that's going to be asynchronous because it has to be used from the UI Thread (not sure why those limitations exist on native).

Otherwise you can calculate it yourself by getting video width/height -> aspect ratio, getting view width/height -> aspect ratio, calculate center crop (either height or width overflow), then put it on screen.

tony-xlh, 2024-05-07:
Having a method to draw the overlay would be very helpful.

xulihang, 2024-07-04:
Setting the resizeMode to contain can make the SVG match with the camera preview in my case.