margelo/react-native-vision-camera #2846: Open Flash light  by click on the Icon

zoobibackups, 2024-05-06:
### Question

how I open flash light when I click on Icon

### VisionCamera Version

"react-native-vision-camera": "^4.0.1"

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22💭+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

zoobibackups, 2024-05-06:
> ### Question
> how I open flash light when I click on Icon
> 
> ### What I tried
const toggleFlash =() => {
      cameraRef.current?.CameraDevice({
        hasTorch: 'on'
      })
    }
> ### VisionCamera Version
> "react-native-vision-camera": "^4.0.1"
>
> ### Additional information
> * [ ]  I am using Expo
> * [x]  I have read the [Troubleshooting [...]

bruno-centanaro, 2024-05-06:
Why not have a state variable for the torch?

zoobibackups, 2024-05-07:
const [torch,setTorch] = useState(false);
const toggleFlash =() => {
    cameraRef.current?.CameraDevice({
    hasTorch: 'on',
    })
    setTorch(!torch)
    }
 <Camera
              hasTorch={torch}
              ref={cameraRef}
              style={StyleSheet.absoluteFill}
              device={device}
              isActive={isActive}
            />

but error is that cameraRef.current.CameraDevice is not a function it is undefined

bruno-centanaro, 2024-05-07:
but why is it not a prop of the camera component?

closed 2024-05-08 by mrousavy