margelo/react-native-vision-camera #2850: Flash documented as `auto` by default, but actually `off`

jkaufman, 2024-05-06:
### What's happening?

Flash [is documented as defaulted to `auto`](https://github.com/mrousavy/react-native-vision-camera/blob/cbfa0c6e65b376f08e96681d3de08486560b4a4c/package/src/types/PhotoFile.ts#L8), but no default value is set.

### Reproduceable Code

```tsx
n/a
```

### Relevant log output

```shell
n/a
```

### Camera Device

```json
n/a
```

### Device

n/a

### VisionCamera Version

4.0.1

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2024-05-08:
Thanks, could you send a PR to change that? Thank you.

closed 2024-05-08 by mrousavy

jkaufman, 2024-05-08:
https://github.com/mrousavy/react-native-vision-camera/pull/2861