margelo/react-native-vision-camera #2851: 🐛 Task :react-native-vision-camera:compileDebugKotlin FAILED on React Native 0.74.X Android Platform

sekhuat, 2024-05-07:
### What's happening?

When install `react-native-vision-camera` to a React Native 0.74.X project, it will throw the `Task :react-native-vision-camera:compileDebugKotlin FAILED` error during compilation for Android. Tested both V3 and V4 of `react-native-vision-camera` module both have the same error.

However, if I using React Native 0.73.X then there will be no issue at all.

Screen Capture On Android Studio with RN 0.74.1
[image]

### Reproduceable Code

```tsx
// Using latest RN 0.74.1 have the issue.
// Create the latest version (0.74.1) of react native project
npx react-native@latest init testcam

// Install the latest version of react-native-vision-camera (4.0.2)
yarn add react-native-vision-camera
npx pod-install

// Compile for Android and will have the issue
npx react-native run-android

// Using older RN 0.73.6 will work without issue
// Create the older version (0.73.6) of react native project
npx react-native@latest init testcam2 --version 0.73.6

// Install the latest version of react-native-vision-camera (4.0.2)
yarn add react-native-vision-camera
npx pod-install

// Compile for Android and will work properly
npx react-native run-android
```

### Relevant log output

```shell
React Native Console Log
========================
> Configure project :react-native-vision-camera
[VisionCamera] Thank you for using VisionCamera ❤️
[VisionCamera] If you enjoy using VisionCamera, please consider sponsoring this project: http://github.com/sponsors/mrousavy
[VisionCamera] node_modules found at /Users/sekhuat/Development/Frontier/testCam/node_modules
[VisionCamera] VisionCamera_enableFrameProcessors is set to true!
[VisionCamera] react-native-worklets-core not found, Frame Processors are disabled!
[VisionCamera] VisionCamera_enableCodeScanner is set to false!

> Task :react-native-vision-camera:compileDebugKotlin

> Task :react-native-vision-camera:compileDebugKotlin FAILED

Deprecated Gradle features were used in this build, making it incompatible with Gradle 9.0.

You can use '--warning-mode all' to show the individual deprecation warnings and determine if they come from your own scripts or plugins.

Android Studio Log
==================
Executing tasks: [:app:assembleDebug] in project /Users/sekhuat/Development/Frontier/testCam/android

> Task :gradle-plugin:checkKotlinGradlePluginConfigurationErrors
> Task :gradle-plugin:compileKotlin UP-TO-DATE
> Task :gradle-plugin:compileJava NO-SOURCE
> Task :gradle-plugin:pluginDescriptors UP-TO-DATE
> Task :gradle-plugin:processResources UP-TO-DATE
> Task :gradle-plugin:classes UP-TO-DATE
> Task :gradle-plugin:jar UP-TO-DATE

> Configure project :react-native-vision-camera
[VisionCamera] Thank you for using VisionCamera ❤️
[... 284 lines cut ...]
### Camera Device

```json
NA
```

### Device

Android Pixel Simulator

### VisionCamera Version

4.0.2

### Can you reproduce this issue in the VisionCamera Example app?

No, I cannot reproduce the issue in the Example app

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

jonjamz, 2024-05-07:
0.74 is not supported yet AFAIK.

danieloprado, 2024-05-07:
A simple workaround using patch-package:

react-native-vision-camera+4.0.3.patch
```diff
diff --git a/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/frameprocessors/VisionCameraProxy.kt b/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/frameprocessors/VisionCameraProxy.kt
index d697bef..3e149f1 100644
--- a/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/frameprocessors/VisionCameraProxy.kt
+++ b/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/frameprocessors/VisionCameraProxy.kt
@@ -13,6 +13,7 @@ import com.mrousavy.camera.core.ViewNotFoundError
 import com.mrousavy.camera.react.CameraView
 import java.lang.ref.WeakReference
 
+@OptIn(FrameworkAPI::class)
 @Suppress("KotlinJniMissingFunction") // we use fbjni.
 class VisionCameraProxy(private val reactContext: ReactApplicationContext) {
   companion object {
```

efstathiosntonas, 2024-05-08:
@danieloprado the patch does not include the `FrameworkAPI` import, proper patch:

```patch
--- a/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/frameprocessors/VisionCameraProxy.kt
+++ b/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/frameprocessors/VisionCameraProxy.kt
@@ -7,12 +7,14 @@ import com.facebook.jni.HybridData
 import com.facebook.proguard.annotations.DoNotStrip
 import com.facebook.react.bridge.ReactApplicationContext
 import com.facebook.react.bridge.UiThreadUtil
+import com.facebook.react.common.annotations.FrameworkAPI
 import com.facebook.react.turbomodule.core.CallInvokerHolderImpl
 import com.facebook.react.uimanager.UIManagerHelper
 import com.mrousavy.camera.core.ViewNotFoundError
 import com.mrousavy.camera.react.CameraView
 import java.lang.ref.WeakReference

+@OptIn(FrameworkAPI::class)
 @Suppress("KotlinJniMissingFunction") // we use fbjni.
 class VisionCameraProxy(private val reactContext: ReactApplicationContext) {
   companion object {
```

danieloprado, 2024-05-08:
Actually should be:

```diff
diff --git a/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/frameprocessors/VisionCameraProxy.kt b/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/frameprocessors/VisionCameraProxy.kt
index d697bef..8de418b 100644
--- a/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/frameprocessors/VisionCameraProxy.kt
+++ b/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/frameprocessors/VisionCameraProxy.kt
@@ -7,12 +7,14 @@ import com.facebook.jni.HybridData
 import com.facebook.proguard.annotations.DoNotStrip
 import com.facebook.react.bridge.ReactApplicationContext
 import com.facebook.react.bridge.UiThreadUtil
+import com.facebook.react.common.annotations.FrameworkAPI
 import com.facebook.react.turbomodule.core.CallInvokerHolderImpl
 import com.facebook.react.uimanager.UIManagerHelper
 import com.mrousavy.camera.core.ViewNotFoundError
 import com.mrousavy.camera.react.CameraView
 import java.lang.ref.WeakReference
 
+@OptIn(FrameworkAPI::class)
 @Suppress("KotlinJniMissingFunction") // we use fbjni.
 class VisionCameraProxy(private val reactContext: ReactApplicationContext) {
   companion object {

```
Need to add the first lines

efstathiosntonas, 2024-05-08:
oops, failed copy paste :face_palm:, thanks @danieloprado

mrousavy, 2024-05-08:
Hey - this is a duplicate of https://github.com/mrousavy/react-native-vision-camera/issues/2614.

New architecture support (which also implies react-native 0.74 support) is tracked in that issue. If you need this, please sponsor the issue/fund the development of that major refactor/migration by donating to the Polar pool mentioned in the issue above. 

Thank you.

closed 2024-05-08 by mrousavy

Gennaro1712, 2024-05-19:
> Actually should be:
>
> ```diff
> diff --git a/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/frameprocessors/VisionCameraProxy.kt [...]

Good afternoon, how can I implement this solution that you mention?

danieloprado, 2024-05-19:
> Good afternoon, how can I implement this solution that you mention?

Create a file at patches/react-native-vision-camera+4.0.3.patch, then install [patch-package](https://www.npmjs.com/package/patch-package)

GeekyCoder25, 2024-05-23:
I've added the two lines on line 10 and line 17 but I'm still getting another error 
[image]

RxmvnLD, 2024-05-25:
Hi, im facing this issue cuz some library broke our project (we dont know which one) and we're forced to update RN :S

Using yarn patch im getting this error, does anyone knows why?
```
YN0000: · Yarn 4.2.2
➤ YN0000: ┌ Resolution step
➤ YN0001: │ Error: react-native-vision-camera@patch:react-native-vision-camera@npm%3A4.0.5#~/.yarn/patches/react-native-vision-camera-npm-4.0.5-0f2bdc9daa.patch: hunk header integrity check failed (expected @@ +0 +10 @@, got @@ +0 +1 @@)
    at $vt (/Users/rxmvnld/.yarn/releases/yarn-4.2.2.cjs:704:7821)
    at Xvt (/Users/rxmvnld/.yarn/releases/yarn-4.2.2.cjs:704:5939)
    at Lv (/Users/rxmvnld/.yarn/releases/yarn-4.2.2.cjs:704:7507)
    at Qj (/Users/rxmvnld/.yarn/releases/yarn-4.2.2.cjs:710:384)
    at _v.getCandidates (/Users/rxmvnld/.yarn/releases/yarn-4.2.2.cjs:710:3324)
    at async Pd.getCandidates (/Users/rxmvnld/.yarn/releases/yarn-4.2.2.cjs:141:1271)
    at async Pd.getCandidates (/Users/rxmvnld/.yarn/releases/yarn-4.2.2.cjs:141:1271)
    at async /Users/rxmvnld/.yarn/releases/yarn-4.2.2.cjs:210:8401
    at async Ky (/Users/rxmvnld/.yarn/releases/yarn-4.2.2.cjs:140:53910)
    at async Fe (/Users/rxmvnld/.yarn/releases/yarn-4.2.2.cjs:210:8383)
➤ YN0000: └ Completed in 1s 399ms
➤ YN0000: · Failed with errors in 1s 409ms
````

Im running 

```
$ yarn patch react-native-vision-camera
```

Then open the temporary folder and making the changes 

```
package com.mrousavy.camera.frameprocessors

import android.util.Log
import androidx.annotation.Keep
import androidx.annotation.UiThread
import com.facebook.jni.HybridData
import com.facebook.proguard.annotations.DoNotStrip
import com.facebook.react.bridge.ReactApplicationContext
import com.facebook.react.bridge.UiThreadUtil
+ import com.facebook.react.common.annotations.FrameworkAPI
import com.facebook.react.turbomodule.core.CallInvokerHolderImpl
import com.facebook.react.uimanager.UIManagerHelper
import com.mrousavy.camera.core.ViewNotFoundError
import com.mrousavy.camera.react.CameraView
import java.lang.ref.WeakReference

+ @OptIn(FrameworkAPI::class)
@Suppress("KotlinJniMissingFunction") // we use fbjni.
```
and finally 

```
$ yarn patch-commit -s /private/var/folders/...
```

varcodeDev, 2024-06-05:
If you using Expo SDK51
The Error is because: 'react-native-vision-camera' does not support yet in Expo SDK51 (react-native 0.74)

Solution: Change to Expo SDK50 (react-native 0.73)
 - step1: yarn add expo@50
 - step 2: npx expo install --fix
- step 3: npx expo run:android --device

GeekyCoder25, 2024-06-05:
How can I revert to sdk50 on my iOS device, the latest version is what on app store

ShahFahad-530, 2024-06-08:
> > Actually should be:
> > ```diff
> > diff --git a/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/frameprocessors/VisionCameraProxy.kt b/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/frameprocessors/VisionCameraProxy.kt
> > index d697bef..8de418b 100644
> > --- a/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/frameprocessors/VisionCameraProxy.kt
> > +++ b/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/frameprocessors/VisionCameraProxy.kt
> > @@ -7,12 +7,14 @@ import com.facebook.jni.HybridData
> >  import com.facebook.proguard.annotations.DoNotStrip
> >  import com.facebook.react.bridge.ReactApplicationContext
> >  import com.facebook.react.bridge.UiThreadUtil
> > +import com.facebook.react.common.annotations.FrameworkAPI
> >  import com.facebook.react.turbomodule.core.CallInvokerHolderImpl
> >  import com.facebook.react.uimanager.UIManagerHelper
> >  import com.mrousavy.camera.core.ViewNotFoundError
> >  import com.mrousavy.camera.react.CameraView
> >  import java.lang.ref.WeakReference
> >  
> > +@OptIn(FrameworkAPI::class)
> >  @Suppress("KotlinJniMissingFunction") // we use fbjni.
> >  class VisionCameraProxy(private val reactContext: ReactApplicationContext) {
> >    companion object {
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > Need to add the first lines
> 
> Hi, im facing this issue cuz some library broke our project (we dont know which one) and we're forced to update RN :S
> 
> Using yarn patch im getting this error, does anyone knows why?
> 
> ```
> YN0000: · Yarn 4.2.2
> ➤ YN0000: ┌ Resolution step
> ➤ YN0001: │ Error: react-native-vision-camera@patch:react-native-vision-camera@npm%3A4.0.5#~/.yarn/patches/react-native-vision-camera-npm-4.0.5-0f2bdc9daa.patch: hunk header integrity check failed (expected @@ +0 +10 @@, got @@ +0 +1 @@)
>     at $vt (/Users/rxmvnld/.yarn/releases/yarn-4.2.2.cjs:704:7821)
>     at Xvt (/Users/rxmvnld/.yarn/releases/yarn-4.2.2.cjs:704:5939)
>     at Lv (/Users/rxmvnld/.yarn/releases/yarn-4.2.2.cjs:704:7507)
>     at Qj (/Users/rxmvnld/.yarn/releases/yarn-4.2.2.cjs:710:384)
>     at _v.getCandidates (/Users/rxmvnld/.yarn/releases/yarn-4.2.2.cjs:710:3324)
>     at async Pd.getCandidates (/Users/rxmvnld/.yarn/releases/yarn-4.2.2.cjs:141:1271)
>     at async Pd.getCandidates (/Users/rxmvnld/.yarn/releases/yarn-4.2.2.cjs:141:1271)
>     at async /Users/rxmvnld/.yarn/releases/yarn-4.2.2.cjs:210:8401
>     at async Ky (/Users/rxmvnld/.yarn/releases/yarn-4.2.2.cjs:140:53910)
>     at async Fe (/Users/rxmvnld/.yarn/releases/yarn-4.2.2.cjs:210:8383)
> ➤ YN0000: └ Completed in 1s 399ms
> ➤ YN0000: · Failed with errors in 1s 409ms
> ```
> 
> Im running
> 
> ```
> $ yarn patch react-native-vision-camera
> ```
> 
> Then open the temporary folder and making the changes
> 
> ```
> package com.mrousavy.camera.frameprocessors
> 
> import android.util.Log
> import androidx.annotation.Keep
> import androidx.annotation.UiThread
> import com.facebook.jni.HybridData
> import com.facebook.proguard.annotations.DoNotStrip
> import com.facebook.react.bridge.ReactApplicationContext
> import com.facebook.react.bridge.UiThreadUtil
> + import com.facebook.react.common.annotations.FrameworkAPI
> import com.facebook.react.turbomodule.core.CallInvokerHolderImpl
> import com.facebook.react.uimanager.UIManagerHelper
> import com.mrousavy.camera.core.ViewNotFoundError
> import com.mrousavy.camera.react.CameraView
> import java.lang.ref.WeakReference
> 
> + @OptIn(FrameworkAPI::class)
> @Suppress("KotlinJniMissingFunction") // we use fbjni.
> ```
> 
> and finally
> 
> ```
> $ yarn patch-commit -s /private/var/folders/...
> ```

Thanks a lot . It worked for me

ShahFahad-530, 2024-06-08:
> Actually should be:
>
> ```diff
> diff --git a/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/frameprocessors/VisionCameraProxy.kt [...]

Thanks bro. It worked for me

zahid502, 2024-06-10:
for **react-native 0.74.1** it's worked for me [https://github.com/mrousavy/react-native-vision-camera/issues/1547#issuecomment-2158914973](comment)

Audrey-Ann, 2024-06-10:
> Hi, im facing this issue cuz some library broke our project (we dont know which one) and we're forced to update RN :S
>
> Using yarn patch im getting this error, does anyone knows why?
>
> ```
> YN0000: · [...]

Will this fix be included in the next patch release (at least) `4.2.x` ?

kumailSFS, 2024-08-28:
> Hey - this is a duplicate of #2614.
>
> New architecture support (which also implies react-native 0.74 support) is tracked in that issue. If you need this, please sponsor the issue/fund the development [...]

I already have these two lines in this file but still getting same error. I'm using latest version 4.5.2

muhammad-kumail, 2024-08-28:
> @danieloprado the patch does not include the `FrameworkAPI` import, proper patch:
> 
> ```diff
> --- a/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/frameprocessors/VisionCameraProxy.kt
> +++ b/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/frameprocessors/VisionCameraProxy.kt
> @@ -7,12 +7,14 @@ import com.facebook.jni.HybridData
>  import com.facebook.proguard.annotations.DoNotStrip
>  import com.facebook.react.bridge.ReactApplicationContext
>  import com.facebook.react.bridge.UiThreadUtil
> +import com.facebook.react.common.annotations.FrameworkAPI
>  import com.facebook.react.turbomodule.core.CallInvokerHolderImpl
>  import com.facebook.react.uimanager.UIManagerHelper
>  import com.mrousavy.camera.core.ViewNotFoundError
>  import com.mrousavy.camera.react.CameraView
>  import java.lang.ref.WeakReference
> 
> +@OptIn(FrameworkAPI::class)
>  @Suppress("KotlinJniMissingFunction") // we use fbjni.
>  class VisionCameraProxy(private val reactContext: ReactApplicationContext) {
>    companion object {
> ```

I already have these two lines in this file but still getting same error. I'm using latest version 4.5.2

sujan-york-ie, 2024-08-30:
> I already have these two lines in this file but still getting same error. I'm using latest version 4.5.2

I am also facing this issue.
After adding these two lines, getting error that - 
import com.facebook.react.common.annotations.FrameworkAPI <---------Unresolved references