margelo/react-native-vision-camera #2853: 🐛 Camera Preview does not align with taken picture

Zoxive, 2024-05-07:
### What's happening?

After upgrading to v4 (from v3) the preview with ``resizeMode='contain'`` is not respecting the cameraFormats aspect ratios.

In my use-case for react-native-vision-camera, I need to allow my users to align fairly precisely with the Preview view so when they do take a picture its aligned.
I am using ML models for object detection, but at the moment after the picture is taken. (I've already dabbled with frame processors to do the ML, but I have some improvements in my ML model to do first)

In the Example App, resizeMode is not used. Is the resizeMode prop mostly dead with v4?

Example Reproduction Code: https://github.com/Zoxive/ReactNativeVisionCamera_PreviewBug

# Video Recordings Examples
## using resizeMode = 'contain'

https://github.com/mrousavy/react-native-vision-camera/assets/124676/efcdc85c-de31-4afe-a988-09b56670b7c7

- Viewport is specified in ReactNative to be 16:9, but there are still blackbars on the Preview, and my image & video aspect ratio is also 16:9 specified via the cameraFormat
- After taking the picture lots of cropping is done, thus making it very difficult to take pictures that need some sort of alignments

## using resizeMode = 'cover'

https://github.com/mrousavy/react-native-vision-camera/assets/124676/b2f2ffb9-02ba-4059-a727-f8802fd08d90

- Cover is closer to what is desired, but still not perfect
- Cover feels like the preview is zoomed in to fit.

Possibly Similar Issue: https://github.com/mrousavy/react-native-vision-camera/issues/2843

### Reproduceable Code

```tsx
const targetFps = 30;
const cameraFormat: FormatFilter[] = [{ fps: targetFps }, { photoAspectRatio: 16 / 9 }, { videoAspectRatio: 16 / 9 }];

function ViewPort() {
	const device = useCameraDevice("back");
	const format = useCameraFormat(device, cameraFormat);
	const fps = Math.min(format?.maxFps ?? 1, targetFps);

	const child =
		device != null ? (
			<Camera
				device={device}
				format={format}
				photo={true}
				fps={fps}
				isActive={true}
				style={styles.camera}
				resizeMode="contain"
			/>
		) : null;

	return <View style={styles.viewPort}>{child}</View>;
}

const styles = StyleSheet.create({
	container: {
		flex: 1,
		backgroundColor: "#fff",
		flexDirection: "column",
	},
	viewPort: {
		aspectRatio: 9 / 16,
		backgroundColor: "red",
	},
});
```

### Relevant log output

[... 332 lines cut ...]
  "hasTorch": true,
  "position": "back",
  "id": "0"
}
```

### Device

Google Pixel 6 Pro

### VisionCamera Version

4.0.3

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [X] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2024-05-08:
Hey!

Preview is a video-stream, so it uses the aspect ratio from `videoResolution`/`videoAspectRatio` in the Format.

You could prioritize video aspect ratio by moving it up, but then the photo still looks different.

I guess some photos just have a different aspect ratio for the photos stream or the video stream.

Does it look proper when you set both `photoAspectRatio` and `videoAspectRatio` to `4 / 3`? Remove the `fps` from the filter as well, that may exclude some filters as it is spot 1 (highest priority)

Zoxive, 2024-05-08:
I removed all filters, only put in a `videoAspectRatio` and nothing changes, also tried combinations of just `photoAspectRatio` and flipping their order.

It just feels like `resizeMode='contain'` is hard coded (somewhere) to 4/3 ish.

I think after all this testing, im just going to use `resizeMode='cover'` (or really just remove the prop as its the default), as that seems like what all example apps use and its fairly close. 

TL;DR
After lots of testing i think best practices with v4 is use no `resizeMode` (aka 'cover'), and ensure that the Preview's `<View>`'s aspect ratio is the same as your photo format.

And maybe recommend to delete this prop since it doesn't seem to do much with v4 just causes confusion? Or I'm still doing something wrong.

mrousavy, 2024-05-08:
> And maybe recommend to delete this prop since it doesn't seem to do much with v4 just causes confusion?

Not true. On iOS it works fine. Also it does change behaviour on Android. Why would I delete that prop?

xulihang, 2024-05-10:
Seems related to this issue: https://github.com/mrousavy/react-native-vision-camera/issues/2845

gnanamoorthyparameshwaran, 2024-05-14:
I am also facing this issue with iPad Pro 10.5 inch(17.4.1) in front camera, for back camera it is working fine and i also tested with android 12 (vivo s1 pro) it is working good.

```
import {useRef, useState} from 'react';
import {Button, Image, Text, View} from 'react-native';
import {
  Camera,
  useCameraDevice,
  useCameraFormat,
  useCameraPermission,
} from 'react-native-vision-camera';

export default function App() {
  const [state, setState] = useState({
    position: 'front',
    url: '',
  });
  const cameraRef = useRef(null);

  const {hasPermission, requestPermission} = useCameraPermission();
  if (!hasPermission) requestPermission();

  const device = useCameraDevice(state.position);

  const format = useCameraFormat(device, [
    {photoAspectRatio: 4 / 3},
    {videoAspectRatio: 4 / 3},
    {
      photoResolution: {
        height: 520,
        width: 390,
      },
    },
    {
      videoResolution: {
        height: 520,
        width: 390,
      },
    },
  ]);

  const CaptureImage = async () => {
    const photo = await cameraRef.current?.takePhoto();
    console.log(photo);
    setState(prev => ({
      ...prev,
      url: photo?.path,
    }));
  };

  if (!hasPermission)
    return (
      <View style={{flex: 1, justifyContent: 'center', alignItems: 'center'}}>
        <Text>Camera permission is required</Text>
      </View>
    );

  if (state.url !== '')
    return (
      <View style={{flex: 1}}>
        <Image
          style={{flex: 1, resizeMode: 'contain'}}
          source={{uri: 'file://' + state.url}}
        />
        <Button
          title="Capture again"
          color={'green'}
          onPress={() => {
            setState(prev => ({
              ...prev,
              url: '',
            }));
          }}
        />
      </View>
    );

  return (
    <View style={{flex: 1, backgroundColor: '#fff'}}>
      <View style={{flex: 1}}>
        <Camera
          ref={cameraRef}
          photo
          isActive
          format={format}
          device={device}
          style={{
            // height: 520,
            // width: 390,
            aspectRatio: 3 / 4,
          }}
        />
      </View>
      <View
        style={{
          flexDirection: 'row',
          justifyContent: 'space-evenly',
        }}>
        <Button title="Capture" color={'green'} onPress={CaptureImage} />
        <Button
          title="Switch"
          color={'red'}
          onPress={() => {
            setState(prev => ({
              ...prev,
              position: prev.position == 'front' ? 'back' : 'front',
            }));
          }}
        />
      </View>
    </View>
  );
}

```

lj45, 2024-06-02:
same issue, in cover mode, the preview size is slightly larger than the actual photo size. How to solve this problem?

gauravrana0-nagarro, 2024-06-24:
I just installed shadow lens app on my android device. the same issue is in Shadow Lens app also

lj45, 2024-06-24:
O(∩_∩)O~邮件已至，谢谢！

BoavistaLudwig, 2024-08-03:
I am facing a similar situation. I want to use photoAspectRatio 1 / 1 to get a square with the same height and width. Not by using resizeMode="cover" (which is the default).

  const Format = useCameraFormat(device, [
    { photoAspectRatio: 1 },
    { photoResolution: "max" },
    { videoAspectRatio: 1 },
  ]);

   <Camera
            ref={camera}
            style={StyleSheet.absoluteFill}
            device={device}
            isActive={hasPermission}
            photo={true}
            /*          resizeMode="contain" */
            format={Format}
            enableZoomGesture={true}
          />
`

With this, I still receive:  LOG  Photo width x height: 3696 3024.

lj45, 2024-08-03:
O(∩_∩)O~邮件已至，谢谢！

MarianAlexandruAlecu, 2024-08-29:
@mrousavy I've tried using [vision-camera-resize-plugin](https://github.com/mrousavy/vision-camera-resize-plugin) to resize the frame to a fixed 720px / 720px, but the preview stays exactly the same on Android - it always takes the video stream aspect ratio.

Shouldn't frame processors be able to resize & crop the frames with no limitations?

lj45, 2024-08-29:
O(∩_∩)O~邮件已至，谢谢！