margelo/react-native-vision-camera #2854: 🐛 [V4] Front camera recording is inverted when going from Back to Front camera during recording on Android

likecarson, 2024-05-07:
### What's happening?

On an Android device, while taking a video recording with the Back camera started, if user switches to Front camera during the same recording the Front video becomes inverted.

Interestingly, if I start the recording using the Front camera then switch to Back camera the recording of the Front video is fine.

### Reproduceable Code

```tsx
const [cameraPosition, setCameraPosition] = useState<'front' | 'back'>('back');
  const cameraDevice = useCameraDevice(cameraPosition);

<Camera
            ref={cameraRef}
            device={cameraDevice}
            torchEnabled={torchEnabled}
            onRotate={useCallback(() => {
              setCameraPosition((p) => (p === 'back' ? 'front' : 'back'));
            }, [])}
            onTorchToggle={() => setTorchEnabled((prev) => !prev)}
          />
```

### Relevant log output

```shell
Back Camera:
{"formats": [{"autoFocusSystem": "none", "fieldOfView": 115.98923300825525, "maxFps": 30, "maxISO": 1600, "minFps": 2, "minISO": 25, "photoHeight": 144, "photoWidth": 176, "supportsDepthCapture": false, "supportsPhotoHdr": false, "supportsVideoHdr": false, "videoHeight": 720, "videoStabilizationModes": [Array], "videoWidth": 1280}, {"autoFocusSystem": "none", "fieldOfView": 115.98923300825525, "maxFps": 30, "maxISO": 1600, "minFps": 2, "minISO": 25, "photoHeight": 240, "photoWidth": 320, "supportsDepthCapture": false, "supportsPhotoHdr": false, "supportsVideoHdr": false, "videoHeight": 720, "videoStabilizationModes": [Array], "videoWidth": 1280}, {"autoFocusSystem": "none", "fieldOfView": 115.98923300825525, "maxFps": 30, "maxISO": 1600, "minFps": 2, "minISO": 25, "photoHeight": 288, "photoWidth": 352, "supportsDepthCapture": false, "supportsPhotoHdr": false, "supportsVideoHdr": false, "videoHeight": 720, "videoStabilizationModes": [Array], "videoWidth": 1280}, {"autoFocusSystem": "none", "fieldOfView": 115.98923300825525, "maxFps": 30, "maxISO": 1600, "minFps": 2, "minISO": 25, "photoHeight": 480, "photoWidth": 640, "supportsDepthCapture": false, "supportsPhotoHdr": false, "supportsVideoHdr": false, "videoHeight": 720, "videoStabilizationModes": [Array], "videoWidth": 1280}, {"autoFocusSystem": "none", "fieldOfView": 115.98923300825525, "maxFps": 30, "maxISO": 1600, "minFps": 2, "minISO": 25, "photoHeight": 720, "photoWidth": 1280, "supportsDepthCapture": false, "supportsPhotoHdr": false, "supportsVideoHdr": false, "videoHeight": 720, "videoStabilizationModes": [Array], "videoWidth": 1280}, {"autoFocusSystem": "none", "fieldOfView": 115.98923300825525, "maxFps": 30, "maxISO": 1600, "minFps": 2, "minISO": 25, "photoHeight": 960, "photoWidth": 1280, "supportsDepthCapture": false, "supportsPhotoHdr": false, "supportsVideoHdr": false, "videoHeight": 720, "videoStabilizationModes": [Array], "videoWidth": 1280}], "hardwareLevel": "limited", "hasFlash": false, "hasTorch": false, "id": "10", "isMultiCam": false, "maxExposure": 6, "maxZoom": 1, "minExposure": -6, "minFocusDistance": 999.999985098839, "minZoom": 1, "name": "10 (BACK) androidx.camera.camera2", "neutralZoom": 1, "physicalDevices": ["ultra-wide-angle-camera"], "position": "back", "sensorOrientation": "landscape-left", "supportsFocus": false, "supportsLowLightBoost": false, "supportsRawCapture": false}

Front Camera:
{"formats": [{"autoFocusSystem": "contrast-detection", "fieldOfView": 77.70146040253039, "maxFps": 30, "maxISO": 1000, "minFps": 7, "minISO": 100, "photoHeight": 1440, "photoWidth": 1920, "supportsDepthCapture": false, "supportsPhotoHdr": false, "supportsVideoHdr": false, "videoHeight": 720, "videoStabilizationModes": [Array], "videoWidth": 1280}, {"autoFocusSystem": "contrast-detection", "fieldOfView": 77.70146040253039, "maxFps": 30, "maxISO": 1000, "minFps": 7, "minISO": 100, "photoHeight": 1080, "photoWidth": 1920, "supportsDepthCapture": false, "supportsPhotoHdr": false, "supportsVideoHdr": false, "videoHeight": 720, "videoStabilizationModes": [Array], "videoWidth": 1280}, {"autoFocusSystem": "contrast-detection", "fieldOfView": 77.70146040253039, "maxFps": 30, "maxISO": 1000, "minFps": 7, "minISO": 100, "photoHeight": 960, "photoWidth": 1920, "supportsDepthCapture": false, "supportsPhotoHdr": false, "supportsVideoHdr": false, "videoHeight": 720, "videoStabilizationModes": [Array], "videoWidth": 1280}, {"autoFocusSystem": "contrast-detection", "fieldOfView": 77.70146040253039, "maxFps": 30, "maxISO": 1000, "minFps": 7, "minISO": 100, "photoHeight": 1200, "photoWidth": 1600, "supportsDepthCapture": false, "supportsPhotoHdr": false, "supportsVideoHdr": false, "videoHeight": 720, "videoStabilizationModes": [Array], "videoWidth": 1280}, {"autoFocusSystem": "contrast-detection", "fieldOfView": 77.70146040253039, "maxFps": 30, "maxISO": 1000, "minFps": 7, "minISO": 100, "photoHeight": 1080, "photoWidth": 1440, "supportsDepthCapture": false, "supportsPhotoHdr": false, "supportsVideoHdr": false, "videoHeight": 720, "videoStabilizationModes": [Array], "videoWidth": 1280}, {"autoFocusSystem": "contrast-detection", "fieldOfView": 77.70146040253039, "maxFps": 30, "maxISO": 1000, "minFps": 7, "minISO": 100, "photoHeight": 960, "photoWidth": 1280, "supportsDepthCapture": false, "supportsPhotoHdr": false, "supportsVideoHdr": false, "videoHeight": 720, "videoStabilizationModes": [Array], "videoWidth": 1280}, {"autoFocusSystem": "contrast-detection", "fieldOfView": 77.70146040253039, "maxFps": 30, "maxISO": 1000, "minFps": 7, "minISO": 100, "photoHeight": 720, "photoWidth": 1280, "supportsDepthCapture": false, "supportsPhotoHdr": false, "supportsVideoHdr": false, "videoHeight": 720, "videoStabilizationModes": [Array], "videoWidth": 1280}, {"autoFocusSystem": "contrast-detection", "fieldOfView": 77.70146040253039, "maxFps": 30, "maxISO": 1000, "minFps": 7, "minISO": 100, "photoHeight": 768, "photoWidth": 1024, "supportsDepthCapture": false, "supportsPhotoHdr": false, "supportsVideoHdr": false, "videoHeight": 720, "videoStabilizationModes": [Array], "videoWidth": 1280}, {"autoFocusSystem": "contrast-detection", "fieldOfView": 77.70146040253039, "maxFps": 30, "maxISO": 1000, "minFps": 7, "minISO": 100, "photoHeight": 600, "photoWidth": 800, "supportsDepthCapture": false, "supportsPhotoHdr": false, "supportsVideoHdr": false, "videoHeight": 720, "videoStabilizationModes": [Array], "videoWidth": 1280}, {"autoFocusSystem": "contrast-detection", "fieldOfView": 77.70146040253039, "maxFps": 30, "maxISO": 1000, "minFps": 7, "minISO": 100, "photoHeight": 480, "photoWidth": 720, "supportsDepthCapture": false, "supportsPhotoHdr": false, "supportsVideoHdr": false, "videoHeight": 720, "videoStabilizationModes": [Array], "videoWidth": 1280}, {"autoFocusSystem": "contrast-detection", "fieldOfView": 77.70146040253039, "maxFps": 30, "maxISO": 1000, "minFps": 7, "minISO": 100, "photoHeight": 480, "photoWidth": 640, "supportsDepthCapture": false, "supportsPhotoHdr": false, "supportsVideoHdr": false, "videoHeight": 720, "videoStabilizationModes": [Array], "videoWidth": 1280}, {"autoFocusSystem": "contrast-detection", "fieldOfView": 77.70146040253039, "maxFps": 30, "maxISO": 1000, "minFps": 7, "minISO": 100, "photoHeight": 360, "photoWidth": 640, "supportsDepthCapture": false, "supportsPhotoHdr": false, "supportsVideoHdr": false, "videoHeight": 720, "videoStabilizationModes": [Array], "videoWidth": 1280}, {"autoFocusSystem": "contrast-detection", "fieldOfView": 77.70146040253039, "maxFps": 30, "maxISO": 1000, "minFps": 7, "minISO": 100, "photoHeight": 288, "photoWidth": 352, "supportsDepthCapture": false, "supportsPhotoHdr": false, "supportsVideoHdr": false, "videoHeight": 720, "videoStabilizationModes": [Array], "videoWidth": 1280}, {"autoFocusSystem": "contrast-detection", "fieldOfView": 77.70146040253039, "maxFps": 30, "maxISO": 1000, "minFps": 7, "minISO": 100, "photoHeight": 240, "photoWidth": 320, "supportsDepthCapture": false, "supportsPhotoHdr": false, "supportsVideoHdr": false, "videoHeight": 720, "videoStabilizationModes": [Array], "videoWidth": 1280}], "hardwareLevel": "full", "hasFlash": true, "hasTorch": true, "id": "1", "isMultiCam": true, "maxExposure": 24, "maxZoom": 12.931958198547363, "minExposure": -24, "minFocusDistance": 9.800000093460085, "minZoom": 1, "name": "1 (FRONT) androidx.camera.camera2", "neutralZoom": 1, "physicalDevices": ["wide-angle-camera", "wide-angle-camera", "telephoto-camera"], "position": "front", "sensorOrientation": "landscape-right", "supportsFocus": true, "supportsLowLightBoost": false, "supportsRawC [... cut ...]
### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2024-05-08:
What does inverted mean? Please attach a sample video.

likecarson, 2024-05-08:
Here is a screen recording using the Android Pixel 6 Pro API 34 emulator. The same happens on physical device:

https://github.com/mrousavy/react-native-vision-camera/assets/147250490/c930e17b-1c46-4fa0-89a3-0e96bd6a1a08

mrousavy, 2024-05-08:
Okay, this is a CameraX bug then, nothing I am doing explicitly - could you create a bug report in the Google CameraX issue tracker about this?
 Thanks!

likecarson, 2024-05-13:
@mrousavy I think I figured out the issue -- at least doing this change fixes it for me:

In this file:
node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/CameraSession.kt

I commented out line 308:
      val video = VideoCapture.Builder(recorder).also { video ->
        // Configure Video Output
        // video.setMirrorMode(MirrorMode.MIRROR_MODE_ON_FRONT_ONLY)

And the issue was resolved. I think this is because in my use case, the recording starts using the back camera and the CameraX must set a certain orientation, and when I switch to Front camera during the same recording the orientation is not updated to relative to Front camera, so mirrorMode causes the inversion. 

I'm not sure if there is a way to adjust the code to check if the orientation is correct before applying the setMirrorMode or maybe this is the desired behavior just an issue with CameraX as you say.

Thanks.

mrousavy, 2024-07-10:
Hey - I think I fixed this issue in the latest releases. I update orientation immediately after switching devices. Can you confirm if that works for you?

closed 2024-07-10 by mrousavy

likecarson, 2024-07-11:
Hi Marc,

Yes the latest version seems to be working now! No more inverted selfie
videos on Android.

I appreciate your follow up and all the work you are doing for this great
package.

Thanks,
Carson

mrousavy, 2024-07-11:
Awesome thank you. If you appreciate my work and dedication please consider sponsoring me on GitHub :)