margelo/react-native-vision-camera #2856: Is there a way to scan the Barcode scanner only to scan once?

iLightFPS, 2024-05-07:
### Question

Is there a way to make the codeScanner only scan the barcode once?

Just switched to Vision Camera from React-Native-Camera where i had a simple usestate. Once the barcode was scanned it stopped scanning and only scanned it once. That solution doesn't work on Vision Camera. 

### What I tried

This is the how the code looks like now. I've managed to make it scan the barcode twice before stopping. But my goal is to make it scan only once and then stop. 

```
let scannerEnabled = true; 

  const codeScanner = useCodeScanner({
    codeTypes: ['ean-13'],
    onCodeScanned: codes => {
      if (!scannerEnabled) return; 

      codes.forEach(code => {
        if (!code.value) {
          console.log('Invalid Barcode Data');
          return;
        }

        console.log('Scanned Barcode:', code.value);

        const product = data.find(
          codeMetadata => codeMetadata.id === code.value,
        );

        console.log('Matched Product:', product);

        if (product) {
          console.log('Product Found:', product);
          setModalVisible(true);
          *Removed*
          scannerEnabled = false;
        } else {
          console.log('Product Not Found');
          setModalVisibleTwo(true);
          setId(code.value);
          scannerEnabled = false;
        }
      });
    },
  });
```

### VisionCamera Version

4.0.2

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22💭+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

jthure, 2024-05-08:
I'm guessing your code is inside a render function, so your `let scannerEnabled = true` is just being redeclared and assigned every time. Try using a ref instead 
```typescript
const scannerEnabled = useRef(true)
...
if(!scannerEnabled.current) return
...
scannerEnabled.current = false
```

mrousavy, 2024-05-08:
Yes, you can just use a Ref and decide to stop scanning inside your app's business logic. That's the caller's job, not something a Camera library should do.

closed 2024-05-08 by mrousavy