margelo/react-native-vision-camera #2857: 🐛 Switching between aspect ratios

ppanwin10, 2024-05-07:
### What's happening?

Hi,

When transitioning between various aspect ratios, I saw a brief pause for recalculating size and color. 

Is there a way to improve it?

Record from Example app.
https://github.com/mrousavy/react-native-vision-camera/assets/77748079/ddfcfe32-d717-4736-8f1d-a9b371a78eb8

This is more visible in my own app
https://github.com/mrousavy/react-native-vision-camera/assets/77748079/65dd84c7-a321-44fc-883d-f239654f587d

### Reproduceable Code

```tsx
enum CameraRatio {
  Square = 1,
  _4_3 = 4 / 3,
  _16_9 = 16 / 9,
}

const [screenAspectRatio, setScreenAspectRatio] = useState(CameraRatio._16_9)
const format = useCameraFormat(device, [
    { fps: targetFps },
    { videoAspectRatio: screenAspectRatio },
    { videoResolution: 'max' },
    { photoAspectRatio: screenAspectRatio },
    { photoResolution: 'max' },
])

<View style={styles.rightButtonRow}>
...
<PressableOpacity
  style={styles.button}
  onPress={() => {
    switch (screenAspectRatio) {
      case CameraRatio.Square:
        setScreenAspectRatio(CameraRatio._4_3)
        break
      case CameraRatio._4_3:
        setScreenAspectRatio(CameraRatio._16_9)
        break
      case CameraRatio._16_9:
        setScreenAspectRatio(CameraRatio.Square)
        break
    }
  }}>
  <Text style={styles.text}>{`Ratio`}</Text>
</PressableOpacity>

...
</View>

<ReanimatedCamera
...
resizeMode={'contain'} 
/>
```

### Relevant log output

```shell
12:45:03.620: [info] 📸 VisionCamera.didSetProps(_:): Updating 1 props: [format]
12:45:03.620: [info] 📸 VisionCamera.configure(_:): configure { ... }: Waiting for lock...
12:45:03.621: [info] 📸 VisionCamera.configure(_:): configure { ... }: Updating CameraSession Configuration... Difference(inputChanged: false, outputsChanged: false, videoStabilizationChanged: false, orientationChanged: false, formatChanged: true, sidePropsChanged: true, torchChanged: false, zoomChanged: true, exposureChanged: false, audioSessionChanged: false, locationChanged: false)
12:45:03.621: [info] 📸 VisionCamera.configureFormat(configuration:device:): Configuring Format (3696x3024 | 352x288@60.0 (ISO: 32.0..3072.0))...
1678593546.056333: 2160x3840 yuv Frame (portrait)
ExamplePlugin: 2160 x 3840 Image. Logging 5 parameters:
[... 1039 lines cut ...]
  "supportsFocus": true,
  "supportsRawCapture": false,
  "isMultiCam": false
}
```

### Device

iPhont 11 IOS17.4.1

### VisionCamera Version

4.0.3

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2024-05-08:
Yea - this is expected - you are updating the Format on a Camera, this just needs to be reconfigured. You can trick this by cutting the view yourself, and then cropping the photo in post processing.

Also, your logs don't include the part where the format gets switched, so I can't really say if there is a way to improve this or not.

I'm specifically looking for these logs:

```
12:44:54.979: [info] 📸 VisionCamera.didSetProps(_:): Updating 1 props: [isActive]
12:44:54.979: [info] 📸 VisionCamera.configure(_:): configure { ... }: Waiting for lock...
12:44:54.979: [info] 📸 VisionCamera.configure(_:): configure { ... }: Updating CameraSession Configuration... Difference(inputChanged: false, outputsChanged: false, videoStabilizationChanged: false, orientationChanged: false, formatChanged: false, sidePropsChanged: false, torchChanged: false, zoomChanged: false, exposureChanged: false, audioSessionChanged: false, locationChanged: true)
```

It should say `1 props: [format]`. If that's the case, then it's as fast as it can get.

closed 2024-05-08 by mrousavy

ppanwin10, 2024-05-08:
> Yea - this is expected - you are updating the Format on a Camera, this just needs to be reconfigured. You can trick this by cutting the view yourself, and then cropping the photo in post [...]

ok thank you, let me go back to see my log. 
Cutting the view manually sounds a good idea!