margelo/react-native-vision-camera #2858: 🐛 useSkiaFrameProcessor makes camera being blank

ppanwin10, 2024-05-08:
### What's happening?

Hi again, apologies for bringing up multiple issues consecutively, but I'm really in need of your assistance.

Everything seems to be functioning properly except when I apply useSkiaFrameProcessor to the camera. The camera screen goes blank.

I tried the exact same process in the example app, and it worked as expected. I've compared the setup in my app with the example app, and everything appears to be identical. It's possible that I've made a mistake somewhere, but I'm reaching out to see if you have any insights into what I might have done wrong.

[image]

### Reproduceable Code

```tsx
pacakge.json
"react-native-vision-camera": "^4.0.3",
"react-native-worklets-core": "^1.2.0",
"@shopify/react-native-skia": "^1.2.3",

const frameProcessor = useSkiaFrameProcessor(frame => {
    'worklet';
    frame.render();

    const centerX = frame.width / 2;
    const centerY = frame.height / 2;
    const rect = Skia.XYWHRect(centerX, centerY, 150, 150);
    const paint = Skia.Paint();
    paint.setColor(Skia.Color('red'));
    frame.drawRect(rect, paint);
}, []);

const format = useCameraFormat(device, [
    {fps: DEFAULT_FPS},
    {photoAspectRatio: camera_ratio},
    {photoResolution: 'max'},
    {photoHdr: true},
    {videoAspectRatio: camera_ratio},
    {videoResolution: 'max'},
    {videoHdr: true},
]);

<ReanimatedCameraView
          ref={camera}
          device={device}
          isActive={camera_isActive}
          style={styles.container}
          photo={true}
          video={true}
          audio={false}
          orientation={cameraOrientation}
          animatedProps={animatedProps}
          format={format}
          resizeMode={'cover'}
          frameProcessor={frameProcessor}
          photoHdr={format?.supportsPhotoHdr}
          videoHdr={format?.supportsVideoHdr}
          photoQualityBalance={'quality'}
          fps={fps}
          onError={onError}
/>
```

### Relevant log output

```shell
17:16:20.390: [info] 📸 VisionCamera.didSetProps(_:): Updating 30 props: [onInitialized, alignItems, cameraId, enableBufferCompression, preview, onStarted, onCodeScanned, collapsable, justifyContent, photoHdr, position, isActive, video, resizeMode, fps, onStopped, onError, onViewReady, enableFrameProcessor, videoHdr, format, orientation, audio, height, width, zoom, photoQualityBalance, exposure, photo, onShutter]
17:16:20.392: [info] 📸 VisionCamera.configure(_:): configure { ... }: Waiting for lock...
17:16:20.401: [info] 📸 VisionCamera.configure(_:): configure { ... }: Updating CameraSession Configuration... Difference(inputChanged: true, outputsChanged: true, videoStabilizationChanged: true, orientationChanged: true, formatChanged: true, sidePropsChanged: true, torchChanged: true, zoomChanged: true, exposureChanged: true, audioSessionChanged: true, locationChanged: true)
17:16:20.401: [info] 📸 VisionCamera.configureDevice(configuration:): Configuring Input Device...
17:16:20.401: [info] 📸 VisionCamera.configureDevice(configuration:): Configuring Camera com.apple.avfoundation.avcapturedevice.built-in_video:6...
17:16:20.409: [info] 📸 VisionCamera.configureDevice(configuration:): Successfully configured Input Device!
[... 109 lines cut ...]
  "supportsLowLightBoost": false,
  "hardwareLevel": "full",
  "position": "back"
}
```

### Device

iPhont 11 IOS17.4.1

### VisionCamera Version

4.0.3

### Can you reproduce this issue in the VisionCamera Example app?

No, I cannot reproduce the issue in the Example app

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2024-05-08:
Hm, did you install Reanimated?

ppanwin10, 2024-05-08:
Hi Marc, yes, I do have it installed. 
"react-native-reanimated": "^3.6.0",

> Hm, did you install Reanimated?

mrousavy, 2024-05-08:
Hm, I don't know. `frame.render()` should be responsible for showing the Frame. 

Does e.g. a red rectangle appear if you render it on screen?

ppanwin10, 2024-05-08:
it does not appear in my app but it is working in example app. Anyway, I will keep debug this, you can close this issue. Thank you

ppanwin10, 2024-05-09:
do you think if this is because RN version? My app is currently in 74.0. Also, the FPS graph is always showing 0 in my app, but working fine in example app.

closed 2024-05-10 by mrousavy

islemsyw, 2024-05-11:
Hello, Can you please help me resolve this issue, when i'm using the frame porcessor and write frame.render(), I get the problem "Property 'render' does not exist on type 'Frame'.

wuguishifu, 2024-05-11:
@islemsyw are you using the `useSkiaFrameProcessor` hook? I was getting that a while ago until I realized I was still using the `useFrameProcessor` hook instead of the Skia one.

islemsyw, 2024-05-12:
I tried with useSkiaFrameProcessor hook but got this problem "react-native-vision-camera"' has no exported member named 'useSkiaFrameProcessor'. Did you mean 'useFrameProcessor'?"
Is it because I'm trying on android?

ppanwin10, 2024-05-13:
> I tried with useSkiaFrameProcessor hook but got this problem "react-native-vision-camera"' has no exported member named 'useSkiaFrameProcessor'. Did you mean 'useFrameProcessor'?" Is it because I'm trying on android?

did you upgrade vision camera package?

bigg-S, 2024-06-04:
Hello,

Everything is functioning properly when I use useFrameProcessor but when I apply useSkiaFrameProcessor to the camera. I get an error that says:

>  ERROR  Camera.onError(session/recoverable-error): An unknown error occurred while creating the Camera Session, but the Camera can recover from it. [session/recoverable-error: An unknown error occurred while creating the Camera Session, but the Camera can recover from it.]

This is the relevant part of my package.json:
"react": "18.2.0",
"react-native": "0.72.4",
"@shopify/react-native-skia": "^1.2.3",
"react-native-reanimated": "^3.11.0",
"react-native-vision-camera": "^4.0.3",
"react-native-worklets-core": "^1.3.0",

I appreciate any help, Thanks

usamaabutt, 2024-12-15:
I am also getting this issue https://github.com/mrousavy/react-native-vision-camera/issues/3297