margelo/react-native-vision-camera #2862: 🐛 Bug happening Android 11 on Android Emulator and Physical Devices

macielrsf, 2024-05-09:
### What's happening?

Camera.onError(session/invalid-output-configuration): Failed to configure the Camera Session because the output/stream configurations are invalid! [session/invalid-output-configuration: Failed to configure the Camera Session because the output/stream configurations are invalid!]

[image]

[image]

### Reproduceable Code

```tsx
import React, {useState, useEffect, useRef} from 'react';
import {
    SafeAreaView,
    StyleSheet,
    Text,
    View,
    ActivityIndicator,
    TouchableOpacity,
    Image,
    Dimensions,
} from 'react-native';
import {Camera, useCameraDevice} from 'react-native-vision-camera';

const {width, height} = Dimensions.get('window');

export default function App() {
    const camera = useRef(null);
    const [cameraPermission, setCameraPermission] = useState();
    const [photoPath, setPhotoPath] = useState();

    useEffect(() => {
        (async () => {
            const cameraPermissionStatus =
                await Camera.requestCameraPermission();
            setCameraPermission(cameraPermissionStatus);
        })();
    }, []);

    const device = useCameraDevice('front');
    console.log(camera);

    const handleTakePhoto = async () => {
        try {
            const photo = await camera.current.takePhoto({
                flash: 'on',
            });
            setPhotoPath(photo.path);
        } catch (e) {
            console.log(e);
        }
    };

    const renderContent = () => {
        if (cameraPermission !== 'granted') {
            return null;
        }
        if (device == null) {
            return <ActivityIndicator size="large" color="#1C6758" />;
        }

        return (
            <View>
                <Camera
                    style={{
                        ...StyleSheet.absoluteFill,
                        flex: 1,
                        width,
                        height,
                    }}
[... 376 lines cut ...]
### Camera Device

```json
{"formats": [{"autoFocusSystem": "contrast-detection", "fieldOfView": 77.70146040253039, "maxFps": 30, "maxISO": 1000, "minFps": 7, "minISO": 100, "photoHeight": 1440, "photoWidth": 1920, "pixelFormats": [Array], "supportsDepthCapture": false, "supportsPhotoHdr": false, "supportsVideoHdr": false, "videoHeight": 720, "videoStabilizationModes": [Array], "videoWidth": 1280}, {"autoFocusSystem": "contrast-detection", "fieldOfView": 77.70146040253039, "maxFps": 30, "maxISO": 1000, "minFps": 7, "minISO": 100, "photoHeight": 1080, "photoWidth": 1920, "pixelFormats": [Array], "supportsDepthCapture": false, "supportsPhotoHdr": false, "supportsVideoHdr": false, "videoHeight": 720, "videoStabilizationModes": [Array], "videoWidth": 1280}, {"autoFocusSystem": "contrast-detection", "fieldOfView": 77.70146040253039, "maxFps": 30, "maxISO": 1000, "minFps": 7, "minISO": 100, "photoHeight": 960, "photoWidth": 1920, "pixelFormats": [Array], "supportsDepthCapture": false, "supportsPhotoHdr": false, "supportsVideoHdr": false, "videoHeight": 720, "videoStabilizationModes": [Array], "videoWidth": 1280}, {"autoFocusSystem": "contrast-detection", "fieldOfView": 77.70146040253039, "maxFps": 30, "maxISO": 1000, "minFps": 7, "minISO": 100, "photoHeight": 1200, "photoWidth": 1600, "pixelFormats": [Array], "supportsDepthCapture": false, "supportsPhotoHdr": false, "supportsVideoHdr": false, "videoHeight": 720, "videoStabilizationModes": [Array], "videoWidth": 1280}, {"autoFocusSystem": "contrast-detection", "fieldOfView": 77.70146040253039, "maxFps": 30, "maxISO": 1000, "minFps": 7, "minISO": 100, "photoHeight": 1080, "photoWidth": 1440, "pixelFormats": [Array], "supportsDepthCapture": false, "supportsPhotoHdr": false, "supportsVideoHdr": false, "videoHeight": 720, "videoStabilizationModes": [Array], "videoWidth": 1280}, {"autoFocusSystem": "contrast-detection", "fieldOfView": 77.70146040253039, "maxFps": 30, "maxISO": 1000, "minFps": 7, "minISO": 100, "photoHeight": 960, "photoWidth": 1280, "pixelFormats": [Array], "supportsDepthCapture": false, "supportsPhotoHdr": false, "supportsVideoHdr": false, "videoHeight": 720, "videoStabilizationModes": [Array], "videoWidth": 1280}, {"autoFocusSystem": "contrast-detection", "fieldOfView": 77.70146040253039, "maxFps": 30, "maxISO": 1000, "minFps": 7, "minISO": 100, "photoHeight": 720, "photoWidth": 1280, "pixelFormats": [Array], "supportsDepthCapture": false, "supportsPhotoHdr": false, "supportsVideoHdr": false, "videoHeight": 720, "videoStabilizationModes": [Array], "videoWidth": 1280}, {"autoFocusSystem": "contrast-detection", "fieldOfView": 77.70146040253039, "maxFps": 30, "maxISO": 1000, "minFps": 7, "minISO": 100, "photoHeight": 768, "photoWidth": 1024, "pixelFormats": [Array], "supportsDepthCapture": false, "supportsPhotoHdr": false, "supportsVideoHdr": false, "videoHeight": 720, "videoStabilizationModes": [Array], "videoWidth": 1280}, {"autoFocusSystem": "contrast-detection", "fieldOfView": 77.70146040253039, "maxFps": 30, "maxISO": 1000, "minFps": 7, "minISO": 100, "photoHeight": 600, "photoWidth": 800, "pixelFormats": [Array], "supportsDepthCapture": false, "supportsPhotoHdr": false, "supportsVideoHdr": false, "videoHeight": 720, "videoStabilizationModes": [Array], "videoWidth": 1280}, {"autoFocusSystem": "contrast-detection", "fieldOfView": 77.70146040253039, "maxFps": 30, "maxISO": 1000, "minFps": 7, "minISO": 100, "photoHeight": 480, "photoWidth": 720, "pixelFormats": [Array], "supportsDepthCapture": false, "supportsPhotoHdr": false, "supportsVideoHdr": false, "videoHeight": 720, "videoStabilizationModes": [Array], "videoWidth": 1280}, {"autoFocusSystem": "contrast-detection", "fieldOfView": 77.70146040253039, "maxFps": 30, "maxISO": 1000, "minFps": 7, "minISO": 100, "photoHeight": 480, "photoWidth": 640, "pixelFormats": [Array], "supportsDepthCapture": false, "supportsPhotoHdr": false, "supportsVideoHdr": false, "videoHeight": 720, "videoStabilizationModes": [Array], "videoWidth": 1280}, {"autoFocusSystem": "contrast-detection", "fieldOfView": 77.70146040253039, "maxFps": 30, "maxISO": 1000, "minFps": 7, "minISO": 100, "photoHeight": 360, "photoWidth": 640, "pixelFormats": [Array], "supportsDepthCapture": false, "supportsPhotoHdr": false, "supportsVideoHdr": false, "videoHeight": 720, "videoStabilizationModes": [Array], "videoWidth": 1280}, {"autoFocusSystem": "contrast-detection", "fieldOfView": 77.70146040253039, "maxFps": 30, "maxISO": 1000, "minFps": 7, "minISO": 100, "photoHeight": 288, "photoWidth": 352, "pixelFormats": [Array], "supportsDepthCapture": false, "supportsPhotoHdr": false, "supportsVideoHdr": false, "videoHeight": 720, "videoStabilizationModes": [Array], "videoWidth": 1280}, {"autoFocusSystem": "contrast-detection", "fieldOfView": 77.70146040253039, "maxFps": 30, "maxISO": 1000, "minFps": 7, "minISO": 100, "photoHeight": 240, "photoWidth": 320, "pixelFormats": [Array], "supportsDepthCapture": false, "supportsPhotoHdr": false, "supportsVideoHdr": false, "videoHeight": 720, "videoStabilizationModes": [Array], "videoWidth": 1280}], "hardwareLevel": "full", "hasFlash": true, "hasTorch": true, "id": "1", "isMultiCam": true, "maxExposure": 24, "maxZoom": 8, "minExposure": -24, "minFocusDistance": 0, "minZoom": 1, "name": "1 (FRONT) androidx.camera.camera2", "neutralZoom": 1, "physicalDevices": ["wide-angle-camera", "wide-angle-camera", "telephoto-camera", "telephoto-camera"], "position": "front", "sensorOrientation": "landscape-right", "supportsFocus": true, "supportsLowLightBoost": false, "supportsRawCapture": false}
```

### Device

Android 10, 11, 12 (Especially on Xiaomi phones)

### VisionCamera Version

4.0.3

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

macielrsf, 2024-05-09:
It was able to resolve this issue by rolling back to version 3.9.2.

iqbaldev46, 2024-05-09:
Might be related. #2777

macielrsf, 2024-05-10:
> Might be related. #2777

@iqbaldev46 thank you so much! This solves my problem

c-info, 2024-05-15:
Having the same issue on RedMi Note 7 Pro with Android 11

"react-native-vision-camera": "^4.0.3",

mrousavy, 2024-05-21:
Hey - just created a PR that maybe fixes this - can you test if that works for you? https://github.com/mrousavy/react-native-vision-camera/pull/2897 🙏

closed 2024-05-21 by mrousavy

den1a, 2024-05-29:
Hello. Same problem.
"react-native-vision-camera": "^4.0.5"

samuelbeaulieu, 2024-06-05:
Hi, I'm still getting this error on my Android test device, Samsung Galaxy S21.

I'm using the latest version 4.0.5.

I'm using the prop `photoQualityBalance="speed"`.

```
 ERROR  Camera.onError(session/invalid-output-configuration): Failed to configure the Camera Session because the output/stream configurations are invalid! [session/invalid-output-configuration: Failed to configure the Camera Session because the output/stream configurations are invalid!]
```

Camera device
```
{"formats": [{"autoFocusSystem": "contrast-detection", "fieldOfView": 80.0605180983776, "maxFps": 30, "maxISO": 3200, "minFps": 7, "minISO": 50, "photoHeight": 3024, "photoWidth": 4032, "supportsDepthCapture": false, "supportsPhotoHdr": true, "supportsVideoHdr": false, "videoHeight": 2160, "videoStabilizationModes": [Array], "videoWidth": 3840}, {"autoFocusSystem": "contrast-detection", "fieldOfView": 80.0605180983776, "maxFps": 30, "maxISO": 3200, "minFps": 7, "minISO": 50, "photoHeight": 2268, "photoWidth": 4032, "supportsDepthCapture": false, "supportsPhotoHdr": true, "supportsVideoHdr": false, "videoHeight": 2160, "videoStabilizationModes": [Array], "videoWidth": 3840}, {"autoFocusSystem": "contrast-detection", "fieldOfView": 80.0605180983776, "maxFps": 30, "maxISO": 3200, "minFps": 7, "minISO": 50, "photoHeight": 1816, "photoWidth": 4032, "supportsDepthCapture": false, "supportsPhotoHdr": true, "supportsVideoHdr": false, "videoHeight": 2160, "videoStabilizationModes": [Array], "videoWidth": 3840}, {"autoFocusSystem": "contrast-detection", "fieldOfView": 80.0605180983776, "maxFps": 30, "maxISO": 3200, "minFps": 7, "minISO": 50, "photoHeight": 1728, "photoWidth": 4032, "supportsDepthCapture": false, "supportsPhotoHdr": true, "supportsVideoHdr": false, "videoHeight": 2160, "videoStabilizationModes": [Array], "videoWidth": 3840}, {"autoFocusSystem": "contrast-detection", "fieldOfView": 80.0605180983776, "maxFps": 30, "maxISO": 3200, "minFps": 7, "minISO": 50, "photoHeight": 2160, "photoWidth": 3840, "supportsDepthCapture": false, "supportsPhotoHdr": true, "supportsVideoHdr": false, "videoHeight": 2160, "videoStabilizationModes": [Array], "videoWidth": 3840}, {"autoFocusSystem": "contrast-detection", "fieldOfView": 80.0605180983776, "maxFps": 30, "maxISO": 3200, "minFps": 7, "minISO": 50, "photoHeight": 3024, "photoWidth": 3024, "supportsDepthCapture": false, "supportsPhotoHdr": true, "supportsVideoHdr": false, "videoHeight": 2160, "videoStabilizationModes": [Array], "videoWidth": 3840}, {"autoFocusSystem": "contrast-detection", "fieldOfView": 80.0605180983776, "maxFps": 30, "maxISO": 3200, "minFps": 7, "minISO": 50, "photoHeight": 1080, "photoWidth": 2400, "supportsDepthCapture": false, "supportsPhotoHdr": true, "supportsVideoHdr": false, "videoHeight": 2160, "videoStabilizationModes": [Array], "videoWidth": 3840}, {"autoFocusSystem": "contrast-detection", "fieldOfView": 80.0605180983776, "maxFps": 30, "maxISO": 3200, "minFps": 7, "minISO": 50, "photoHeight": 1440, "photoWidth": 1920, "supportsDepthCapture": false, "supportsPhotoHdr": true, "supportsVideoHdr": false, "videoHeight": 2160, "videoStabilizationModes": [Array], "videoWidth": 3840}, {"autoFocusSystem": "contrast-detection", "fieldOfView": 80.0605180983776, "maxFps": 30, "maxISO": 3200, "minFps": 7, "minISO": 50, "photoHeight": 1080, "photoWidth": 1920, "supportsDepthCapture": false, "supportsPhotoHdr": true, "supportsVideoHdr": false, "videoHeight": 2160, "videoStabilizationModes": [Array], "videoWidth": 3840}, {"autoFocusSystem": "contrast-detection", "fieldOfView": 80.0605180983776, "maxFps": 30, "maxISO": 3200, "minFps": 7, "minISO": 50, "photoHeight": 824, "photoWidth": 1920, "supportsDepthCapture": false, "supportsPhotoHdr": true, "supportsVideoHdr": false, "videoHeight": 2160, "videoStabilizationModes": [Array], "videoWidth": 3840}, {"autoFocusSystem": "contrast-detection", "fieldOfView": 80.0605180983776, "maxFps": 30, "maxISO": 3200, "minFps": 7, "minISO": 50, "photoHeight": 1080, "photoWidth": 1440, "supportsDepthCapture": false, "supportsPhotoHdr": true, "supportsVideoHdr": false, "videoHeight": 2160, "videoStabilizationModes": [Array], "videoWidth": 3840}, {"autoFocusSystem": "contrast-detection", "fieldOfView": 80.0605180983776, "maxFps": 30, "maxISO": 3200, "minFps": 7, "minISO": 50, "photoHeight": 720, "photoWidth": 1280, "supportsDepthCapture": false, "supportsPhotoHdr": true, "supportsVideoHdr": false, "videoHeight": 2160, "videoStabilizationModes": [Array], "videoWidth": 3840}, {"autoFocusSystem": "contrast-detection", "fieldOfView": 80.0605180983776, "maxFps": 30, "maxISO": 3200, "minFps": 7, "minISO": 50, "photoHeight": 1088, "photoWidth": 1088, "supportsDepthCapture": false, "supportsPhotoHdr": true, "supportsVideoHdr": false, "videoHeight": 2160, "videoStabilizationModes": [Array], "videoWidth": 3840}, {"autoFocusSystem": "contrast-detection", "fieldOfView": 80.0605180983776, "maxFps": 30, "maxISO": 3200, "minFps": 7, "minISO": 50, "photoHeight": 720, "photoWidth": 960, "supportsDepthCapture": false, "supportsPhotoHdr": true, "supportsVideoHdr": false, "videoHeight": 2160, "videoStabilizationModes": [Array], "videoWidth": 3840}, {"autoFocusSystem": "contrast-detection", "fieldOfView": 80.0605180983776, "maxFps": 30, "maxISO": 3200, "minFps": 7, "minISO": 50, "photoHeight": 480, "photoWidth": 720, "supportsDepthCapture": false, "supportsPhotoHdr": true, "supportsVideoHdr": false, "videoHeight": 2160, "videoStabilizationModes": [Array], "vide [... cut ...] e, "hasTorch": true, "id": "0", "isMultiCam": false, "maxExposure": 20, "maxZoom": 8, "minExposure": -20, "minFocusDistance": 7.000000066757203, "minZoom": 1, "name": "0 (BACK) androidx.camera.camera2", "neutralZoom": 1, "physicalDevices": ["wide-angle-camera"], "position": "back", "sensorOrientation": "landscape-left", "supportsFocus": true, "supportsLowLightBoost": false, "supportsRawCapture": false}
```

 Is there any information that would help debug this issue that I should provide?

Thanks

DianaMGalindo, 2024-06-08:
@samuelbeaulieu are you using useSkiaFrameProcessor? I am wondering if this has something to do with trying to use SkiaFrameProcessor in combination with the camera. I start getting a similar error only when I try to implement photo={true} in combination with SkiaFrameProcessor. My camera stops opening and I get the following log:

[image]

samuelbeaulieu, 2024-06-11:
I am not using `useSkiaFrameProcessor`

soyadrianhurtado, 2024-07-03:
I have the same problem with "react-native-vision-camera": "^4.3.2" and "react-native": "0.74.2"

matheuscostadesign, 2024-07-10:
I solved the error by passing the prop: 

- `photoQualityBalance={"speed"}`

Note: The property without braces "{}" `photoQualityBalance="speed"` did not apply the change, just inserting the braces `photoQualityBalance={"speed"}`.

jonjamz, 2024-08-17:
I get this error sporadically in development  (Expo development client) on Pixel 6a using 4.5.1. It rarely occurs and may only happen during a refresh. Reloading the app manually and going to the camera view doesn't seem to trigger it.

I have not specified the `photoQualityBalance` prop.

felipejh, 2024-09-20:
I've been facing the issue on the Pixel 8 Pro. I have tried both `qualityPrioritization="speed"` and `qualityPrioritization="quality"`.

- `react-native-vision-camera: 4.5.2`
- `react-native": 0.74.5`

notnout, 2024-09-22:
I'm also hitting this same issue and it is happening with `4.5.3`.
Based on [this question](https://stackoverflow.com/questions/46997776/camera2-api-error-failed-to-create-capture-session) and also https://github.com/mrousavy/react-native-vision-camera/issues/2777 this sounds related to incorrect resolution configuration. Setting `qualityPrioritization` had no effect.

alielzein98, 2025-01-21:
**I solved this by adding the following:**

   const [isCameraActive, setIsCameraActive] = useState(true);
   useFocusEffect(
        React.useCallback(() => {
            setIsCameraActive(true); // Activate camera on focus
            return () => setIsCameraActive(false); // Deactivate camera on blur
        }, [])
    );

                    <Camera
                        style={styles.cameraInsideScannerBox}
                        device={device}
                        codeScanner={codeScanner}
                        isActive={isCameraActive}
                    />

SveBo, 2025-09-03:
> **I solved this by adding the following:**
>
> const [isCameraActive, setIsCameraActive] = useState(true); useFocusEffect( React.useCallback(() => { setIsCameraActive(true); // Activate camera on focus [...]

This does seem to work, but I think its because the camera is first rendered/mounted deactivated and only activated on the second render.
There is probably a bigger issue behind the scenes, but this is a good workaround for now.