margelo/react-native-vision-camera #2863: 🐛 Unable to Retrieve Active Camera and App Crashes on iOS 17.2 Simulator iPhone 15

huantd0111, 2024-05-09:
### What's happening?

**Description:**
When attempting to utilize the camera feature on iOS 17.2 Simulator iPhone 15, the application crashes while trying to fetch the active camera.

**Steps to Reproduce:**

Initialize the camera devices using useCameraDevices().
Log the camera devices to the console.
Try to retrieve the active camera using devices.back.
Log the active camera to the console.
Render the <Camera> component with the retrieved device.
Expected Behavior:
The app should successfully retrieve the active camera and render it without any crashes.

**Actual Behavior:**
The app crashes when trying to retrieve the active camera.

**Last Exception Backtrace:**
`Last Exception Backtrace:
0   CoreFoundation                	    0x7ff80049b751 __exceptionPreprocess + 226
1   libobjc.A.dylib               	    0x7ff800063904 objc_exception_throw + 48
2   CoreFoundation                	    0x7ff8004b0873 +[NSObject(NSObject) instanceMethodSignatureForSelector:] + 0
3   CoreFoundation                	    0x7ff80049fef4 ___forwarding___ + 1459
4   CoreFoundation                	    0x7ff8004a20f8 _CF_forwarding_prep_0 + 120
5   AVFCapture                    	    0x7ff84c851f43 __47+[AVCaptureDevice isEligibleForReactionEffects]_block_invoke + 239
6   libdispatch.dylib             	    0x7ff800157d3a _dispatch_client_callout + 8
7   libdispatch.dylib             	    0x7ff8001590f3 _dispatch_once_callout + 20
8   AVFCapture                    	    0x7ff84c851e52 +[AVCaptureDevice isEligibleForReactionEffects] + 41
9   AVFCapture                    	    0x7ff84c850075 __48+[AVCaptureDevice setUpReactionEffectsStateOnce]_block_invoke + 349
10  libdispatch.dylib             	    0x7ff800157d3a _dispatch_client_callout + 8
11  libdispatch.dylib             	    0x7ff8001590f3 _dispatch_once_callout + 20
12  AVFCapture                    	    0x7ff84c84ff16 +[AVCaptureDevice setUpReactionEffectsStateOnce] + 86
13  AVFCapture                    	    0x7ff84c86da6f +[AVCaptureFigVideoDevice _devicesWithPriorRegisteredDevices:] + 116
14  AVFCapture                    	    0x7ff84c847a81 _registerVideoDevicesOnce + 227
15  AVFCapture                    	    0x7ff84c848dcd +[AVCaptureDevice _devicesWithDeviceTypes:mediaType:position:] + 355
16  AVFCapture                    	    0x7ff84c810cb9 -[AVCaptureDeviceDiscoverySession _initWithDeviceTypes:mediaType:position:allowIOSMacEnvironment:prefersUnsuspendedAndAllowsAnyPosition:] + 163
17  AVFCapture                    	    0x7ff84c810c00 +[AVCaptureDeviceDiscoverySession discoverySessionWithDeviceTypes:mediaType:position:] + 97
18  unitel_reader                 	       0x10ef99ba5 @nonobjc AVCaptureDeviceDiscoverySession.__allocating_init(deviceTypes:mediaType:position:) + 85
19  unitel_reader                 	       0x10ef99729 closure #1 in CameraViewManager.getAvailableCameraDevices(_:reject:) + 137
20  unitel_reader                 	       0x10ef998d0 partial apply for closure #1 in CameraViewManager.getAvailableCameraDevices(_:reject:) + 16
21  unitel_reader                 	       0x10efa356c withPromise(_:_:) + 92
22  unitel_reader                 	       0x10efa3920 withPromise(resolve:reject:_:) + 144
23  unitel_reader                 	       0x10ef99683 CameraViewManager.getAvailableCameraDevices(_:reject:) + 115
24  unitel_reader                 	       0x10ef9a702 @objc CameraViewManager.getAvailableCameraDevices(_:reject:) + 162
25  CoreFoundation                	    0x7ff8004a238c __invoking___ + 140
26  CoreFoundation                	    0x7ff80049f6b3 -[NSInvocation invoke] + 302
27  CoreFoundation                	    0x7ff80049f923 -[NSInvocation invokeWithTarget:] + 70
28  unitel_reader                 	       0x10eaee590 -[RCTModuleMethod invokeWithBridge:module:arguments:] + 2528 (RCTModuleMethod.mm:584)
29  unitel_reader                 	       0x10eaf305b facebook::react::invokeInner(RCTBridge*, RCTModuleData*, unsigned int, folly::dynamic const&, int, (anonymous namespace)::SchedulingContext) + 2043 (RCTNativeModule.mm:196)
30  unitel_reader                 	       0x10eaf26a5 facebook::react::RCTNativeModule::invoke(unsigned int, folly::dynamic&&, int)::$_0::operator()() const + 133 (RCTNativeModule.mm:113)
31  unitel_reader                 	       0x10eaf2619 invocation function for block in facebook::react::RCTNativeModule::invoke(unsigned int, folly::dynamic&&, int) + 25 (RCTNativeModule.mm:104)
32  libdispatch.dylib             	    0x7ff800156a90 _dispatch_call_block_and_release + 12
33  libdispatch.dylib             	    0x7ff800157d3a _dispatch_client_callout + 8
34  libdispatch.dylib             	    0x7ff800166ac0 _dispatch_main_queue_drain + 1420
35  libdispatch.dylib             	    0x7ff800166526 _dispatch_main_queue_callback_4CF + 31
36  CoreFoundation                	    0x7ff8003f7dc4 __CFRUNLOOP_IS_SERVICING_THE_MAIN_DISPATCH_QUEUE__ + 9
37  CoreFoundation                	    0x7ff8003f26ff __CFRunLoopRun + 2463
38  CoreFoundation                	    0x7ff8003f197d CFRunLoopRunSpecific + 557
39  GraphicsServices              	    0x7ff80fe9d08f GSEventRunModal + 137
40  UIKitCore                     	    0x7ff805bbb53d -[UIApplication _run] + 972
41  UIKitCore                     	    0x7ff805bbffab UIApplicationMain + 123
42  unitel_reader                 	       0x10e2fa110 main + 96 (main.m:8)
43  dyld_sim                      	       0x113a8d3e0 start_sim + 10
44  dyld                          	       0x12267241f start + 1903
`

**Environment:**

Operating System: iOS 17.2
[... 206 lines cut ...]
### Camera Device

```json
{"back": undefined, "external": undefined, "front": undefined, "unspecified": undefined}
```

### Device

iOS 17.2 Simulator: iPhone 15

### VisionCamera Version

2.16.8

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [X] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2024-05-10:
VisionCamera V2 is no longer maintained. Upgrade to VisionCamera V4

closed 2024-05-10 by mrousavy

luannv1203, 2024-07-05:
i upgraded to v4 working on ios 17.5.1 but crash on ios 17.2.1 on physical device