margelo/react-native-vision-camera #2864: [V4][Android]🔧 sdk 33 no more supported

francesco-clementi-92, 2024-05-09:
### How were you trying to build the app?

I upgraded from v3 to v4 and now when I try to run the application the build fail asking to increase the compileSdkVersion to 34.
I'm using the feature code scanner

### Full build logs

```tsx
Execution failed for task ':app:checkProdDebugAarMetadata'.
> A failure occurred while executing com.android.build.gradle.internal.tasks.CheckAarMetadataWorkAction
   > 7 issues were found when checking AAR metadata:
     
       1.  Dependency 'androidx.camera:camera-extensions:1.4.0-alpha04' requires libraries and applications that
           depend on it to compile against version 34 or later of the
           Android APIs.
     
           :app is currently compiled against android-33.
     
           Also, the maximum recommended compile SDK version for Android Gradle
           plugin 7.4.2 is 33.
     
           Recommended action: Update this project's version of the Android Gradle
           plugin to one that supports 34, then update this project to use
           compileSdkVerion of at least 34.
     
           Note that updating a library or application's compileSdkVersion (which
           allows newer APIs to be used) can be done separately from updating
           targetSdkVersion (which opts the app in to new runtime behavior) and
           minSdkVersion (which determines which devices the app can be installed
           on).
     
       2.  Dependency 'androidx.camera:camera-video:1.4.0-alpha04' requires libraries and applications that
           depend on it to compile against version 34 or later of the
           Android APIs.
     
           :app is currently compiled against android-33.
     
           Also, the maximum recommended compile SDK version for Android Gradle
           plugin 7.4.2 is 33.
     
           Recommended action: Update this project's version of the Android Gradle
           plugin to one that supports 34, then update this project to use
           compileSdkVerion of at least 34.
     
           Note that updating a library or application's compileSdkVersion (which
           allows newer APIs to be used) can be done separately from updating
           targetSdkVersion (which opts the app in to new runtime behavior) and
           minSdkVersion (which determines which devices the app can be installed
           on).
     
       3.  Dependency 'androidx.camera:camera-lifecycle:1.4.0-alpha04' requires libraries and applications that
           depend on it to compile against version 34 or later of the
           Android APIs.
     
           :app is currently compiled against android-33.
     
           Also, the maximum recommended compile SDK version for Android Gradle
           plugin 7.4.2 is 33.
     
           Recommended action: Update this project's version of the Android Gradle
           plugin to one that supports 34, then update this project to use
           compileSdkVerion of at least 34.
     
           Note that updating a library or application's compileSdkVersion (which
           allows newer APIs to be used) can be done separately from updating
           targetSdkVersion (which opts the app in to new runtime behavior) and
           minSdkVersion (which determines which devices the app can be installed
           on).
     
       4.  Dependency 'androidx.camera:camera-camera2:1.4.0-alpha04' requires libraries and applications that
[... 110 lines cut ...]
```

### VisionCamera Version

4.0.3

### Target platforms

Android

### Operating system

MacOS

### Can you build the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2024-05-10:
Yea, so set your compileSdk to 34.
CameraX needs it, there's no way around it other than to bump it in your build.gradle

closed 2024-05-10 by mrousavy

amosaxe, 2024-05-10:
I know the error log is self explanatory but still @mrousavy I will recommend updating https://react-native-vision-camera.com/docs/guides/troubleshooting for removing such confusion.

mrousavy, 2024-05-10:
ah its outdated in the docs - well if you can shoot over a PR that'd be much appreciated :)

akshatatyokel, 2024-09-27:
@francesco-clementi-92 Hi did you find any solution? Because updating is not working working