margelo/react-native-vision-camera #2866: 🐛 Memory Leak on camera.takePhoto after 6 six shots.

mdottavi, 2024-05-09:
### What's happening?

I integrated latest RNVC 4.0.3 in my project.

After taking 6 photos correctly (I checked files are saved in cache folder of the device) 
the takePhoto method get systematically stuck at the seventh execution (await props.camera.current.takePhoto does never end). 

If I switch to another View and get back to the Camera View it resets and I can still get 6 goot photos before stucking again.

I tried to give some time from one shot and another but it does not change the issue. Fully reproducable Code is provided.

Devices: Motorola G84 5G 

The error from LogCat at seventh shot is:
[ImageReader-4080x3072f22m9-8855-21](id:22970000002d,api:4,p:1476,c:8855) waitForFreeSlotThenRelock TIMED_OUT dequeuedCount=0, acquiredCount=3, mMaxAcquiredBufferCount=9, mMaxDequeuedBufferCount=8
[ImageReader-4080x3072f22m9-8855-21](id:22970000002d,api:4,p:1476,c:8855) dequeueBuffer: No free buffer is found!
getBufferLockedCommon: Stream 3: Can't dequeue next output buffer: Connection timed out (-110)

From the traces it look some buffer is not flushed after image acquisition. 
Do I need to flush/empty some data ?

### Reproduceable Code

```tsx
import React, { useRef, useState, useCallback, useEffect } from 'react';
import Ripple from 'react-native-material-ripple';
import { View } from 'react-native';
import { Camera, useCameraDevice, useCameraPermission}  from 'react-native-vision-camera';

const CameraStyles = {
    containerPortrait: {flex: 1,backgroundColor: '#000000' },
    cameraWrapperPortrait: {flex: 1,justifyContent: 'center',alignItems: 'center',overflow: 'hidden' },
    cameraPortrait: { flex: 1, aspectRatio: 3/4 },
    buttonBarPortrait: { height: 90, padding: 10 },
    buttonsPortrait: { flexDirection: 'row',flex: 1, justifyContent: 'space-between', alignItems: 'center' },
    captureButton: { width: 40, height: 40, borderRadius: 20, backgroundColor: 'rgba(140, 140, 140, 0.3)', borderColor: 'white', borderWidth: 5}
};

function _TakePictureButton ( props){
    console.log("TakePictureButton Rendering...");
    const takePhoto = useCallback(async () => {
        console.log("takePhoto: started");
        try {
          if (props.camera.current == null) throw new Error('Camera ref is null!')
    
          console.log('Taking photo...')
          const photo = await props.camera.current.takePhoto({
            qualityPrioritization: 'quality',
            flash: props.flash,
            enableShutterSound: true,
          })
          console.log('Photo Taken!')
          props.onClick(photo, 'photo')
        } catch (e) {
          console.error('Failed to take photo!', e)
        }
      }, [props.camera, props.flash, props.flash]);
    
    return (
        <Ripple style={CameraStyles.captureButton} 
            rippleColor='rgb(256, 256, 256)' 
            rippleOpacity={1} 
            rippleDuration={400} 
            rippleCentered={true}
            rippleSize={78} 
            onPress={takePhoto}
            >
        </Ripple>
    );
}
[... 171 lines cut ...]
  "hasTorch": true,
  "position": "back",
  "id": "0"
}
```

### Device

Motorola G84 5G

### VisionCamera Version

4.0.3

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mdottavi, 2024-05-09:
Motorola G84 5G (Android 14)
Same thing happens (6 photos good) seventh stucks  also on Android OnePlus Nord 3 (Android 14)

mdottavi, 2024-05-10:
The issue does not show up in the CameraExample as it is....
BUT just removing the 
format={format} 
parameter passed to the Camera Component it happens exactly the same.

mrousavy, 2024-05-10:
hm interesting.

mrousavy, 2024-05-21:
Can you test again with VisionCamera 4.0.5?