margelo/react-native-vision-camera #2867: 🐛 Android | Crashing on getCameraSelectorLensFacing

omerts, 2024-05-10:
### What's happening?

From production, got an app crashed, with the following exception from crashlytics:
```
Fatal Exception: java.lang.IllegalArgumentException: The given lens facing integer: 255 can not be recognized.
       at androidx.camera.camera2.internal.LensFacingUtil.getCameraSelectorLensFacing(LensFacingUtil.java:56)
       at androidx.camera.camera2.internal.Camera2CameraInfoImpl.getLensFacing(Camera2CameraInfoImpl.java:203)
       at com.mrousavy.camera.core.CameraDeviceDetails.<init>(CameraDeviceDetails.kt:41)
       at com.mrousavy.camera.react.CameraDevicesManager.getDevicesJson(CameraDevicesManager.kt:86)
       at com.mrousavy.camera.react.CameraDevicesManager.sendAvailableDevicesChangedEvent(CameraDevicesManager.kt:94)
       at com.mrousavy.camera.react.CameraDevicesManager$initialize$1.invokeSuspend(CameraDevicesManager.kt:71)
       at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
       at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:108)
       at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
       at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
       at java.lang.Thread.run(Thread.java:923)
 ```

### Reproduceable Code

```tsx
<InternalCamera
            codeScanner={codeScanner}
            enableZoomGesture={true}
            style={styles.preview}
            ref={camera}
            device={device}
            audio={false}
            torch={supportsFlash && flash ? 'on' : 'off'}
            isActive={isActive}
            photo={!onBarCodeDetected}
          />
```

### Relevant log output

```shell
Came from crashlytics so I don't have the physical device
```

### Camera Device

```json
Done have the actual device in hand, this is from crashlytics:
Device
Brand:Sanmu - FIH Foxconn
Model:A13 Pro
Orientation:Portrait
RAM free: 1.07 GB
Disk free: 43.4 GB
Operating system
Version:Android 11
Orientation:Portrait
Rooted:No
```

### Device

Sanmu - FIH Foxconn A13 Pro

### VisionCamera Version

4.0.1

### Can you reproduce this issue in the VisionCamera Example app?

No, I cannot reproduce the issue in the Example app

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

zzz08900, 2024-05-13:
This is not a RNVC issue, its a misbehaving phone OS issue.
Ask the device vendor to fix their broken camera implementation.

mrousavy, 2024-05-13:
Yep, exactly as @zzz08900 said. If they cannot implement Cameras properly on their phone, there's nothing I can do.

Also, I wouldn't really be able to help without native logs. You can also set up Crashlytics or Sentry to give you native logs.

closed 2024-05-13 by mrousavy

SameerAther, 2024-06-12:
`androidx.camera.camera2.internal.LensFacingUtil in getCameraSelectorLensFacing at line 56
androidx.camera.camera2.internal.Camera2CameraInfoImpl in getLensFacing at line 203
com.mrousavy.camera.core.CameraDeviceDetails in <init> at line 41
com.mrousavy.camera.react.CameraDevicesManager in getDevicesJson at line 86
com.mrousavy.camera.react.CameraDevicesManager in sendAvailableDevicesChangedEvent at line 94
com.mrousavy.camera.react.CameraDevicesManager$initialize$1 in invokeSuspend at line 71
kotlin.coroutines.jvm.internal.BaseContinuationImpl in resumeWith at line 33
kotlinx.coroutines.DispatchedTask in run at line 108
java.util.concurrent.ThreadPoolExecutor in runWorker at line 1167
java.util.concurrent.ThreadPoolExecutor$Worker in run at line 641
java.lang.Thread in run at line 923`

Issue on Sentry

Device: Infinix PR652B
OS: Android 11

shubhamdeol, 2024-11-24:
Related issue
https://issuetracker.google.com/issues/315957160?pli=1