margelo/react-native-vision-camera #2871: 🐛 Cannot read property 'useCameraDevice' of undefined

XY0987, 2024-05-12:
### What's happening?

I tried to change the version, but it did not match the RN version, and I did not want to switch the RN version. I can successfully run the front-end part of the code in this warehouse, but I do not know why this is

### Reproduceable Code

```tsx
import React, {useEffect, useState} from 'react';
import {PermissionsAndroid, StyleSheet, Text, View} from 'react-native';
import {useCameraDevice, Camera} from 'react-native-vision-camera';

export default function App() {
  const [isHas, setIsHas] = useState(false);
  const camaPermission = 'android.permission.CAMERA';
  useEffect(() => {
    PermissionsAndroid.check(camaPermission).then(result => {
      if (!result) {
        PermissionsAndroid.request(camaPermission)
          .then(status => {
            setIsHas(true);
          })
          .catch(err => {
            console.log(err);
          });
      } else {
        setIsHas(true);
      }
    });
  }, []);

  const device = useCameraDevice('back');
  if (device == null) return <Text>null</Text>;
  return (
    <View style={style.container}>
      {isHas ? (
        <Camera style={{height: 500}} device={device} isActive={true} />
      ) : (
        <Text>Have no authority</Text>
      )}
    </View>
  );
}

const style = StyleSheet.create({
  container: {
    padding: 20,
  },
});
```

### Relevant log output

```shell
Real machine simulation error
![F724FF54A90AF0580981394E589642E7](https://github.com/mrousavy/react-native-vision-camera/assets/105161665/31ac651d-deec-4856-a6e8-80dbf6aa23ba)

![QQ截图20240512172206](https://github.com/mrousavy/react-native-vision-camera/assets/105161665/c6bcc985-6aef-4009-a1a0-5ba4c4b121e9)
```

### Camera Device

```json
{
"react-native": "0.71.1",
    "react-native-vision-camera": "^3.9.2",
    "react-native-worklets-core": "^1.3.2",
}
```

### Device

Android，xiaomi

### VisionCamera Version

3.9.2

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2024-05-13:
Hey - use react-native-worklets-core 0.x.x for VC 3.x.x, and react-native-worklets-core 1.x.x for VC 4.x.x

closed 2024-05-13 by mrousavy

XY0987, 2024-05-13:
This is still true after the switch，If I copy the latest front-end code and reference it directly in the project instead of using the react-native-vision-camera exported code, I can run it directly，
```json
"react-native-worklets-core": "^0.5.0",
```