margelo/react-native-vision-camera #2876: 💭 Disabled location showing an error log

ppanwin10, 2024-05-14:
### Question

I disabled the location config by using $VCEnableLocation = false. And I received the following error message in console. Want to confirm that this error log can be ignored.

[system/location-not-enabled: Location is not enabled, so VisionCamera did not compile the Location APIs. Set $VCEnableLocation in your Podfile, or enableLocation in the expo config plugin and rebuild.]

### What I tried

$VCEnableLocation = false.

enableLocation={false}

### VisionCamera Version

4.0.3

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22💭+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2024-05-14:
Hm, this error should only come when you set `enableLocation={true}`. Are you sure you are not passing true here?

ppanwin10, 2024-05-14:
Yes I am sure `enableLocation={false}` in my code.

I am not familiar with swift code but if I understand correctly, this error returns if VISION_CAMERA_ENABLE_LOCATION is off, even though `configuration.enableLocation` is false. Is there any problem here?

[image]

ppanwin10, 2024-05-14:
BTW, during my testing of location tagging in my app, I noticed that the saved photo doesn't include location information even I enabled location. I haven't tested it with the example app yet; I just wanted to see if anyone else encountered the same issue. For testing, I'm using an iPhone 11.

mrousavy, 2024-05-15:
Ahhhh, `configuration.enableLocation` will be true because the device changed, that's all chained. I've created a PR to fix this: https://github.com/mrousavy/react-native-vision-camera/pull/2882

PR #2882 (2024-05-15): fix: Fix location-not-enabled throwing even if location is disabled

## What

Fixes `location-not-enabled` error being thrown even if location was disabled.

## Changes

## Tested on

## Related issues

- Fixes #2876
closed 2024-05-15 by mrousavy