margelo/react-native-vision-camera #2877: Slow response while scanning 1D barcode in iOS intermittently.

sathiyaraj45, 2024-05-14:
### What's happening?

Sometimes it works perfectly for 1D barcode scanning, but sometimes it doesn't work as expected. Even when we move the camera closer to the object, it is not detected, but 2D barcodes work perfectly all the time. IF anything i missed the config. Thanks in advance. 

### Reproduceable Code

```tsx
<Camera style={styles.camera} device={cameraDevice} isActive={cameraStatus} codeScanner={codeScanner}                                   >
   <View style={styles.overlay}>
        <View style={styles.border}>
              <View style={styles.centerLine} />
        </View>
   </View>
</Camera>
```

### Relevant log output

```shell
N/A
```

### Camera Device

```json
N/A
```

### Device

ipad

### VisionCamera Version

3.9.1

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2024-05-14:
No logs, no device info, and not full code (`useCodeScanner(..)` usage is missing) - not enough info -> closing.

closed 2024-05-14 by mrousavy

sathiyaraj45, 2024-05-15:
```
const codeScanner: CodeScanner = {
       
        codeTypes: ['upc-e', 'ean-13', 'code-128', 'code-39', 'itf', 'qr', 'pdf-417'],
        //codeTypes: ['itf'],
        onCodeScanned: (codes, frame) => {
            console.log(`Scanned ${codes.length} codes!`)
            console.log(frame);
            console.log(codes);
            setCameraSatus(false);
            setCameraModalVisible(!cameraModalVisible);
        }
        ,
        //500x500
        regionOfInterest: {
            x: 0.44,
            y: 0.10,
            width: 0.16,
            height: 0.90
        }
    }
 
```

sathiyaraj45, 2024-05-15:
How can we make the region of interest bordered with a different color?