margelo/react-native-vision-camera #2878: 🐛 Unable to use more than 3 video outputs on Android

santiagofm, 2024-05-14:
### What's happening?

When I try to use more than 3 camera outputs on Android it throws an exception:
```
Camera.onError(unknown/unknown): No supported surface combination is found for camera device - Id : 0.  May be attempting to bind too many use cases. Existing surfaces: [] New configs: [androidx.camera.core.impl.ImageAnalysisConfig@5500c94, androidx.camera.core.impl.ImageCaptureConfig@2ce66a6, androidx.camera.core.streamsharing.StreamSharingConfig@6ea4732, androidx.camera.core.impl.ImageAnalysisConfig@f8c78e7]
```

I'm using the example app code for the code scanner use case and adding a few more "video outputs" for running a frame processor & being able to take photos programatically.

I need to be able to run 4 "video ouputs" at the same time: preview the camera, take photos programatically, scan codes using the built-in code-scanner & run a frame processor:

### Reproduceable Code

```tsx
import * as React from 'react'
import { useCallback } from 'react'
import { StyleSheet, View } from 'react-native'
import type { Code } from 'react-native-vision-camera'
import { useCameraDevice, useCodeScanner, useFrameProcessor } from 'react-native-vision-camera'
import { Camera } from 'react-native-vision-camera'
import { useIsForeground } from './hooks/useIsForeground'
import { StatusBarBlurBackground } from './views/StatusBarBlurBackground'
import { useIsFocused } from '@react-navigation/core'

export function CodeScannerPage(): React.ReactElement {
  // 1. Use a simple default back camera
  const device = useCameraDevice('back')

  // 2. Only activate Camera when the app is focused and this screen is currently opened
  const isFocused = useIsFocused()
  const isForeground = useIsForeground()
  const isActive = isFocused && isForeground

  // 4. On code scanned, console.log
  const onCodeScanned = useCallback((codes: Code[]) => {
    console.log(`Scanned ${codes.length} codes:`, codes)
    const value = codes[0]?.value
    if (value == null) return

    console.log(value)
  }, [])

  // 5. Initialize the Code Scanner to scan QR codes and Barcodes
  const codeScanner = useCodeScanner({
    codeTypes: ['qr', 'ean-13'],
    onCodeScanned: onCodeScanned,
  })

  const frameProcessor = useFrameProcessor((frame) => {
    // nothing to do here
  }, [])

  return (
    <View style={styles.container}>
      {device != null && (
        <Camera
          style={StyleSheet.absoluteFill}
          device={device}
          isActive={isActive}
          photo={true}
          codeScanner={codeScanner}
          enableZoomGesture={true}
          frameProcessor={frameProcessor}
        />
      )}
      <StatusBarBlurBackground />
    </View>
  )
}

[... 193 lines cut ...]
  "hasTorch": false,
  "position": "back",
  "id": "0"
}
```

### Device

Pixel 6 API 30 emulator, also repo on real Xiaomi Redmi 13C

### VisionCamera Version

main branch

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

santiagofm, 2024-05-14:
While debugging I came accross the [`bindToLifecycle`](https://github.com/mrousavy/react-native-vision-camera/blob/main/package/android/src/main/java/com/mrousavy/camera/core/CameraSession.kt#L433) doc with clearly states:
```
...
Currently up to 3 use cases may be bound to a Lifecycle at any time. Exceeding capability of target camera device will throw an IllegalArgumentException.
...
```
So this issue seems to be an api limitation?

I can consistently repro this when I ahve more than 3 "video outputs" enabled at the same time(preview, photo, codescanner, frame processors) and it works fine when I remove any of them.

mrousavy, 2024-05-15:
Can you try setting `qualityPrioritization="speed"`?

I built VisionCamera to not have that limitation (that was a huge issue in native Camera APIs, so it's hard to work around this issue)

santiagofm, 2024-05-15:
> Can you try setting `qualityPrioritization="speed"`?
> 
> I built VisionCamera to not have that limitation (that was a huge issue in native Camera APIs, so it's hard to work around this issue)

I guess you meant `photoQualityBalance` but yeah, same result.

Isn't it a limitation on the camera api itself?

```
...
Currently up to 3 use cases may be bound to a Lifecycle at any time. Exceeding capability of target camera device will throw an IllegalArgumentException.
...
```

santiagofm, 2024-05-15:
@mrousavy  Sorry if I'm missing something implementation wise, but it seems like we are assuming _all_ devices will support binding any number of use cases or at least the camera api will disregard. Which doesn't seem to be the case and the camera API will just throw an exception.

Seems like we should be using the `hardware_level` of the device to [determine the number of use cases that are supported](https://issuetracker.google.com/issues/243168482)? Not sure how the one thing translates to the other though.

I will try to fold the code scanner into my custom frame processor to reduce the number of use cases and workaround this. We cannot use multiple `Camera` components at the same time right? Even if one of them has the preview disabled?

mrousavy, 2024-05-16:
> but it seems like we are assuming all devices will support binding any number of use cases

No;

- On iOS, I am explicitly checking if this is supported and Frame Processors and Video Capture have been carefully put together into a single video stream
- On Android, CameraX already uses a similar feature under the hood called "stream sharing". So you could use any number of outputs here, although that feature is still in beta.

closed 2024-05-21 by mrousavy

santiagofm, 2024-05-21:
> Hey - just created a PR to **maybe** fix this - can you test if that works for you? #2897 🙏

Will test it later today 🙌🏻 ! 

I ended up implementing the barcode recognizer with mlkit on a frame processor to reduce the number of outputs btw, that seems to work fine.

santiagofm, 2024-05-22:
> Hey - just created a PR to **maybe** fix this - can you test if that works for you? #2897 🙏

Hey! I just tested on a Xiaomi Redmi 13C and the issue it's still throwing the error. I'm using 4 video outputs: preview, take photo programatically, built-in code scanner, frame processors.

mrousavy, 2024-05-22:
Well then that means that the StreamSharing feature is not working properly on that specific phone.

We'd need to report that bug to Google/CameraX.
I don't have the time to do that right now, so I'd really appreciate it if you took the time to do that (I mean it's also in your interest to get that fixed)