margelo/react-native-vision-camera #2879: 💭 Differences between iPhone native camera and available camera devices

ppanwin10, 2024-05-14:
### Question

When comparing the iPhone native camera with the device list returned by Camera.getAvailableCameraDevices(), I noticed that the list doesn't include a front device that supports auto-focus, and none of the front or back devices support HDR, although these features are supported in the native camera. Is this a expected behavior?

iPhone 11 IOS 17.4.1

### What I tried

```
const des = Camera.getAvailableCameraDevices();
for (const d of des) {
      console.log(d.name);
      console.log('supportFocus', d.supportsFocus);

      // find if there is any formats has hdr supported
      const hasHDRFormat = d.formats.some(f => {
        return f.supportsPhotoHdr || f.supportsVideoHdr;
      });
      console.log('hasHDR', hasHDRFormat);
}

// result
 LOG  Back Camera
 LOG  supportFocus true
 LOG  hasHDR false
 LOG  Front Camera
 LOG  supportFocus false
 LOG  hasHDR false
 LOG  Front TrueDepth Camera
 LOG  supportFocus false
 LOG  hasHDR false
 LOG  Back Dual Wide Camera
 LOG  supportFocus true
 LOG  hasHDR false
 LOG  Back Ultra Wide Camera
 LOG  supportFocus false
 LOG  hasHDR false
```

### VisionCamera Version

4.0.3

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22💭+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

ppanwin10, 2024-05-15:
ah, just saw this issue someone already raised, it makes to why focus is not working then. https://github.com/mrousavy/react-native-vision-camera/issues/2622

mrousavy, 2024-05-15:
- Video HDR is true 10-bit HDR in VisionCamera. If true 10-bit HDR is not supported, the devices will automatically use EDR (extended dynamic range), which doubles frame rate to capture more exposure. Looks like your iPhone 11 doesn't support 10-bit HDR, only EDR. Which is sometimes also just called HDR, even though it isn't.
- Photo HDR is something the iOS APIs don't support, so I'd have to do that myself by capturing multiple photos as fast as possible (a "bracketed photo sequence"), then combining them via a custom fusion algorithm, such as the Robertson-HDR algorithm (an example implementation: [schulz0r/CoreImage-HDR](https://github.com/schulz0r/CoreImage-HDR)). Currently, I didn't have 100+ hours of free time to implement this, but if you need this feature I'll gladly implement it through my contracting agency :) 
- Supports focus; yea that's weird. Maybe Apple is doing something else that's a private API, but I'm just using their focus method, nothing I can really do if they throw an error there. Maybe create an Apple bug report.

mrousavy, 2024-05-15:
You can check this out in ShadowLens - I record 10-bit video HDR if possible.

closed 2024-05-15 by mrousavy