margelo/react-native-vision-camera #2881: BLANK SCREEN WHILE USING PHOTO={true} on version 4.0.3🐛

aglmnzr16, 2024-05-15:
### What's happening?

I wanted to take a picture, but im getting blank screen. If u try to set photo={false} its gonna be fine, but i can't take a picture, if i set photo to false

### Reproduceable Code

```tsx
const { requestPermission, hasPermission } = useCameraPermission();
    const devices = useCameraDevice('back')
    const camera = useRef<Camera>(null);

    useEffect(() => {
        if (!hasPermission) {
          requestPermission();
        }
      }, [hasPermission, requestPermission]);

    

    const takePhoto = async () => {
        if (camera.current) {
            const photo: PhotoFile = await camera.current.takePhoto({
                flash: 'off',
            });
            console.log(photo);
        }
    };

    const onError = useCallback((error: CameraRuntimeError) => {
        console.error(error)
      }, [])
    

    if (!devices) return <View style={styles.container}>
                            <Text>No device</Text>
                        </View>

    return (
        <View
            style={{
                flex : 1,
                backgroundColor : colors.layoutWhite,
            }}>
            <HeaderComponent title='Kembali' background="#1C1B1F" onBack={() => navigation.goBack()} />
            
            <Camera
                key={devices.id}
                ref={camera}
                style={StyleSheet.absoluteFill}
                device={devices}
                isActive={true}
                // photo={true}
                photo
                onError={onError}
            />
            <View style={styles.buttonContainer}>
                <TouchableOpacity style={styles.button}>
                    <Icon name="image" size={30} color="#fff" />
                </TouchableOpacity>
                <TouchableOpacity  onPress={takePhoto}>
                    <CameraIcon />
                </TouchableOpacity>
                <TouchableOpacity style={styles.button}>
                    <Icon name="camera-party-mode" size={30} color="#fff" />
                </TouchableOpacity>
            </View>
        </View>
    )
```

### Relevant log output

```shell
tcontext=u:r:isolated_app:s0:c230,c257,c512,c768 tclass=dir permissive=0
05-15 17:01:34.075 11273 11273 W .android.chrome: type=1400 audit(0.0:1149420): avc: denied { search } for name="11398" dev="proc" ino=6778939 scontext=u:r:untrusted_app:s0:c125,c256,c512,c768 tcontext=u:r:isolated_app:s0:c230,c257,c512,c768 tclass=dir permissive=0
05-15 17:01:34.079  1060  1086 W BatteryExternalStatsWorker: modem info is invalid: ModemActivityInfo{ mTimestamp=225668300 mSleepTimeMs=69960594 mIdleTimeMs=-22520608 mTxTimeMs[]=[4, 0, 0, 3602, 7659] mRxTimeMs=178799158 mEnergyUsed=0}
05-15 17:01:34.080  1060  1912 D DisplayPowerController: mSettingBrightness = -1 as request 31
05-15 17:01:34.141   595   632 D AALLightSensor: newLux = 75, [57, 57] -> 75
05-15 17:01:34.207  4079  4079 D SensorManager:  acc x 0.114 y 8.208 z 4.719 ts 225668429772277 count 11 to com.facebook.react.common.ShakeDetector@4fec894
05-15 17:01:34.445  4079  4079 D SensorManager:  acc x 0.16 y 8.232 z 4.774 ts 225668668130436 count 11 to com.facebook.react.common.ShakeDetector@4fec894
read: unexpected EOF!
```

### Camera Device

```json
{
  "formats": [],
  "sensorOrientation": "landscape-left",
  "hardwareLevel": "full",
  "maxZoom": 10,
  "minZoom": 1,
  "maxExposure": 4,
  "supportsLowLightBoost": false,
  "neutralZoom": 1,
  "physicalDevices": [
    "wide-angle-camera"
  ],
  "supportsFocus": true,
  "supportsRawCapture": false,
  "isMultiCam": false,
  "minFocusDistance": 5,
  "minExposure": -4,
  "name": "0 (BACK) androidx.camera.camera2",
  "hasFlash": true,
  "hasTorch": true,
  "position": "back",
  "id": "0"
}
```

### Device

Vivo Y17

### VisionCamera Version

4.0.3

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2024-05-15:
There has to be some kind of error. Please paste the full logs, not just 6 lines of debug messages.

yaser-elbatal, 2024-05-15:
i get the same thing blank screen with no error

aglmnzr16, 2024-05-16:
> There has to be some kind of error. Please paste the full logs, not just 6 lines of debug messages.

cannot paste full logs, its a lot message

aglmnzr16, 2024-05-16:
> There has to be some kind of error. Please paste the full logs, not just 6 lines of debug messages.

--------- beginning of crash
05-13 02:28:26.073 10461 10625 F libc    : Fatal signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x0 in tid 10625 (GLThread 134), pid 10461 (uy.growlauncher)
05-13 02:28:26.353 10735 10735 F DEBUG   : Softversion: PD1901_A_1.21.38
05-13 02:28:26.353 10735 10735 F DEBUG   : Time: 2024-05-13 02:28:26
05-13 02:28:26.353 10735 10735 F DEBUG   : *** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
05-13 02:28:26.354 10735 10735 F DEBUG   : Build fingerprint: 'vivo/PD1901/PD1901:9/P00610/compiler01071830:user/release-keys'
05-13 02:28:26.354 10735 10735 F DEBUG   : Revision: '0'
05-13 02:28:26.354 10735 10735 F DEBUG   : ABI: 'arm'
05-13 02:28:26.354 10735 10735 F DEBUG   : pid: 10461, tid: 10625, name: GLThread 134  >>> launcher.powerkuy.growlauncher <<<
05-13 02:28:26.354 10735 10735 F DEBUG   : signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x0
05-13 02:28:26.354 10735 10735 F DEBUG   : Cause: null pointer dereference
05-13 02:28:26.354 10735 10735 F DEBUG   :     r0  ea4cd64a  r1  c01830c8  r2  80000000  r3  00000000
05-13 02:28:26.354 10735 10735 F DEBUG   :     r4  c4132d84  r5  0000075f  r6  00000760  r7  c05fe800
05-13 02:28:26.354 10735 10735 F DEBUG   :     r8  e7443810  r9  ed631d8c  r10 00000000  r11 00f42400
05-13 02:28:26.355 10735 10735 F DEBUG   :     ip  00000020  sp  c0183028  lr  c458a208  pc  c459ef08
05-13 02:28:26.357 10735 10735 F DEBUG   : 
05-13 02:28:26.357 10735 10735 F DEBUG   : backtrace:
05-13 02:28:26.357 10735 10735 F DEBUG   :     #00 pc 0013ff08  /data/app/launcher.powerkuy.growlauncher-cmiD-8fOAySFP8m93ngYZw==/lib/arm/libPowerKuy.so
05-13 02:28:26.357 10735 10735 F DEBUG   :     #01 pc 0012b204  /data/app/launcher.powerkuy.growlauncher-cmiD-8fOAySFP8m93ngYZw==/lib/arm/libPowerKuy.so
05-13 02:28:26.357 10735 10735 F DEBUG   :     #02 pc 00230098  /data/app/launcher.powerkuy.growlauncher-cmiD-8fOAySFP8m93ngYZw==/lib/arm/libPowerKuy.so
05-13 02:36:24.249 10782 10912 F libc    : Fatal signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x176 in tid 10912 (GLThread 150), pid 10782 (uy.growlauncher)
05-13 02:36:24.576 12509 12509 F DEBUG   : Softversion: PD1901_A_1.21.38
05-13 02:36:24.576 12509 12509 F DEBUG   : Time: 2024-05-13 02:36:24
05-13 02:36:24.577 12509 12509 F DEBUG   : *** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
05-13 02:36:24.577 12509 12509 F DEBUG   : Build fingerprint: 'vivo/PD1901/PD1901:9/P00610/compiler01071830:user/release-keys'
05-13 02:36:24.577 12509 12509 F DEBUG   : Revision: '0'
05-13 02:36:24.577 12509 12509 F DEBUG   : ABI: 'arm'
05-13 02:36:24.578 12509 12509 F DEBUG   : pid: 10782, tid: 10912, name: GLThread 150  >>> launcher.powerkuy.growlauncher <<<
05-13 02:36:24.578 12509 12509 F DEBUG   : signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x176
05-13 02:36:24.578 12509 12509 F DEBUG   : Cause: null pointer dereference
05-13 02:36:24.578 12509 12509 F DEBUG   :     r0  00000176  r1  00000176  r2  00000073  r3  00000002
05-13 02:36:24.578 12509 12509 F DEBUG   :     r4  00000176  r5  bf480bf4  r6  c3f6a084  r7  c42b2be7
05-13 02:36:24.578 12509 12509 F DEBUG   :     r8  bf480ac0  r9  00000025  r10 c338bd00  r11 bf480ad4
05-13 02:36:24.579 12509 12509 F DEBUG   :     ip  00000002  sp  bf480ab0  lr  c449545c  pc  ed5a0c1a
05-13 02:36:24.581 12509 12509 F DEBUG   : 
05-13 02:36:24.581 12509 12509 F DEBUG   : backtrace:
05-13 02:36:24.581 12509 12509 F DEBUG   :     #00 pc 0001ac1a  /system/lib/libc.so (strlen+38)
05-13 02:36:24.581 12509 12509 F DEBUG   :     #01 pc 0023f458  /data/app/launcher.powerkuy.growlauncher-cmiD-8fOAySFP8m93ngYZw==/lib/arm/libPowerKuy.so
05-13 02:40:48.906 12560 12878 F libc    : stack corruption detected (-fstack-protector)
05-13 02:40:53.714 12560 12878 F libc    : Fatal signal 6 (SIGABRT), code -6 (SI_TKILL) in tid 12878 (uy.growlauncher), pid 12560 (uy.growlauncher)
05-13 02:40:54.034 13304 13304 F DEBUG   : Softversion: PD1901_A_1.21.38
05-13 02:40:54.035 13304 13304 F DEBUG   : Time: 2024-05-13 02:40:54
05-13 02:40:54.035 13304 13304 F DEBUG   : *** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
05-13 02:40:54.036 13304 13304 F DEBUG   : Build fingerprint: 'vivo/PD1901/PD1901:9/P00610/compiler01071830:user/release-keys'
05-13 02:40:54.036 13304 13304 F DEBUG   : Revision: '0'
05-13 02:40:54.036 13304 13304 F DEBUG   : ABI: 'arm'
05-13 02:40:54.036 13304 13304 F DEBUG   : pid: 12560, tid: 12878, name: uy.growlauncher  >>> launcher.powerkuy.growlauncher <<<
05-13 02:40:54.036 13304 13304 F DEBUG   : signal 6 (SIGABRT), code -6 (SI_TKILL), fault addr --------
05-13 02:40:54.036 13304 13304 F DEBUG   : Abort message: 'stack corruption detected (-fstack-protector)'
05-13 02:40:54.036 13304 13304 F DEBUG   :     r0  00000000  r1  0000324e  r2  00000006  r3  00000008
05-13 02:40:54.036 13304 13304 F DEBUG   :     r4  00003110  r5  0000324e  r6  c05ff34c  r7  0000010c
05-13 02:40:54.037 13304 13304 F DEBUG   :     r8  ed631d8c  r9  ed631d8c  r10 b51b8804  r11 00f42400
05-13 02:40:54.037 13304 13304 F DEBUG   :     ip  c05ff2e8  sp  c05ff338  lr  ed5aae65  pc  ed5a2cb6
05-13 02:40:54.038 13304 13304 F DEBUG   : 
05-13 02:40:54.038 13304 13304 F DEBUG   : backtrace:
05-13 02:40:54.038 13304 13304 F DEBUG   :     #00 pc 0001ccb6  /system/lib/libc.so (abort+58)
05-13 15:48:49.726  5570  5570 E AndroidRuntime: FATAL EXCEPTION: main
[... cut ...]
05-14 14:43:56.223  5872  5872 E AndroidRuntime: 	at android.os.Looper.loop(Looper.java:224)
05-14 14:43:56.223  5872  5872 E AndroidRuntime: 	at android.app.ActivityThread.main(ActivityThread.java:7081)
05-14 14:43:56.223  5872  5872 E AndroidRuntime: 	at java.lang.reflect.Method.invoke(Native Method)
05-14 14:43:56.223  5872  5872 E AndroidRuntime: 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:536)
05-14 14:43:56.223  5872  5872 E AndroidRuntime: 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:928)

mrousavy, 2024-05-16:
```
05-13 15:48:49.726 5570 5570 E AndroidRuntime: java.lang.OutOfMemoryError: Failed to allocate a 514319528 byte allocation with 2097152 free bytes and 249MB until OOM, max allowed footprint 9429192, growth limit 268435456
```

well that's an out of memory error?

mrousavy, 2024-05-21:
Hey - just created a PR to temporarily fix this - can you test if that works for you? https://github.com/mrousavy/react-native-vision-camera/pull/2897 🙏

closed 2024-05-21 by mrousavy