margelo/react-native-vision-camera #2885: 🐛 Camera.onError(session/recoverable-error): An unknown error occurred while creating the Camera Session, but the Camera can recover from it.

brunosp-49, 2024-05-17:
### What's happening?

Hi guys, I'm having some problems to take pictures using react-native-vision-camera, the video works just fine, but when I try to take some picture, i just give an error "Camera.onError(session/recoverable-error): An unknown error occurred while creating the Camera Session, but the Camera can recover from it.",  I do not know why, I tried to add the property onInitialized, to just let the camera take photo when it is ready, but then it never send the asnwer, I'm trying to findo the solution for days, that is my code:

### Reproduceable Code

```tsx
import React, { FC, useEffect, useRef, useState } from 'react';
import { Modal, StyleSheet, View } from 'react-native';
import { Camera, useCameraDevice, useCameraPermission, useMicrophonePermission } from 'react-native-vision-camera';
import { VideoButton } from './components/videoButton';
import { PhotoButton } from './components/photoButton';
import { ModeSwitch } from './components/modeSwitch';
import { CloseButton } from './components/closeButton';
import VideoOrImage from '../inputFom/diplay/vieoOrImage';

interface Props {
    isVisible: boolean;
    onClose: () => void;
}

export const CameraComponent: FC<Props> = ({ isVisible, onClose }) => {
    const device = useCameraDevice('back');
    // eslint-disable-next-line @typescript-eslint/no-unused-vars
    const { hasPermission, requestPermission } = useCameraPermission();
    // eslint-disable-next-line @typescript-eslint/no-unused-vars
    const { hasPermission: hasMicPermission, requestPermission: requestMicPermission } = useMicrophonePermission();
    const [permission, setPermission] = useState<null | boolean>(null);
    const [cameraMode, setCameraMode] = useState<1 | 2>(1);
    const [isRecording, setIsRecording] = useState(false);
    const [preview, setPreview] = useState(false);
    const [uriPreview, setUriPreview] = useState<string | undefined>();
    // const [flashIsActive, setFlashIsActive] = useState(false);

    const cameraRef = useRef<Camera>(null);

    const startRecording = () => {
        if (!cameraRef.current || !device) { return; }
        setIsRecording(true);

        cameraRef.current.startRecording({
            videoCodec: 'h265',
            fileType: 'mp4',
            videoBitRate: 'normal',
            onRecordingFinished: (video) => {
                setUriPreview(video.path);
                setPreview(true);
            },
            onRecordingError: (error) => {
                console.log(error);
            },
        });
    };

    const stopRecording = async () => {
        if (!cameraRef.current || !device) { return; }
        await cameraRef.current.stopRecording();
        setIsRecording(false);
    };

    const takePhoto = async () => {
        if (!cameraRef.current || !device) { return; }
        const photo = await cameraRef.current.takePhoto();
        setUriPreview(photo.path);
        setPreview(true);
    };

    useEffect(() => {
        (async () => {
            const status = await requestPermission();
[... 133 lines cut ...]
  "hasTorch": false,
  "position": "back",
  "id": "0"
}
```

### Device

Pixel 6

### VisionCamera Version

4.0.3

### Can you reproduce this issue in the VisionCamera Example app?

No, I cannot reproduce the issue in the Example app

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

moi-xiey, 2024-05-17:
I am also facing this issue with XIaoMi MIX 2S(Android 10) with props video=true, with video=false it`s ok

mrousavy, 2024-05-21:
Try to set `photoQualityBalance="quality"` or `"speed"`

mrousavy, 2024-05-21:
Hey - just created a PR to temporarily fix this - can you test if that works for you? https://github.com/mrousavy/react-native-vision-camera/pull/2897 🙏

closed 2024-05-21 by mrousavy

brunosp-49, 2024-05-21:
Ok I'm gonna try, just a second

henriquedepaularodrigues, 2024-05-29:
> Try to set `photoQualityBalance="quality"` or `"speed"`

Works for me!

"react-native": "0.73.6",
"react-native-vision-camera": "^4.0.3"

```tsx
const myCameraScreen()=>{
// ... more code
  const devices = useCameraDevices();
  const device = devices.find(dvc => dvc.position === 'front');
  const format = useCameraFormat(device, [{ photoResolution: 'max' }]);
//... more code
return(
<Camera
          ref={cameraRef}
          onInitialized={() => console.log('Camera is ready!')}
          device={device}
          isActive={true}
          photoQualityBalance="quality"
          style={styles.camera}
          photo
        />
)
}
```