margelo/react-native-vision-camera #2891: 💭 Two separate frame processors at different target FPS

wuguishifu, 2024-05-20:
### Question

I have two separate native frame processors that I would like to run. However, I only want to run one of the frame processors at 15 hz (it's async and I don't need high temporal resolution), and the other one at 60-120 hz (not async, need the high temporal resolution but I've already optimized it so that it can run at 60-120 hz).

### What I tried

Currently, I've been keeping track of the total frame count in a shared value and just running the other frame processor every couple frames. I was wondering if there was a more optimal way to do it such as using the runAtTargetFps() in separate worklets.

This is my current implementation:

```ts
const format = useCameraFormat...

const frameCount = useSharedValue(0)

const frameProcessor = useFrameProcessor((frame: Frame) => {
    'worklet'
    frameCount.value++;
    if (frameCount.value % (Math.min((format?.maxFps ?? 30), 120) / 15) {
        lowHzFrameProcessor(frame)
    }

    highHzFrameProcessor(frame)
});
```

Note: on Android, I'm not calling `frame.image.close()` in the low frequency frame processor so that the image doesn't get garbage collected before the high frequency processor runs.

### VisionCamera Version

3.9.2

### Additional information

- [X] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22💭+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2024-05-21:
why not just

```ts
const frameProcessor = useFrameProcessor((frame) => {
  'worklet'
  runAtTargetFps(15, () => {
    lowHzFrameProcessor(frame)
  })

  // optional; either `runAsync`, or just call directly. whatever works for you
  runAsync(frame, () => {
    'worklet'
    highHzFrameProcessor(frame)
  })
})
```

?

wuguishifu, 2024-05-21:
🤦 Thanks! I think a combination of working late on a Sunday + brainfart made me not realize I could do that. I think I just misunderstood how the runAtTargetFps function worked. Appreciate the answer!

mrousavy, 2024-05-21:
haha no worries - glad it helped!

wuguishifu, 2024-05-25:
Sorry to bring this up again, but I'm still occasionally running into issues where the image is closed too early. Here's my current implementation:

```ts
const frameProcessor = useFrameProcessor((frame: Frame) => {
	'worklet'
	runAtTargetFps(5, () => {
		runAsync(frame, () => {
			'worklet'
			runSlowPlugin(frame);
		});
	});

	runFastPlugin(frame);
});
```

```kt
// slow plugin implementation
override fun callback(frame: Frame, params: MutableMap<String, Any>?): Boolean {
	val image = frame.image ?: return false

	runVeryLongProcess(image)

	image.close()
	return false
}

// fast plugin implementation
override fun callback(frame: Frame, params: MutableMap<String, Any>?): Double {
	val image = frame.image ?: return 0.0

	// do some processing
	val value = ...

	image.close()
	return val
}
```

I've seen the information about writing an asynchronous frame copy, but I'm not sure how to do it in kotlin. The documentation seems to indicate that I can do something like:

```kt
val frameCopy = Frame(
	frame.image,
	frame.timestamp,
	frame.orientation,
	frame.isMirrored
)

runVeryLongProcess(frameCopy.image)
```

But I'm still occasionally getting the "image is closed" issue. A temporary fix I had was to add a parameter to the fast plugin to prevent it from closing the image and then always closing it in the slow plugin. However, if the slow one happened to run very quickly then it would throw the error, and I had to keep track of the frame count to determine if I should try running the 5 fps frame plugin.

Is there any documentation on how to copy the image in kotlin without it being closed? Alternatively, is there a way to detect if the image is already closed and just return early instead of trying to continue the plugins?

mrousavy, 2024-05-27:
> ```kt
>  // I've been closing it in this plugin because it's 
>  // the plugin that always runs and for the most part 
>  // it seems to work fine
>  image.close()
> ```

Why? Why are you closing the Frame? The documentation tells you that you should not close the Frame, VisionCamera handles that for you.

closed 2024-05-27 by mrousavy

mrousavy, 2024-05-28:
Again, you should not close the Frame.
That's why there's no documentation - you should not close it.
It is closed automatically when all runAsync and frameProcessor calls have completed.

If you intend to keep it longer than a runAsync call (eg when it goes out of scope in your native plugin) call incrementRefCount() first, and then decrementRefCount() after you're done.