margelo/react-native-vision-camera #2893: APK Size increase After implementing react-native-vision-camera lib double the apk size

vinod1988, 2024-05-21:
### What's happening?

I was using **camera kit** lib but due to some issue i switched to **react-native-vision-camera** and my apk size increase double.

### Reproduceable Code

```tsx
<Camera
                style={{ flex: 1 }}
                device={device}
                isActive={true}
                focusable={true}
                torch={torchMode ? "on" : "off"}
                enableZoomGesture={true}
                zoom={zoomLevel}
                codeScanner={codeScanner}
              />
```

### Relevant log output

```shell
above code is working fine but when exporting the apk my apk size double the earlier size
```

### Camera Device

```json
<Camera
                style={{ flex: 1 }}
                device={device}
                isActive={true}
                focusable={true}
                torch={torchMode ? "on" : "off"}
                enableZoomGesture={true}
                zoom={zoomLevel}
                codeScanner={codeScanner}
              />
```

### Device

Oppo android version 13

### VisionCamera Version

4.0.3

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

Farhankhaliid, 2024-05-21:
Same APK  size increased to  double.

PerovArtemWork, 2024-05-24:
@mrousavy  Before update  - 39.1 MB ,after 76.7 MB.  "react-native-vision-camera": "4.0.5"

pierroo, 2025-01-15:
> Did you disable the CodeScanner?
> 
> Maybe that's what's increasing the storage.
> 
> Can you try to remove the MLKit CodeScanner dependency from VisionCamera's build.gradle, then also removing it from everywhere in the code where it shows errors?
> 
> Would like to see the sizes after that.

I would be interested in that as well.
Is there a way to deactivate MLKit CodeScanner without touching files inside node_modules that would be erased on every npm reinstall?