margelo/react-native-vision-camera #2895: 🐛 `finally` on a `Promise` stops camera on Android

bglgwyng, 2024-05-21:
### What's happening?

The following reproducible example causes the camera to stop.
It only happens when you attach a `finally` chain to a `Promise`, not `then` or `catch`.

The real-world example includes the usage of await for run-on-JS function return.
Also this problem only exists on Android.

I used react-native-worklets-core 1.3.3.

### Reproduceable Code

```tsx
const frameProcessor = useFrameProcessor(
  (frame) => {
    "worklet";
    Promise.resolve().finally(() => {});
    Promise.resolve().then(() => {}); // it's ok
    Promise.resolve().catch(() => {}); // it's ok
  },
[])
```

### Relevant log output

```shell
no relevant log output
```

### Camera Device

```json
{
  "formats": [],
  "sensorOrientation": "landscape-left",
  "hardwareLevel": "full",
  "maxZoom": 10,
  "minZoom": 0.6000000238418579,
  "maxExposure": 20,
  "supportsLowLightBoost": false,
  "neutralZoom": 1,
  "physicalDevices": [
    "wide-angle-camera",
    "ultra-wide-angle-camera",
    "wide-angle-camera",
    "telephoto-camera"
  ],
  "supportsFocus": true,
  "supportsRawCapture": false,
  "isMultiCam": true,
  "minFocusDistance": 7.000000066757203,
  "minExposure": -20,
  "name": "0 (BACK) androidx.camera.camera2",
  "hasFlash": true,
  "hasTorch": true,
  "position": "back",
  "id": "0"
}
```

### Device

iPhone 13 (iOS 17.5)

### VisionCamera Version

4.0.4

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [X] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2024-05-21:
This is an issue with Worklets. Please create an issue in react-native-worklets-core.

closed 2024-05-21 by mrousavy

bglgwyng, 2024-05-23:
@mrousavy  I tried to reproduce the behavior with react-native-worklets-core only but nothing happened.
```typescript
Worklets.defaultContext
  .runAsync(() => {
    'worklet'
    Promise.resolve().finally(() => {})
  })
  ```

The behavior of VisionCamera with `finally` is stopping the camera preview.
Which behavior do you expect from this code?

mrousavy, 2024-05-27:
We also saw a similar issue on a separate Worklet context (in our upcoming 3d library), so this is def. an issue with Worklets, just maybe not the `defaultContext`. Try to create a new separate context.