margelo/react-native-vision-camera #2899: 🐛 Android: maxImages (6) has already been acquired

subskill-anezzar, 2024-05-22:
### What's happening?

Hello 👋

I am using  VisionCamera with frame processors enabled with @ismaelmoreiraa/vision-camera-ocr:**3.0.1**.
It's a simple code that apply a Regex Validation out of a frame trough the FrameProcessor.

It's work perfectly in local (**Expo 50**) but when I build and use on Android, the application crash with the trace below.
I saw this issue that was previously fixed apparently (it could be related maybe) -> https://github.com/mrousavy/react-native-vision-camera/pull/2352/commits/42ae1796ed63b7deb754470bff6c6e2e4e1e5df5

I've try a lot of things, like tweaking the version and all, before coming here and hope get some help about this.

I certainly did something wrong since I didn't saw similar new issues, any idea ?

Thanks in advance !

### Reproduceable Code

```tsx
export default function ScanScreen() {
    const device = useCameraDevice('back');
    // ... 
    const ocr = useSharedValue({})
    const processing = useSharedValue(false)
    const parseOcr = useRunOnJS(() => {
        let matchFound = false;
        const scannedBlocks = ocr.value.result?.blocks;
        if (scannedBlocks?.length > 0) {
            scannedBlocks.forEach((block) => {
                block.lines.forEach((line) => {
                  // REGEX CHECK
                });
            });
        }

        if (!matchFound) {
            processing.value = false;
        }
    }, [])

    /** Process camera frames */
    const frameProcessor = useFrameProcessor(
        (frame) => {
            'worklet'
            if (processing.value === false) {
                processing.value = true;
                runAsync(frame, () => {
                    'worklet'
                    runAtTargetFps(10, () => {
                        'worklet'
                        const image = scanOCR(frame);
                        if (image) {
                            try {
                                ocr.value = image;
                                parseOcr();
                            } finally {
                                processing.value = false;
                            }
                        } else {
                            processing.value = false;
                        }
                    })
                });
            }
        },
        [],
    )

    const permissionFunction = async () => {
        ExpoCam.getCameraPermissionsAsync().then((res) => {
                if (!res.canAskAgain) {
                    Linking.openSettings();
                } else {
                    getPermission()
                }
            }
        )
    }
    
    return (

        <Camera
            style={{height: Dimensions.get('window').height + (Platform.OS !== "ios" ? StatusBar.currentHeight : 0)}}
            frameProcessor={frameProcessor}
            device={device}
            pixelFormat={"yuv"}
            isActive={true}
            photo={true}
            format={format}
        >
        </Camera>
    );
};
```

### Relevant log output

```shell
Failed to acquire image.
java.lang.IllegalStateException: maxImages (6) has already been acquired, call #close before acquiring more.
	at android.media.ImageReader.acquireNextImage(ImageReader.java:560)
	at androidx.camera.core.AndroidImageReaderProxy.acquireNextImage(AndroidImageReaderProxy.java:88)
	at androidx.camera.core.SafeCloseImageReaderProxy.acquireNextImage(SafeCloseImageReaderProxy.java:88)
	at androidx.camera.core.ImageAnalysisBlockingAnalyzer.acquireImage(ImageAnalysisBlockingAnalyzer.java:41)
	at androidx.camera.core.ImageAnalysisAbstractAnalyzer.onImageAvailable(ImageAnalysisAbstractAnalyzer.java:128)
	at androidx.camera.core.SafeCloseImageReaderProxy.lambda$setOnImageAvailableListener$1$androidx-camera-core-SafeCloseImageReaderProxy(SafeCloseImageReaderProxy.java:213)
	at androidx.camera.core.SafeCloseImageReaderProxy$$ExternalSyntheticLambda1.onImageAvailable(Unknown Source:4)
	at androidx.camera.core.AndroidImageReaderProxy.lambda$setOnImageAvailableListener$0$androidx-camera-core-AndroidImageReaderProxy(AndroidImageReaderProxy.java:169)
	at androidx.camera.core.AndroidImageReaderProxy$$ExternalSyntheticLambda0.run(Unknown Source:4)
```

### Camera Device

```json
{
  "formats": [],
  "sensorOrientation": "landscape-left",
  "hardwareLevel": "limited",
  "maxZoom": 4,
  "minZoom": 1,
  "maxExposure": 20,
  "supportsLowLightBoost": false,
  "neutralZoom": 1,
  "physicalDevices": [
    "wide-angle-camera"
  ],
  "supportsFocus": true,
  "supportsRawCapture": false,
  "isMultiCam": false,
  "minFocusDistance": 0,
  "minExposure": -20,
  "name": "0 (BACK) androidx.camera.camera2",
  "hasFlash": true,
  "hasTorch": true,
  "position": "back",
  "id": "0"
}
```

### Device

Galaxy A12

### VisionCamera Version

4.0.5

### Can you reproduce this issue in the VisionCamera Example app?

No, I cannot reproduce the issue in the Example app

### Additional information

- [X] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

subskill-anezzar, 2024-05-27:
Finally, I did the **STRATEGY_KEEP_ONLY_LATEST** (https://github.com/mrousavy/react-native-vision-camera/issues/2826) modification, since I first wanted to prevent lags while using scanOCR of `vision-camera-ocr`. I still don't know what caused `runAsync` to create the Release crash (see above).

Anyway, I hope getting around this without patching the package with that last modification since it is apparently not a good idea but it's all I got :)

closed 2024-05-27 by subskill-anezzar

mrEuler, 2024-08-07:
I am having same error, even though I am using runAsync inside the useSkiaFrameProcessor. On debug build works fine, but when comming to prod build, 9 out of 10 runs the app crashes with the logs same as in original post ^^^ . @subskill-anezzar did you solve this maybe somehow?