margelo/react-native-vision-camera #2901: 🔧 Not able to build expo dev client when using any of the given frame processing plugin for OCR

Swikriti0998, 2024-05-22:
### How were you trying to build the app?

I tried adding an OCR and a barcode scanner frame processor plugin, the barcode is working absolutely fine but while trying to build the dev client build for the OCR plugin it's giving error.

### Full build logs

```tsx
[RUN_FASTLANE] › Packaging react-native-worklets-core Pods/react-native-worklets-core » libreact-native-worklets-core.a
[RUN_FASTLANE] › Packaging react-native Pods/React-CoreModules » libReact-CoreModules.a
[RUN_FASTLANE] › Packaging react-native-vision-camera Pods/VisionCamera » libVisionCamera.a
[RUN_FASTLANE] › Packaging react-native-vision-camera Pods/VisionCamera » libVisionCamera.a
[RUN_FASTLANE] › Executing react-native-vision-camera Pods/VisionCamera » Copy generated compatibility header
[RUN_FASTLANE] › Compiling react-native-vision-camera-v3-text-recognition Pods/react-native-vision-camera-v3-text-recognition » react-native-vision-camera-v3-text-recognition-dummy.m
[RUN_FASTLANE] › Compiling react-native-vision-camera-v3-text-recognition Pods/react-native-vision-camera-v3-text-recognition » VisionCameraV3TextRecognition.m
[RUN_FASTLANE] 
❌  (node_modules/react-native-vision-camera-v3-text-recognition/ios/VisionCameraV3TextRecognition.m:9:9)

   7 | #import <VisionCamera/FrameProcessorPlugin.h>
   8 | #import <VisionCamera/FrameProcessorPluginRegistry.h>
>  9 | #import <VisionCameraProxy.h>
     |         ^ 'VisionCameraProxy.h' file not found with <angled> include; use "quotes" instead
  10 | #import <VisionCamera/Frame.h>
  11 | @import MLKitVision;
  12 | @interface VisionCameraTextRecognitionV3Plugin : FrameProcessorPlugin

[RUN_FASTLANE] 
❌  (node_modules/react-native-vision-camera/ios/FrameProcessors/VisionCameraProxy.h:12:2)

  10 | 
  11 | #ifndef __cplusplus
> 12 | #error VisionCameraProxy.h has to be compiled with C++!
     |  ^ VisionCameraProxy.h has to be compiled with C++!
  13 | #endif
  14 | 
  15 | #import <Foundation/Foundation.h>

[RUN_FASTLANE] 
❌  (node_modules/react-native-worklets-core/cpp/dispatch/WKTDispatchQueue.h:3:10)

  1 | #pragma once
  2 | 
> 3 | #include <condition_variable>
    |          ^ 'condition_variable' file not found
  4 | #include <cstdint>
  5 | #include <cstdio>
  6 | #include <functional>

[RUN_FASTLANE] › Compiling react-native-vision-camera-v3-text-recognition Pods/react-native-vision-camera-v3-text-recognition » VisionCameraV3TextRecognition.m
[RUN_FASTLANE] 
❌  (node_modules/react-native-vision-camera-v3-text-recognition/ios/VisionCameraV3TextRecognition.m:9:9)

   7 | #import <VisionCamera/FrameProcessorPlugin.h>
   8 | #import <VisionCamera/FrameProcessorPluginRegistry.h>
>  9 | #import <VisionCameraProxy.h>
     |         ^ 'VisionCameraProxy.h' file not found with <angled> include; use "quotes" instead
  10 | #import <VisionCamera/Frame.h>
  11 | @import MLKitVision;
  12 | @interface VisionCameraTextRecognitionV3Plugin : FrameProcessorPlugin

[RUN_FASTLANE] 
❌  (node_modules/react-native-vision-camera/ios/FrameProcessors/VisionCameraProxy.h:12:2)

  10 | 
  11 | #ifndef __cplusplus
> 12 | #error VisionCameraProxy.h has to be compiled with C++!
     |  ^ VisionCameraProxy.h has to be compiled with C++!
  13 | #endif
  14 | 
  15 | #import <Foundation/Foundation.h>

[... 71 lines cut ...]
```

### VisionCamera Version

4.0.1

### Target platforms

iOS

### Operating system

MacOS

### Can you build the VisionCamera Example app?

No, I cannot build the Example app either

### Additional information

- [X] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

tfcornerstone, 2024-05-22:
@Swikriti0998 That is an issue with the frame processor plugin. Mention this problem in the issues of that project.

mrousavy, 2024-05-27:
Yea I'm sorry about that. The header path should be `<VisionCamera/VisionCameraProxyHolder.h>` now - can you update that and send them a PR? Thank you.

closed 2024-05-27 by mrousavy

neha-s999, 2024-10-30:
I am still facing this error
❌ Pods/react-native-vision-camera-v3-text-recognition: 'VisionCameraProxy.h' file not found with include; use "quotes" instead
└─/Users/tribalscale/script-runner-driver-mobile/node_modules/react-native-vision-camera-v3-text-recognition/ios/VisionCameraV3TextRecognition.m:9:9
❌ Pods/react-native-vision-camera-v3-text-recognition: 'condition_variable' file not found
└─/Users/tribalscale/script-runner-driver-mobile/node_modules/react-native-worklets-core/cpp/dispatch/WKTDispatchQueue.h:3:10
› Executing expo-modules-core Pods/ExpoModulesCore » Copy generated compatibility header

This is my package.json
{
"name": "scriptrunner-driver-mobile",
"main": "expo-router/entry",
"version": "1.0.0",
"scripts": {
"start": "expo start",
"reset-project": "node ./scripts/reset-project.js",
"android": "expo run:android",
"ios": "expo run:ios",
"web": "expo start --web",
"test": "jest --watchAll",
"lint": "eslint .",
"prepare": "husky"
},
"jest": {
"preset": "jest-expo"
},
"dependencies": {
"@aws-amplify/react-native": "^1.1.6",
"@aws-amplify/ui-react-native": "^2.2.14",
"@expo/vector-icons": "^14.0.2",
"@react-native-async-storage/async-storage": "^2.0.0",
"@react-native-community/geolocation": "^3.4.0",
"@react-native-community/netinfo": "^11.4.1",
"@react-navigation/native": "^6.1.18",
"aws-amplify": "^6.6.4",
"axios": "^1.7.7",
"expo": "~51.0.28",
"expo-camera": "^15.0.16",
"expo-constants": "~16.0.2",
"expo-font": "~12.0.9",
"expo-image": "^1.12.15",
"expo-image-manipulator": "^12.0.5",
"expo-linking": "~6.3.1",
"expo-location": "^17.0.1",
"expo-router": "~3.5.23",
"expo-splash-screen": "~0.27.5",
"expo-status-bar": "~1.12.1",
"expo-system-ui": "~3.0.7",
"expo-web-browser": "~13.0.3",
"react": "18.2.0",
"react-dom": "18.2.0",
"react-native": "0.74.5",
"react-native-config": "^1.5.3",
"react-native-error-boundary": "^1.2.5",
"react-native-geolocation-service": "^5.3.1",
"react-native-gesture-handler": "~2.16.1",
"react-native-get-random-values": "^1.11.0",
"react-native-image-picker": "^7.1.2",
"react-native-keyboard-aware-scroll-view": "^0.9.5",
"react-native-maps": "^1.18.0",
"react-native-paper": "^5.12.5",
"react-native-permissions": "^4.1.5",
"react-native-reanimated": "~3.10.1",
"react-native-safe-area-context": "^4.10.5",
"react-native-screens": "3.31.1",
"react-native-signature-capture": "^0.4.12",
"react-native-svg": "^15.8.0",
"react-native-svg-transformer": "^1.5.0",
"react-native-vision-camera": "4.0.5",
"react-native-vision-camera-v3-text-recognition": "^1.1.1",
"react-native-web": "~0.19.10",
"react-native-webview": "^13.12.2",
"react-native-worklets-core": "^1.3.3"
},
"devDependencies": {
"@babel/core": "^7.20.0",
"@types/jest": "^29.5.13",
"@types/react": "~18.2.45",
"@types/react-native": "^0.72.8",
"@types/react-native-maps": "^0.24.2",
"@types/react-test-renderer": "^18.0.7",
"@typescript-eslint/eslint-plugin": "^8.7.0",
"@typescript-eslint/parser": "^8.7.0",
"eslint": "^8.57.1",
"eslint-config-expo": "^7.1.2",
"husky": "^9.1.6",
"jest": "^29.7.0",
"jest-expo": "~51.0.3",
"lint-staged": "^15.2.10",
"react-test-renderer": "18.2.0",
"ts-jest": "^29.2.5",
"typescript": "~5.3.3"
},
"lint-staged": {
"*.{js,jsx,ts,tsx}": [
"npm run lint"
]
},
"private": true
}