margelo/react-native-vision-camera #2903: 🔧 Execution failed for task ':react-native-vision-camera:compileReleaseKotlin'.

flessard, 2024-05-22:
### How were you trying to build the app?

I try to build my app with expo command : eas build --profile production --platform android --local --clear-cache but i got compile error : 

` FAILURE: Build failed with an exception.
[RUN_GRADLEW] * What went wrong:
[RUN_GRADLEW] Execution failed for task ':react-native-vision-camera:compileReleaseKotlin'.
[RUN_GRADLEW] > A failure occurred while executing org.jetbrains.kotlin.compilerRunner.GradleCompilerRunnerWithWorkers$GradleKotlinCompilerWorkAction
[RUN_GRADLEW]    > Compilation error. See log for more details
[RUN_GRADLEW] * Try:
[RUN_GRADLEW] > Run with --stacktrace option to get the stack trace.
[RUN_GRADLEW] > Run with --info or --debug option to get more log output.
`

### Full build logs

```tsx
[RUN_GRADLEW] w: file:///private/var/folders/tf/mn7y87ks0j112k2l95f0_0140000gn/T/eas-build-local-nodejs/1daa6cc1-e676-41a9-9233-5e21677c9c88/build/node_modules/@rnmapbox/maps/android/src/main/mapbox-v11-compat/v10/com/rnmapbox/rnmbx/v11compat/StyleFactory.kt:199:36 Parameter 'expression' is never used
[RUN_GRADLEW] w: file:///private/var/folders/tf/mn7y87ks0j112k2l95f0_0140000gn/T/eas-build-local-nodejs/1daa6cc1-e676-41a9-9233-5e21677c9c88/build/node_modules/@rnmapbox/maps/android/src/main/mapbox-v11-compat/v10/com/rnmapbox/rnmbx/v11compat/StyleFactory.kt:203:14 This API is experimental. It may be changed in the future without notice.
[RUN_GRADLEW] w: file:///private/var/folders/tf/mn7y87ks0j112k2l95f0_0140000gn/T/eas-build-local-nodejs/1daa6cc1-e676-41a9-9233-5e21677c9c88/build/node_modules/@rnmapbox/maps/android/src/main/mapbox-v11-compat/v10/com/rnmapbox/rnmbx/v11compat/StyleFactory.kt:203:52 Parameter 'transition' is never used
[RUN_GRADLEW] w: file:///private/var/folders/tf/mn7y87ks0j112k2l95f0_0140000gn/T/eas-build-local-nodejs/1daa6cc1-e676-41a9-9233-5e21677c9c88/build/node_modules/@rnmapbox/maps/android/src/main/mapbox-v11-compat/v10/com/rnmapbox/rnmbx/v11compat/StyleFactory.kt:207:14 This API is experimental. It may be changed in the future without notice.
[RUN_GRADLEW] w: file:///private/var/folders/tf/mn7y87ks0j112k2l95f0_0140000gn/T/eas-build-local-nodejs/1daa6cc1-e676-41a9-9233-5e21677c9c88/build/node_modules/@rnmapbox/maps/android/src/main/mapbox-v11-compat/v10/com/rnmapbox/rnmbx/v11compat/StyleFactory.kt:207:42 Parameter 'expression' is never used
[RUN_GRADLEW] w: file:///private/var/folders/tf/mn7y87ks0j112k2l95f0_0140000gn/T/eas-build-local-nodejs/1daa6cc1-e676-41a9-9233-5e21677c9c88/build/node_modules/@rnmapbox/maps/android/src/main/mapbox-v11-compat/v10/com/rnmapbox/rnmbx/v11compat/StyleFactory.kt:211:14 This API is experimental. It may be changed in the future without notice.
[RUN_GRADLEW] w: file:///private/var/folders/tf/mn7y87ks0j112k2l95f0_0140000gn/T/eas-build-local-nodejs/1daa6cc1-e676-41a9-9233-5e21677c9c88/build/node_modules/@rnmapbox/maps/android/src/main/mapbox-v11-compat/v10/com/rnmapbox/rnmbx/v11compat/StyleFactory.kt:211:42 Parameter 'expression' is never used
[RUN_GRADLEW] w: file:///private/var/folders/tf/mn7y87ks0j112k2l95f0_0140000gn/T/eas-build-local-nodejs/1daa6cc1-e676-41a9-9233-5e21677c9c88/build/node_modules/@rnmapbox/maps/android/src/main/mapbox-v11-compat/v10/com/rnmapbox/rnmbx/v11compat/StyleFactory.kt:215:14 This API is experimental. It may be changed in the future without notice.
[RUN_GRADLEW] w: file:///private/var/folders/tf/mn7y87ks0j112k2l95f0_0140000gn/T/eas-build-local-nodejs/1daa6cc1-e676-41a9-9233-5e21677c9c88/build/node_modules/@rnmapbox/maps/android/src/main/mapbox-v11-compat/v10/com/rnmapbox/rnmbx/v11compat/StyleFactory.kt:215:48 Parameter 'transition' is never used
[RUN_GRADLEW] w: file:///private/var/folders/tf/mn7y87ks0j112k2l95f0_0140000gn/T/eas-build-local-nodejs/1daa6cc1-e676-41a9-9233-5e21677c9c88/build/node_modules/@rnmapbox/maps/android/src/main/mapbox-v11-compat/v10/com/rnmapbox/rnmbx/v11compat/StyleFactory.kt:219:14 This API is experimental. It may be changed in the future without notice.
[RUN_GRADLEW] w: file:///private/var/folders/tf/mn7y87ks0j112k2l95f0_0140000gn/T/eas-build-local-nodejs/1daa6cc1-e676-41a9-9233-5e21677c9c88/build/node_modules/@rnmapbox/maps/android/src/main/mapbox-v11-compat/v10/com/rnmapbox/rnmbx/v11compat/StyleFactory.kt:219:38 Parameter 'expression' is never used
[RUN_GRADLEW] w: file:///private/var/folders/tf/mn7y87ks0j112k2l95f0_0140000gn/T/eas-build-local-nodejs/1daa6cc1-e676-41a9-9233-5e21677c9c88/build/node_modules/@rnmapbox/maps/android/src/main/mapbox-v11-compat/v10/com/rnmapbox/rnmbx/v11compat/StyleFactory.kt:223:53 Parameter 'transition' is never used
[RUN_GRADLEW] w: file:///private/var/folders/tf/mn7y87ks0j112k2l95f0_0140000gn/T/eas-build-local-nodejs/1daa6cc1-e676-41a9-9233-5e21677c9c88/build/node_modules/@rnmapbox/maps/android/src/main/mapbox-v11-compat/v10/com/rnmapbox/rnmbx/v11compat/StyleFactory.kt:227:43 Parameter 'expression' is never used
[RUN_GRADLEW] w: file:///private/var/folders/tf/mn7y87ks0j112k2l95f0_0140000gn/T/eas-build-local-nodejs/1daa6cc1-e676-41a9-9233-5e21677c9c88/build/node_modules/@rnmapbox/maps/android/src/main/mapbox-v11-compat/v10/com/rnmapbox/rnmbx/v11compat/StyleFactory.kt:231:43 Parameter 'expression' is never used
[RUN_GRADLEW] w: file:///private/var/folders/tf/mn7y87ks0j112k2l95f0_0140000gn/T/eas-build-local-nodejs/1daa6cc1-e676-41a9-9233-5e21677c9c88/build/node_modules/@rnmapbox/maps/android/src/main/mapbox-v11-compat/v10/com/rnmapbox/rnmbx/v11compat/StyleFactory.kt:235:51 Parameter 'transition' is never used
[RUN_GRADLEW] w: file:///private/var/folders/tf/mn7y87ks0j112k2l95f0_0140000gn/T/eas-build-local-nodejs/1daa6cc1-e676-41a9-9233-5e21677c9c88/build/node_modules/@rnmapbox/maps/android/src/main/mapbox-v11-compat/v10/com/rnmapbox/rnmbx/v11compat/StyleFactory.kt:239:41 Parameter 'expression' is never used
[RUN_GRADLEW] w: file:///private/var/folders/tf/mn7y87ks0j112k2l95f0_0140000gn/T/eas-build-local-nodejs/1daa6cc1-e676-41a9-9233-5e21677c9c88/build/node_modules/@rnmapbox/maps/android/src/main/mapbox-v11-compat/v10/com/rnmapbox/rnmbx/v11compat/StyleFactory.kt:243:41 Parameter 'expression' is never used
[RUN_GRADLEW] w: file:///private/var/folders/tf/mn7y87ks0j112k2l95f0_0140000gn/T/eas-build-local-nodejs/1daa6cc1-e676-41a9-9233-5e21677c9c88/build/node_modules/@rnmapbox/maps/android/src/main/mapbox-v11-compat/v10/com/rnmapbox/rnmbx/v11compat/StyleFactory.kt:247:38 Parameter 'rasterColor' is never used
[RUN_GRADLEW] w: file:///private/var/folders/tf/mn7y87ks0j112k2l95f0_0140000gn/T/eas-build-local-nodejs/1daa6cc1-e676-41a9-9233-5e21677c9c88/build/node_modules/@rnmapbox/maps/android/src/main/mapbox-v11-compat/v10/com/rnmapbox/rnmbx/v11compat/StyleFactory.kt:251:62 Parameter 'expression' is never used
[RUN_GRADLEW] w: file:///private/var/folders/tf/mn7y87ks0j112k2l95f0_0140000gn/T/eas-build-local-nodejs/1daa6cc1-e676-41a9-9233-5e21677c9c88/build/node_modules/@rnmapbox/maps/android/src/main/mapbox-v11-compat/v10/com/rnmapbox/rnmbx/v11compat/StyleFactory.kt:255:62 Parameter 'expression' is never used
[RUN_GRADLEW] w: file:///private/var/folders/tf/mn7y87ks0j112k2l95f0_0140000gn/T/eas-build-local-nodejs/1daa6cc1-e676-41a9-9233-5e21677c9c88/build/node_modules/@rnmapbox/maps/android/src/main/mapbox-v11-compat/v10/com/rnmapbox/rnmbx/v11compat/StyleFactory.kt:259:70 Parameter 'transition' is never used
[RUN_GRADLEW] w: file:///private/var/folders/tf/mn7y87ks0j112k2l95f0_0140000gn/T/eas-build-local-nodejs/1daa6cc1-e676-41a9-9233-5e21677c9c88/build/node_modules/@rnmapbox/maps/android/src/main/mapbox-v11-compat/v10/com/rnmapbox/rnmbx/v11compat/StyleFactory.kt:263:66 Parameter 'expression' is never used
[RUN_GRADLEW] w: file:///private/var/folders/tf/mn7y87ks0j112k2l95f0_0140000gn/T/eas-build-local-nodejs/1daa6cc1-e676-41a9-9233-5e21677c9c88/build/node_modules/@rnmapbox/maps/android/src/main/mapbox-v11-compat/v10/com/rnmapbox/rnmbx/v11compat/StyleFactory.kt:267:74 Parameter 'expression' is never used
[RUN_GRADLEW] w: file:///private/var/folders/tf/mn7y87ks0j112k2l95f0_0140000gn/T/eas-build-local-nodejs/1daa6cc1-e676-41a9-9233-5e21677c9c88/build/node_modules/@rnmapbox/maps/android/src/main/mapbox-v11-compat/v10/com/rnmapbox/rnmbx/v11compat/StyleFactory.kt:271:90 Parameter 'transition' is never used
[RUN_GRADLEW] w: file:///private/var/folders/tf/mn7y87ks0j112k2l95f0_0140000gn/T/eas-build-local-nodejs/1daa6cc1-e676-41a9-9233-5e21677c9c88/build/node_modules/@rnmapbox/maps/android/src/main/mapbox-v11-compat/v10/com/rnmapbox/rnmbx/v11compat/StyleFactory.kt:275:73 Parameter 'expr [... cut ...]
### Additional information

- [X] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

coderjinhui, 2024-05-22:
same with me!!!!! but the task is :react-native-vision-camera:compileDebugKotlin FAILED

coderjinhui, 2024-05-22:
I think the document should update the version limitation. As I find the old issue form #2783 
You can see this library currently is not supported when RN > 0.74 !!!
Fix with downgrade to 0.73.
Really missing the important information in documentation!

flessard, 2024-05-22:
If i downgrade to  0.73, work fine...Thanks

closed 2024-05-22 by flessard

cyrusyan021, 2025-08-28:
I'm still getting this issue with react-native@0.77.0, is there any workarounds ?