margelo/react-native-vision-camera #2904: 🔧

ZhenQian-keystone, 2024-05-23:
### How were you trying to build the app?

i am trying to build android,  i am not use camera v4 , but i still meet same this error. this error confuse me a lot of time , who can save my life. 

this is error 
```
> Task :react-native-vision-camera:compileReleaseKotlin FAILED
e: file:///Users/jianjianjianjian/Documents/fuck_rn_fuck_rn/keystone-hot/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/frameprocessor/VisionCameraProxy.kt:31:79 This API is provided only for React Native frameworks and not intended for general users. This API can change between minor versions in alignment with React Native frameworks and won't be considered a breaking change.
e: file:///Users/jianjianjianjian/Documents/fuck_rn_fuck_rn/keystone-hot/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/frameprocessor/VisionCameraProxy.kt:36:19 This API is provided only for React Native frameworks and not intended for general users. This API can change between minor versions in alignment with React Native frameworks and won't be considered a breaking change.
e: file:///Users/jianjianjianjian/Documents/fuck_rn_fuck_rn/keystone-hot/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/frameprocessor/VisionCameraProxy.kt:36:47 This API is provided only for React Native frameworks and not intended for general users. This API can change between minor versions in alignment with React Native frameworks and won't be considered a breaking change.
e: file:///Users/jianjianjianjian/Documents/fuck_rn_fuck_rn/keystone-hot/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/frameprocessor/VisionCameraProxy.kt:72:73 This API is provided only for React Native frameworks and not intended for general users. This API can change between minor versions in alignment with React Native frameworks and won't be considered a breaking change.

```

i use camera v3
```
    "react-native-vision-camera": "^3.3.1",
```

here is my computer env

```
System:
    OS: macOS 14.4.1
    CPU: (12) arm64 Apple M2 Pro
    Memory: 75.92 MB / 16.00 GB
    Shell: 5.9 - /bin/zsh
  Binaries:
    Node: 18.20.2 - /usr/local/bin/node
    Yarn: 1.22.22 - /opt/homebrew/bin/yarn
    npm: 10.5.0 - /usr/local/bin/npm
    Watchman: 2024.05.06.00 - /opt/homebrew/bin/watchman
  Managers:
    CocoaPods: 1.15.2 - /opt/homebrew/bin/pod
  SDKs:
    iOS SDK:
      Platforms: DriverKit 23.5, iOS 17.5, macOS 14.5, tvOS 17.5, visionOS 1.2, watchOS 10.5
    Android SDK: Not Found
  IDEs:
    Android Studio: 2023.3 AI-233.14808.21.2331.11709847
    Xcode: 15.4/15F31d - /usr/bin/xcodebuild
  Languages:
    Java: 11.0.23 - /Users/jianjianjianjian/.jenv/shims/javac
  npmPackages:
    @react-native-community/cli: Not Found
    react: 18.2.0 => 18.2.0 
    react-native: 0.71.11 => 0.71.11 
    react-native-macos: Not Found
  npmGlobalPackages:
    *react-native*: Not Found
```

### Full build logs

```tsx
> Task :react-native-safe-area-context:compileReleaseKotlin
w: file:///Users/jianjianjianjian/Documents/fuck_rn_fuck_rn/keystone-hot/node_modules/react-native-safe-area-context/android/src/main/java/com/th3rdwave/safeareacontext/SafeAreaView.kt:50:23 'getter for uiImplementation: UIImplementation!' is deprecated. Deprecated in Java

> Task :react-native-vision-camera:compileReleaseKotlin FAILED
e: file:///Users/jianjianjianjian/Documents/fuck_rn_fuck_rn/keystone-hot/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/frameprocessor/VisionCameraProxy.kt:31:79 This API is provided only for React Native frameworks and not intended for general users. This API can change between minor versions in alignment with React Native frameworks and won't be considered a breaking change.
e: file:///Users/jianjianjianjian/Documents/fuck_rn_fuck_rn/keystone-hot/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/frameprocessor/VisionCameraProxy.kt:36:19 This API is provided only for React Native frameworks and not intended for general users. This API can change between minor versions in alignment with React Native frameworks and won't be considered a breaking change.
e: file:///Users/jianjianjianjian/Documents/fuck_rn_fuck_rn/keystone-hot/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/frameprocessor/VisionCameraProxy.kt:36:47 This API is provided only for React Native frameworks and not intended for general users. This API can change between minor versions in alignment with React Native frameworks and won't be considered a breaking change.
e: file:///Users/jianjianjianjian/Documents/fuck_rn_fuck_rn/keystone-hot/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/frameprocessor/VisionCameraProxy.kt:72:73 This API is provided only for React Native frameworks and not intended for general users. This API can change between minor versions in alignment with React Native frameworks and won't be considered a breaking change.

> Task :react-native-vision-camera:configureCMakeRelWithDebInfo[arm64-v8a]
C/C++: CMake Warning at /Users/jianjianjianjian/Library/Android/sdk/ndk/24.0.8215888/build/cmake/android-legacy.toolchain.cmake:415 (message):
C/C++:   An old version of CMake is being used that cannot automatically detect
C/C++:   compiler attributes.  Compiler identification is being bypassed.  Some
C/C++:   values may be wrong or missing.  Update to CMake 3.19 or newer to use
C/C++:   CMake's built-in compiler identification.
[... 172 lines cut ...]
```

### VisionCamera Version

react-native-vision-camera@^3.3.1:   version "3.9.2"

### Target platforms

Android

### Operating system

MacOS

### Can you build the VisionCamera Example app?

Yes, I can successfully build the Example app here

### Additional information

- [X] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

RiusmaX, 2024-05-23:
Same here with react-native-vision-camera v4.0.5 and RN 0.74.0

ZhenQian-keystone, 2024-05-23:
> Same here with react-native-vision-camera v4.0.5 and RN 0.74.0

yes, so confused  , dont know how to solve that

Ashikes, 2024-05-23:
> Same here with react-native-vision-camera v4.0.5 and RN 0.74.0

Try downgrading the RN version to 0.73

ZhenQian-keystone, 2024-05-24:
> > Same here with react-native-vision-camera v4.0.5 and RN 0.74.0
> 
> Try downgrading the RN version to 0.73

but my rn version is 0.71.11, meet same issue, do you konw how to fix it

ZhenQian-keystone, 2024-05-24:
why  "react-native-vision-camera": "^3.3.1" use two versions of react-native ?
```
Resolve files of :app:releaseRuntimeClasspath

Resolve files of :app:releaseRuntimeClasspath Resolve files of :app:releaseRuntimeClasspath Resolve files of :app:releaseRuntimeClasspath > react-android-0.74.1-release.aar > 44.3 MiB/143.7 MiB downloaded Resolve files of :app:releaseRuntimeClasspath Resolve files of :app:releaseRuntimeClasspath Resolve files of :app:releaseRuntimeClasspath Resolve files of :app:releaseRuntimeClasspath IDLE Resolve files of :app:releaseRuntimeClasspath Resolve files of :app:releaseRuntimeClasspath Resolve files of :app:releaseRuntimeClasspath > react-native-0.71.0-rc.0-release.aar > 38.4 MiB/70.2 MiB downloaded
```

ZhenQian-keystone, 2024-05-24:
here is my gradle version and kotlin version
```
------------------------------------------------------------
Gradle 7.5.1
------------------------------------------------------------

Build time:   2022-08-05 21:17:56 UTC
Revision:     d1daa0cbf1a0103000b71484e1dbfe096e095918

Kotlin:       1.6.21
Groovy:       3.0.10
Ant:          Apache Ant(TM) version 1.10.11 compiled on July 10 2021
JVM:          1.8.0_202 (Oracle Corporation 25.202-b08)
OS:           Mac OS X 10.16 x86_64
```

chenyunshan, 2024-05-24:
[image]
my rn 0.74.1

1500256797, 2024-05-24:
First, let's analyze the issue. 

Although your react-native version in `package.json` is less than 0.74, and your r`eact-native-vision-camera is ^3.3.1 `(the version in yarn.lock is 3.9.2), everything seems normal and there shouldn't be any problems. However, the issue still occurs.

This is because, before react-native-vision-camera v3.3.1, the android/build.gradle module in the project used the following syntax:

`implementation 'com.facebook.react:react-android:+'`

https://github.com/mrousavy/react-native-vision-camera/blob/37507345786b3b0f0b63831d499f6de3ba587190/package/android/build.gradle#L145

This line means that when building the camera package, if you don't explicitly specify the react-android version in the build.gradle file of your react-native project, Gradle will download the latest version 0.74.1 from the Maven repository at https://mvnrepository.com/artifact/com.facebook.react/react-android. This is the reason for the bug.

The solution is to use the "force" keyword to enforce the use of react-android version 0.71.11:
```
allprojects {
  configurations.all {
    resolutionStrategy {
    force "com.facebook.react:react-android:0.71.11"
    }
  }
}
```
However, sometimes the problem may not be resolved with just this step. If you are maintaining an older project that uses both react-native-vision-camera and other react-native libraries like react-native-webview, you may encounter "duplicate class" issues.

For example:

https://github.com/react-native-webview/react-native-webview/blob/master/android/build.gradle#L107

`implementation 'com.facebook.react:react-native:+' `This will download the latest version 0.71.0-rc.0 dependency from https://mvnrepository.com/artifact/com.facebook.react/react-native.

If your project has a similar situation, it will be difficult to compile because although your project references react-native and react-android, which seem unrelated, they actually have the same package names and class names. Hence, the compiler will fail.

To solve this problem, you need to change all occurrences of implementation `'com.facebook.react:react-native:+'` to `implementation 'com.facebook.react:react-android:+'` to resolve the package and class name conflicts.

Usually, we don't need to manually adjust the build.gradle file of each package. We can add the following code at the end of the android/build.gradle file in the root directory:
```
subprojects {
    project.configurations.all {
        resolutionStrategy.eachDependency { details ->
            if (details.requested.group == 'com.facebook.react' && details.requested.name == 'react-native') {
            details.useTarget group: details.requested.group, name: 'react-android', version: '0.71.11'
            }
        }
    }
}
```
This code will change all occurrences of implementation `'com.facebook.react:react-native:+' `to implementation `'com.facebook.react:react-android:0.71.11' `by renaming, resolving the package and class name conflicts.

ZhenQian-keystone, 2024-05-24:
> First, let's analyze the issue.
>
> Although your react-native version in `package.json` is less than 0.74, and your r`eact-native-vision-camera is ^3.3.1 `(the version in yarn.lock is 3.9.2), [...]

thanks , it works for me, you save me life

ZhenQian-keystone, 2024-05-24:
> ![Q636$AXS`6J 3$LJ%X FIQF](https://private-user-images.githubusercontent.com/6806308/333491928-916baaa6-1ed2-424a-b3ad-1e33f9026171.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTY1MzYxODYsIm5iZiI6MTcxNjUzNTg4NiwicGF0aCI6Ii82ODA2MzA4LzMzMzQ5MTkyOC05MTZiYWFhNi0xZWQyLTQyNGEtYjNhZC0xZTMzZjkwMjYxNzEucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDUyNCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA1MjRUMDczMTI2WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ZDM4NWQ5NThkNzQxNmNkNjVjZTliMmQ5ZGE5MTQ0MjRiZTY1ZjBjOWQwM2E5OWMzODBiZTRiY2U3N2Q0MzM0NiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.VuEoMha4gRunu-VP_w4MrpJtvbalc7GE8hzJzQCSl98) my rn 0.74.1

Try downgrading the RN version

mrousavy, 2024-05-27:
VisionCamera does not support RN 0.74 yet. See this issue for updates: https://github.com/mrousavy/react-native-vision-camera/issues/2614 and consider sponsoring if you need this sooner.

closed 2024-05-27 by mrousavy

AbdulrehmanTs, 2024-09-28:
Hey, what worked for me was: 
1. Free up space from physical Android device/emulator.
2. clear gradle cache 
```
cd android
./gradlew clean
```

!Reason of error: "Requested internal only, but not enough space": This indicates that the device or emulator you are using to run your React Native project does not have enough internal storage space to install the APK.