margelo/react-native-vision-camera #2905: 💭 Android: Frameprocessor

sairajKalkundre, 2024-05-23:
### Question

When I use useFrameProcessor in android, the frame.width is always greater than frame.height. Is this correct? In IOS I see the frame.height is greater than frame.width. 

### VisionCamera Version

4.0.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22💭+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2024-05-27:
Yes.

mrousavy, 2024-05-27:
It is correct. That's because the Frame is rotated on Android (camera sensors are just rotated).

See `orientation`.

closed 2024-05-27 by mrousavy

sairajKalkundre, 2024-05-29:
@mrousavy Just want to confirm on your comment, even if the camera is in portrait we will get orientation as Landscape-left in android?

mrousavy, 2024-05-29:
Yes because it's the Frame's orientation.
Not the phone's orientation.

sairajKalkundre, 2024-05-29:
Is their any way I can make it Portrait?

mrousavy, 2024-05-29:
what lol

the sensor orientation is landscape

sairajKalkundre, 2024-05-29:
🫣Okay, thanks for the quick response. I think I should get more native knowledge.