margelo/react-native-vision-camera #2908: 🐛 Memory leak Android

ismaelsousa, 2024-05-23:
### What's happening?

While recording the camera is consuming all memory and not releasing it.

### Reproduceable Code

```tsx
Code of shadow lens
```

### Relevant log output

```shell
--------- beginning of crash
05-23 11:18:49.858 16518 16531 E AndroidRuntime: FATAL EXCEPTION: binder:16518_2
05-23 11:18:49.858 16518 16531 E AndroidRuntime: Process: com.mrousavy.shadowlens, PID: 16518
05-23 11:18:49.858 16518 16531 E AndroidRuntime: java.lang.OutOfMemoryError: Failed to allocate a 80 byte allocation with 35808 free bytes and 34KB until OOM, target footprint 268435456, growth limit 268435456; giving up on allocation because <1% of heap free after GC.
05-23 11:18:49.858 16518 16531 E AndroidRuntime: 	at android.hardware.camera2.impl.CaptureResultExtras$1.createFromParcel(CaptureResultExtras.java:43)
05-23 11:18:49.858 16518 16531 E AndroidRuntime: 	at android.hardware.camera2.impl.CaptureResultExtras$1.createFromParcel(CaptureResultExtras.java:40)
05-23 11:18:49.858 16518 16531 E AndroidRuntime: 	at android.os.Parcel.readTypedObject(Parcel.java:3999)
05-23 11:18:49.858 16518 16531 E AndroidRuntime: 	at android.hardware.camera2.ICameraDeviceCallbacks$Stub.onTransact(ICameraDeviceCallbacks.java:148)
05-23 11:18:49.858 16518 16531 E AndroidRuntime: 	at android.os.Binder.execTransactInternal(Binder.java:1299)
05-23 11:18:49.858 16518 16531 E AndroidRuntime: 	at android.os.Binder.execTransact(Binder.java:1253)
05-23 11:18:49.878 16518 16531 I savy.shadowlens: Waiting for a blocking GC Alloc
05-23 11:18:49.889 16518 18617 I savy.shadowlens: Clamp target GC heap from 279MB to 256MB
05-23 11:18:49.889 16518 18617 I savy.shadowlens: Alloc concurrent copying GC freed 4(88B) AllocSpace objects, 0(0B) LOS objects, 0% free, 255MB/256MB, paused 167us,87us total 61.738ms
05-23 11:18:49.889 16518 16518 I savy.shadowlens: WaitForGcToComplete blocked Alloc on Alloc for 188.682ms
05-23 11:18:49.889 16518 16518 I savy.shadowlens: Starting a blocking GC Alloc
05-23 11:18:49.889 16518 18413 I savy.shadowlens: WaitForGcToComplete blocked Alloc on Alloc for 181.765ms
05-23 11:18:49.889 16518 18413 I savy.shadowlens: Starting a blocking GC Alloc
05-23 11:18:49.889 16518 18617 W savy.shadowlens: Throwing OutOfMemoryError "Failed to allocate a 2712 byte allocation with 136 free bytes and 136B until OOM, target footprint 268435456, growth limit 268435456" (VmSize 13918424 kB)
05-23 11:18:49.889 16518 16526 I savy.shadowlens: WaitForGcToComplete blocked Background on Alloc for 134.055ms
05-23 11:18:49.889 16518 18413 I savy.shadowlens: Forcing collection of SoftReferences for 64B allocation
05-23 11:18:49.889 16518 18413 I savy.shadowlens: Starting a blocking GC Alloc
05-23 11:18:49.889 16518 16518 I savy.shadowlens: Waiting for a blocking GC Alloc
05-23 11:18:49.889 16518 16591 I savy.shadowlens: Waiting for a blocking GC Alloc
05-23 11:18:49.890 16518 16531 I savy.shadowlens: Waiting for a blocking GC Alloc
05-23 11:18:49.890 16518 18617 I savy.shadowlens: Waiting for a blocking GC Alloc
05-23 11:18:49.892 16518 16528 I savy.shadowlens: Waiting for a blocking GC Alloc
05-23 11:18:49.950 16518 16768 I savy.shadowlens: Waiting for a blocking GC Alloc
05-23 11:18:49.951 16518 18413 I savy.shadowlens: Clamp target GC heap from 279MB to 256MB
05-23 11:18:49.951 16518 18413 I savy.shadowlens: Alloc concurrent copying GC freed 338(28KB) AllocSpace objects, 0(0B) LOS objects, 0% free, 255MB/256MB, paused 200us,121us total 61.158ms
05-23 11:18:49.951 16518 18413 W savy.shadowlens: Throwing OutOfMemoryError "Failed to allocate a 64 byte allocation with 28696 free bytes and 28KB until OOM, target footprint 268435456, growth limit 268435456; giving up on allocation because <1% of heap free after GC." (VmSize 13918424 kB)
05-23 11:18:49.951 16518 18413 E AndroidRuntime: FATAL EXCEPTION: ExoPlayer:Playback
05-23 11:18:49.951 16518 18413 E AndroidRuntime: Process: com.mrousavy.shadowlens, PID: 16518
05-23 11:18:49.951 16518 18413 E AndroidRuntime: java.lang.OutOfMemoryError: Failed to allocate a 64 byte allocation with 28696 free bytes and 28KB until OOM, target footprint 268435456, growth limit 268435456; giving up on allocation because <1% of heap free after GC.
05-23 11:18:49.951 16518 18413 E AndroidRuntime: 	at android.media.MediaCodec.getBuffer(Native Method)
05-23 11:18:49.951 16518 18413 E AndroidRuntime: 	at android.media.MediaCodec.getOutputBuffer(MediaCodec.java:4149)
05-23 11:18:49.951 16518 18413 E AndroidRuntime: 	at androidx.media3.exoplayer.mediacodec.AsynchronousMediaCodecAdapter.getOutputBuffer(AsynchronousMediaCodecAdapter.java:218)
05-23 11:18:49.951 16518 18413 E AndroidRuntime: 	at androidx.media3.exoplayer.mediacodec.MediaCodecRenderer.drainOutputBuffer(MediaCodecRenderer.java:1894)
05-23 11:18:49.951 16518 18413 E AndroidRuntime: 	at androidx.media3.exoplayer.mediacodec.MediaCodecRenderer.render(MediaCodecRenderer.java:816)
05-23 11:18:49.951 16518 18413 E AndroidRuntime: 	at androidx.media3.exoplayer.video.MediaCodecVideoRenderer.render(MediaCodecVideoRenderer.java:852)
05-23 11:18:49.951 16518 18413 E AndroidRuntime: 	at androidx.media3.exoplayer.ExoPlayerImplInternal.doSomeWork(ExoPlayerImplInternal.java:1047)
05-23 11:18:49.951 16518 18413 E AndroidRuntime: 	at androidx.media3.exoplayer.ExoPlayerImplInternal.handleMessage(ExoPlayerImplInternal.java:522)
05-23 11:18:49.951 16518 18413 E AndroidRuntime: 	at android.os.Handler.dispatchMessage(Handler.java:107)
05-23 11:18:49.951 16518 18413 E AndroidRuntime: 	at android.os.Looper.loopOnce(Looper.java:238)
05-23 11:18:49.951 16518 18413 E AndroidRuntime: 	at android.os.Looper.loop(Looper.java:357)
05-23 11:18:49.951 16518 18413 E AndroidRuntime: 	at android.os.HandlerThread.run(HandlerThread.java:85)
05-23 11:18:49.953 16518 16518 I savy.shadowlens: WaitForGcToComplete blocked Alloc on Alloc for 63.211ms
05-23 11:18:49.953 16518 16518 I savy.shadowlens: Starting a blocking GC Alloc
05-23 11:18:49.953 16518 16526 I savy.shadowlens: WaitForGcToComplete blocked Background on Alloc for 63.475ms
05-23 11:18:49.953 16518 16531 I savy.shadowlens: Waiting for a blocking GC Alloc
05-23 11:18:49.953 16518 18617 I savy.shadowlens: Waiting for a blocking GC Alloc
05-23 11:18:49.954 16518 16591 I savy.shadowlens: Waiting for a blocking GC Alloc
05-23 11:18:49.954 16518 16768 I savy.shadowlens: Waiting for a blocking GC Alloc
05-23 11:18:49.954 16518 16528 I savy.shadowlens: Waiting for a blocking GC Alloc
05-23 11:18:49.976 16518 16531 I savy.shadowlens: WaitForGcToComplete blocked Alloc on Background for 97.792ms
05-23 11:18:49.976 16518 16531 I savy.shadowlens: Starting a blocking GC Alloc
05-23 11:18:49.976 16518 16591 I savy.shadowlens: Waiting for a blocking GC Alloc
05-23 11:18:49.976 16518 18617 I savy.shadowlens: Waiting for a blocking GC Alloc
[... 322 lines cut ...]

```json
Code of shadow lens
- back camera
```

### Device

MOTO G52

### VisionCamera Version

Latest shadow lens

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

ismaelsousa, 2024-05-23:
Maybe related to #2866

rodgomesc, 2024-05-24:
it seems really similar to that unsupported codec thing for camera X, have you tried to play with h264 and different camera resolutions?

ismaelsousa, 2024-05-24:
it's using the `h264`

mrousavy, 2024-05-27:
Hmm interesting. CameraX is always h264. Might be a CameraX bug, I'll try to upgrade versions soon and see if that fixes anything.

Thanks for reporting!

mrousavy, 2024-05-27:
How long do you have to record before it crashes?

ismaelsousa, 2024-05-27:
I usually record 3 videos with 2-3m but my device has just 4 GB of RAM (max ~ 1.5GB available)