margelo/react-native-vision-camera #2909: 💭 [iOS] Did someone manage to get to reasonable FPS with ML Kit Pose Detection (native frame processor plugin)?

VasilP202, 2024-05-24:
### Question

I integrated MLKit Pose Detection with the help of various tutorials [(1)](https://medium.com/dogtronic/real-time-pose-detection-in-react-native-using-mlkit-e1819847c340), [(2)](https://react-native-vision-camera.com/docs/guides/frame-processors-plugins-ios).
Inside the frame processor in the React Native code, every frame I call a native function responsible for detecting the pose on the current frame. Once the result is obtained, a pose is drawn using Skia. I'm using Vision Camera V4. 
I was only able to get 8 FPS without dropping any frames. Is there a way to get a higher FPS using the current setup I described? If it's not technically possible, can you recommend another setup that also includes MediaPipe (BlazePose)?
As far as I know, ML Kit does not provide GPU support for iOS applications.

### What I tried

- I tried to optimize the detection by reducing the resolution, etc. I analyzed the calculation time of one frame - the longest part of the frame processing is the native pose detection library call itself.
- I tried [react-native-fast-tflite](https://github.com/mrousavy/react-native-fast-tflite) library with MediaPipe tflite model I got from [here](https://github.com/google-ai-edge/mediapipe/blob/master/docs/solutions/models.md#pose). I wasn't able to get it to work (detected poses were incorrect, perhaps my setup was wrong).

### VisionCamera Version

4.0.0

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22💭+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2024-05-27:
Hey - I get 90 FPS with face detection, maybe pose detection is similar.

Show your code please.

VasilP202, 2024-05-27:
@mrousavy Thank you for response. 

I managed to solve the problem today. For anyone else, in my case problem was the constant reinitialization of the detector object (`MLKPoseDetector` class in case of Pose Detection).

closed 2024-05-27 by VasilP202

techwithbob, 2024-07-06:
How do you get your FPS higher that will works in real-time?