margelo/react-native-vision-camera #2910: 💭 Lifecycle methods for native frame processors

kaancembertas, 2024-05-24:
### Question

Is there any chance to catch lifecycle of a native frame processor. Current implementation works like when a frameprocessor is registered, native frame processor creates a singleton instance. I need to reset my native data whenever camera session ends. However I could not figure that out. If there are some lifecycle callbacks like "onDispose, onCameraSessionEnd etc.", I can do what I need.

### VisionCamera Version

4.0.4

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22💭+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

nmajumder12, 2024-05-24:
> ### Question
> Is there any chance to catch lifecycle of a native frame processor. Current implementation works like when a frameprocessor is registered, native frame processor creates a singleton [...]

There is a render function within the Camera.tsx

kaancembertas, 2024-05-24:
@nmajumder12 I mean native side like swift and java, not javascript

mrousavy, 2024-05-27:
Hm this is a good feature request. Currently it's not possible - the FP Plugin will be deleted (`dealloc` on iOS, `finalize()` on Android) as soon as it is no longer needed anywhere. be careful to not make it global when you depend on such deleter methods though:

```ts
// BAD: will live forever!
export const Plugin = VisionCameraProxy.initFrameProcessorPlugin('myPlugin')

// GOOD: will live as long as the consumer component is mounted
export function usePlugin() {
  return useMemo(() => VisionCameraProxy.initFrameProcessorPlugin('myPlugin'), [])
}
```

Is this enough or do you actually need to know when the Camera stopped it's session? Why do you need that?

kaancembertas, 2024-05-27:
It seems enough for me, may be this code block should be added to documentation. I need it because I extract some timeseries data from frame processor, whenever I need another extraction session I need to clear previous calculations so that I can calculate my data properly.

Thanks! @mrousavy

mrousavy, 2024-05-27:
> may be this code block should be added to documentation

The one with hook? Yea good point, not every JS developer is familiar with how objects/memory works under the hood...

closed 2024-05-27 by mrousavy