margelo/react-native-vision-camera #2911: How to access GPS location tags in photo metadata

iamnotreddy, 2024-05-24:
### Question

I'm trying to access the GPS metadata from the photo metadata property but am unable to locate it. My understanding is that it should appear in the metadata object in a property called `'{GPS}'`. 

I went into the native code and I was able to log the expected GPS metadata, so I'm unsure if this is an actual issue or just something I'm doing wrong so I'll post my findings. Curious if anyone else has run into these issues with accessing GPS metadata and / or figured a way to resolve it. 

When I log the metadata property from the photo inside react native: 

```
const photo = await camera.current?.takePhoto();
const metadata = photo?.metadata;
console.log('Raw Metadata:', JSON.stringify(metadata));
```

I don't see the GPS metadata tag (removed some fields for brevity): 

```
Raw Metadata: {
  "DPIWidth": 72,
  "{TIFF}": {
    "ResolutionUnit": 2,
    "Software": "17.4.1",
    "Make": "Apple",
    "DateTime": "2024:05:24 10:25:21",
    "XResolution": 72,
    "HostComputer": "iPhone 13",
    "Model": "iPhone 13",
    "YResolution": 72
  },
  "kCGImageDestinationICCProfile": null,
  "{MakerApple}": {
   ...
  },
  "DPIHeight": 72,
  "Orientation": 6,
  "{Exif}": {
    "DateTimeOriginal": "2024:05:24 10:25:21",
    "ExposureTime": 0.0017699115044247787,
    "FNumber": 1.6,
    ...
  }
}
```

I went into the native code and I was able to log the GPS data inside the `MetadataProvider` file. This is what its`replacementMetadata` function returns (also abridged some fields):

```
["DPIHeight": 72, 
"kCGImageDestinationICCProfile": ..., 
"{MakerApple}": {
    1 = 14;
    ...
}, 
"DPIWidth": 72, 
"{Exif}": 
["WhiteBalance": 0, "UserComment": "Captured with VisionCamera by mrousavy", 
....
"DateTimeDigitized": 2024:05:24 10:25:21], 
"Orientation": 6, 
"{TIFF}": {
    DateTime = "2024:05:24 10:25:21";
    HostComputer = "iPhone 13";
    ...
}, 
"{GPS}": {
    Altitude = "10.41956996917725";
    AltitudeRef = 0;
    DOP = [redacted]
    DateStamp = "2024:05:24";
    ImgDirection = "188.7100982666016";
    ImgDirectionRef = T;
    Latitude = [redacted]
    LatitudeRef = N;
    Longitude = [redacted];
    LongitudeRef = W;
    TimeStamp = "10:25:21";
}]
```
The GPS data above appears to be correct and is what I would hope to see inside the photo metadata. This also makes me think it's not a permissions issue on my end. 

Something else I noticed in the native code: when I log `photo.metadata` inside the `photoOutput` function inside `PhotoCaptureDelegate` file, the GPS tag does NOT appear (but I see EXIF, TIFF etc.). 

I don't have too much understanding of the native code and how it all ties together but hoping this information is helpful. Please let me know if any additional context is needed from my side.

### What I tried

I've ensured the following:
- `enableLocation` is set to true in the `<Camera>` component
-  added `NSLocationWhenInUseUsageDescription` to `Info.plist`
-  confirmed I'm correctly obtaining location permissions: `hasPermission` is true from the `useLocationPermission` h

### VisionCamera Version

4.0.5

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22💭+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2024-05-27:
Hm well maybe it just doesn't have the replacement data at this point yet.
It's definitely in the file, so if you read it you will get GPS metadata.

Also on Android there's no way to get that metadata currently - which sucks. So this is not really a public API at all.

iamnotreddy, 2024-05-29:
@mrousavy - appreciate you getting back to me!

just to clarify, are referring to reading the GPS metadata from [this photo file](https://react-native-vision-camera.com/docs/guides/taking-photos#getting-the-photos-data)? my original understanding was that I could read the GPS metadata from the photo [metadata](https://react-native-vision-camera.com/docs/api/interfaces/PhotoFile#metadata) so that might make sense why I'm not seeing it

aesqe, 2024-06-02:
@iamnotreddy I also don't see the GPS data if I `console.log` the `camera.takePhoto` output. However, @mrousavy is right and the data is contained in the file metadata. https://github.com/francisco-sanchez-molina/react-native-exif can read that data and the values are correct in my case, at least on the latest iOS using iPhone 13.

iamnotreddy, 2024-06-03:
Thanks for confirming on the `takePhoto` output @aesqe! That makes sense, I will try to get the GPS tag from the file metadata