margelo/react-native-vision-camera #2915: 🐛  takePhoto() function working very slowly on Android

Jonnboy91, 2024-05-27:
### What's happening?

Using: const photo = await cameraRef.current?.takePhoto() and pressing the photo button will take about 3-4 seconds before the photo is being shown. You have enough time to click on the take photo button again.
On the Reproduceable Code, the console logs are about 3-4 seconds apart.

I tried just taking a photo with the example app and get this:
ERROR  Failed to take photo! [unknown/unknown: Capture request failed with reason ERROR]
With my front-camera it seemed to work but with back camera I get this error ^
Front-camera also took a while to load up.

### Reproduceable Code

```tsx
import { Camera as VisionCamera } from 'react-native-vision-camera'

const cameraRef = useRef<VisionCamera>(null)

console.log('takePhoto')
const photo = await cameraRef.current?.takePhoto()
console.log('photo', photo)
```

### Relevant log output

```shell
-
```

### Camera Device

```json
{
  "formats": [],
  "sensorOrientation": "landscape-left",
  "hardwareLevel": "full",
  "maxZoom": 10,
  "minZoom": 1,
  "maxExposure": 18,
  "supportsLowLightBoost": false,
  "neutralZoom": 1,
  "physicalDevices": [
    "wide-angle-camera"
  ],
  "supportsFocus": true,
  "supportsRawCapture": false,
  "isMultiCam": false,
  "minFocusDistance": 10,
  "minExposure": -18,
  "name": "2 (BACK) androidx.camera.camera2",
  "hasFlash": true,
  "hasTorch": true,
  "position": "back",
  "id": "2"
}
```

### Device

One Plus 12R

### VisionCamera Version

4.0.5

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

EwaGuziejko, 2024-05-27:
I have the same issue - on Realme it takes up to 8 sec. Any ideas how to fix it would be really appreciated

mrousavy, 2024-05-27:
Setting `photoQualityBalance="speed"` is the fastest way to capture photos.
Maybe also choose a `format` with lower photo resolution if you want faster captures.

Other than that, this is what Android gives me. My API is extremely fast and it's up to Android to implement improvements on the Camera layer.

closed 2024-05-27 by mrousavy

Jonnboy91, 2024-05-27:
I've set the photoQualityBalance="speed" already as well, that didn't work here. I haven't tested the format yet, will have a look at those.
If those don't seem to do any difference in speed, I will check the logs and send them here.